iiab/iiab #2484: OpenStreetMap (OSM) stalls all IIAB installs for about 38 minutes?

holta, 2020-08-20:
@georgejhunt can you give more context as to why this is suddenly now happening?  IIAB installs all stall without warning for about 38 minutes here:

```
TASK [osm-vector-maps : Install the World Map to zoom 10] **********************
```

What options might there be to eliminate this stage or speed it up, so we avoid scaring away our most important allies i.e. newcomers ?!

Problem detail/summary: https://github.com/iiab/iiab/issues/2481#issuecomment-676820225

tim-moody, 2020-08-20:
There are a couple of big data downloads as part of install. This one and kalite EN. I wonder if it makes sense to put them on the adm cons job queue; would need a script, but probably want that anyway, For me the data comes from iarchive at about 1.5Mbs, so pretty slow. Maybe we need another place for it or at least for world to 10..

holta, 2020-08-20:
True, but for comparison, KA Life's en.zip downloads in 60-90 seconds typically.

(Whereas this OSM task appears to be much more than an order-of-magnitude slower, i.e. 38 min is "Not Ok" most everyone, newcomers & all, needing to kick the tires very early on!)

holta, 2020-08-28:
Resolved.

closed 2020-08-28 by holta