iiab/iiab #2496: http://box/osm-vector-maps/installer/ often (1) offers each Map Pack twice (2) shows no Map Packs at all

holta, 2020-08-28:
@jvonau showed me this is reproducible almost every time in Firefox, after cookies are deleted &mdash; i.e. what appears on the left side under "Select One at a Time".

If a 2nd tab is quickly opened, also with URL http://box/osm-vector-maps/installer/ this also tends also show each available Map Pack twice.

If a 3rd tab (and any additional tabs, also with URL http://box/osm-vector-maps/installer/) are opened, they usually fail to show _any_ available Map Packs :/  (In contrast, these 3rd-and-additional tabs tend to work in the Chrome browser.)

CLARIFS:

(1) The 1st tab (after Firefox cookies are deleted) sometimes shows each of the 12 Map Packs exactly once each (properly) but this seems to be rare :/

(2) These problems with http://box/osm-vector-maps/installer/ were confirmed in Firefox-on-Linux and Firefox-on-Windows.  Unclear why these problems appear much worse in Firefox than Chrome.

holta, 2020-08-31:
Is this possibly solved by PR https://github.com/iiab/maps/pull/27 ?  ("7.2 maps...fix a race condition in getting map-catalog.json")

georgejhunt, 2020-08-31:
> Is this possibly solved by PR iiab/maps#27
> <https://github.com/iiab/maps/pull/27> ? ("7.2 maps...fix a race
> condition in getting map-catalog.json")

Thanks for your help. It's much easier to fix something when it fails all
the time, so testing it on firefox was the breakthrough.

You will need to run "./runroles osm-vector-maps" to fix a machine that is
already installed once

holta, 2020-08-31:
Fantastic @georgejhunt if this is likely fixed.

Thanks to @jvonau for noticing the situation with Firefox.

closed 2020-08-31 by holta