iiab/iiab #2497: Concerning 1 line install on Raspberry pi 3b+

pumpkinpi85, 2020-08-28:
Architecture: raspberry pi 3b+ > 1tb expansion board (no micro sd > booting from usb)

What am I dealing with here:

curl d.iiab.io/install.txt | sudo bash 

(see image)

[image]

holta, 2020-08-28:
@pumpkinpi85 can you please provide the URL that results from running the `iiab-diagnostics` command?

(Run it as root if possible, Thanks!)

pumpkinpi85, 2020-08-28:
Thanks @holta. Is this what you're looking for? 

COMPLETE! Your diagnostics file (, 0 lines) is:

   /etc/iiab/diag/7.2_raspbian-10_2020-08-27_22:56:52_MDT_sudo iiab-diagnostics

Publish it to a web pastebin? [Y/n] y

PUBLISHING TO URL... /usr/bin/iiab-diagnostics: line 238: $outfile: ambiguous redirect

holta, 2020-08-28:
Hi @pumpkinpi85, let's try to understand more:

- What exact version of Raspberry Pi OS are you running?
  - 32-bit or the new/experimental 64-bit?
  - Lite or "with desktop" ?
  - Is your OS from 2020-08-20 at https://www.raspberrypi.org/downloads/raspberry-pi-os/ _(or an earlier version?)_
- What's the full output of `systemctl status kolibri` please, with a screenshot/photo if that's easiest?
- Did your Raspberry Pi 3 B+ run out of memory (RAM) ?  That's one possibility anyway, as Kolibri (part of BIG-sized IIAB installs) can sometimes be very resource-intensive.
  - Any chance you might have access to a Raspberry Pi 4 with 2GB or more of memory, to try installing there?

Thanks !

jvonau, 2020-08-28:
> /etc/iiab/diag/7.2_raspbian-10_2020-08-27_22:56:52_MDT_sudo iiab-diagnostics

I reproduced the iiab-diagnostics issue locally. Please don't use a file name with spaces in it.

jvonau, 2020-08-28:
> @pumpkinpi85 can you please provide the URL that results from running the `iiab-diagnostics` command?
> 
> (Run it as root if possible, Thanks!)

Always use `sudo iiab-diagnostics` to ensure all the information can be gathered.

holta, 2020-08-28:
@jvonau is right on.

@pumpkinpi85 can you please try again, as follows:

1) Run `sudo iiab-diagnostics`
2) When it asks for a nickname, _do not use any spaces._
3) Reply here with the resulting pastebin's URL, Thanks!

pumpkinpi85, 2020-08-28:
> @pumpkinpi85 can you please try again, as follows:
> 
> 1. Run `sudo iiab-diagnostics`
> 2. When it asks for a nickname, _do not use any spaces._
> 3. Reply here with the resulting pastebin's URL, Thanks!

My apologies for that oversight and thank you kindly for you support. Here is the resulting past bin's URL:

PUBLISHING TO URL... /usr/bin/pastebinit:42: DeprecationWarning: dist() and linux_distribution() functions are deprecated in Python 3.5
  release = platform.linux_distribution()[0].lower()
/usr/bin/pastebinit:413: DeprecationWarning: pasteURLopener style of invoking requests is deprecated. Use newer urlopen functions/methods
  url_opener = pasteURLopener()
http://sprunge.us/iWL3oP

holta, 2020-08-28:
Thanks @pumpkinpi85 that helps a lot:

1) Can you please also provide the full output of `systemctl status kolibri` ?

2) Does the situation (output just above) improve if you run `systemctl start kolibri` or reboot?

holta, 2020-08-28:
CLARIF: it's probably necessary to become root for (2) above. 
 So run this with sudo to be safe:

`sudo systemctl start kolibri`

pumpkinpi85, 2020-08-28:
> * What exact version of Raspberry Pi OS are you running?
>
>   * 32-bit or the new/experimental 64-bit?
>   * Lite or "with desktop" ?
>   * Is your OS from 2020-08-20 at [...]

- I am running the latest Raspberry OS with recommended software and desktop (booting from USB 1TB expansion board; not micro SD card)
- (see image for 'systemctl status kolibri')
- I do not have a RPi 4 unfortunately 
- Architecture is as follows: raspberry pi 3b+, X820 Metal Case/Enclosure + Power Control Switch + Cooling Fan Kit, X820 V3.0 SSD SATA Storage Board, 1TB expansion storage & X735 Power Management & Auto Cooling Expansion Board With Safe Shutdown
- Concerning memory (see images)

[image]

[image]

[image]

[image]

pumpkinpi85, 2020-08-28:
> CLARIF: it's probably necessary to become root for (2) above.
> So run this with sudo to be safe:
> 
> `sudo systemctl start kolibri`

root@iiab:/home/pi# sudo systemctl start kolibri
Job for kolibri.service failed because the control process exited with error code.
See "systemctl status kolibri.service" and "journalctl -xe" for details.
root@iiab:/home/pi#

jvonau, 2020-08-28:
@holta 
>ansible_version = 2.9.12

holta, 2020-08-28:
@pumpkinpi85 you're running a buggy version of Ansible (2.9.12) that IBM / Red Hat unfortunately pushed a couple weeks ago.  Very sorry they did that, and in fact are overhauling their release process as a result.  Background: #2481, #2479

I'd highly recommend you do a completely fresh install, as we implemented a workaround earlier this week (#2482) to use Ansible 2.9.6 instead!

ASIDE/OPTIONAL/RECOMMENDED: new OS images for Raspberry Pi OS are available since earlier this week — in both 32-bit and 64-bit flavors — definitely consider these: https://github.com/iiab/iiab/wiki/IIAB-Platforms#operating-systems

holta, 2020-08-28:
@pumpkinpi85 you should be all set, installing IIAB on a fresh OS.

So I'm closing this for now.

However feel free to re-open (or start a new ticket) if any questions arise!

pumpkinpi85, 2020-08-28:
> @pumpkinpi85 you should be all set, installing IIAB on a fresh OS.
> 
> So I'm closing this for now.
> 
> However feel free to re-open (or start a new ticket) if any questions arise!

I did a fresh install of Raspberry OS, but am now facing this on the 1 line install of iiab:

[image]

holta, 2020-08-28:
Apologies pumpkinpi85 that slipped thru, as a result of PR #2495.

@jvonau should definitely be able to solve this for you later today, or overnight.

Thanks for your patience!

jvonau, 2020-08-29:
Reproduced on aarm32
> grep arch /opt/iiab/iiab/iiab-install.log | more
        "ansible_architecture": "x86_64", 
            "cap_dac_read_search", 
        "ansible_userspace_architecture": "x86_64", 
        "ansible_architecture": "x86_64", 
            "cap_dac_read_search", 
        "ansible_userspace_architecture": "x86_64", 

>iiab-admin@box:~ $ sudo grep arch /opt/iiab/iiab/iiab-install.log | more
        "ansible_architecture": "aarch64", 
        "ansible_machine": "aarch64", 
            "cap_dac_read_search", 
        "ansible_architecture": "aarch64", 
        "ansible_machine": "aarch64", 
            "cap_dac_read_search", 

First time I ran into a situation that a variable exists on one arch but not the other. #2498 needs to be merged first.
Once this is merged `cd /opt/iiab/iiab` `sudo git pull origin master` `sudo iiab` to continue

jvonau, 2020-08-29:
Ok now the fix is merged proceed with the above instructions. Sorry for my opps and thanks for installing IIAB, happy testing.

holta, 2020-08-29:
Thanks @pumpkinpi85: you should be all set to finalize your IIAB install now, per @jvonau's instructions above.

Let us know if any questions.

pumpkinpi85, 2020-08-30:
@holta & @jvonau 

Installation was a success (according to "IIAB INSTALLATION (/usr/sbin/iiab) IS ALREADY COMPLETE -- per existence of:
/etc/iiab/install-flags/iiab-complete -- nothing to do.")

However might there be anything about my setup that prevents a successful installation of the access point? Client computer, side-by-side to the RPi that includes a wifi dongle, cannot "see" the network ssid (I do know the means of creating an access point, but I know not enough if doing so will interfere with the functionality of IIAB). Everything depends on it. Please advise.

[image]
[image]

holta, 2020-08-30:
Hi @pumpkinpi85

1) After inserting the WiFi dongle into your Raspberry Pi, did you run the following?  (As required when networking hardware changes.)

   ```
   cd /opt/iiab/iiab
   sudo ./iiab-network
   ```

2) If the above doesn't work, please again run `sudo iiab-diagnostics` and send us the resulting URL, Thanks!

jvonau, 2020-08-30:
> @holta & @jvonau
>
> Installation was a success (according to "IIAB INSTALLATION (/usr/sbin/iiab) IS ALREADY COMPLETE -- per existence of:
> /etc/iiab/install-flags/iiab-complete -- nothing to do.") [...]

Everything looks normal for a RPi3B+ using the internal WiFi as a Hotspot. (ap0 in the br0, hostapd & dnsmasq running).
>Client computer, side-by-side to the RPi that includes a wifi dongle, cannot "see" the network ssid 

Can any other device in your house like a cellphone "see the ssid"? Can you clarify where this 'WiFi dongle' is attached to? From the dump it doesn't look it's attached to the IIAB, but have to ask to draw a clearer picture. 
>(I do know the means of creating an access point, but I know not enough if doing so will interfere with the functionality of IIAB). Everything depends on it. Please advise.
> 
> http://sprunge.us/uoPD0Z

Wonder if the wifi firmware doesn't like the CA country code..... As a quick test can you
edit /etc/wpa_supplicant/wpa_supplicant.conf,  change the country= line to be US, save
edit /etc/iiab/local_vars.yml, change the host_country_code = line to be US, save
`cd /opt/iiab/iiab` `sudo ./iiab-network` to refresh the settings. `reboot`

pumpkinpi85, 2020-08-30:
By way of ./iiab-network, changing the country code and a system reboot - I was able to see the ssid network (not password protected for some reason) from a client computer and mobile phone, but "unable to join network" in both cases. 

It also seems as though that I only had one attempt to connect as since I can no longer see the network ssid. (update: if I run 
"./iiab-network" again I can see the unprotected network ssid, but fails to connect and disappears)

The wifi dongle is a USB interface.

http://sprunge.us/b1WN1p

holta, 2020-08-30:
@pumpkinpi85 & @jvonau

Let's arrange a call if that might be much more efficient?

And or direct connection to Nathan's IIAB (using OpenVPN) if he wants that!

If either helps, send an email at the bottom here: http://wiki.laptop.org/go/IIAB/FAQ#What_are_the_best_places_for_community_support.3F

jvonau, 2020-08-30:
> The wifi dongle is a USB interface.

What is the intended use of this interface? and why does it not appear in the dump?

>gw_active = False
internet_available = False

Ok, so the network cable was unattached

>[   12.679646] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  1 2015 07:29:38 version 7.45.18 (r538002) FWID 01-6a2c8ad4
[   18.538016] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[  403.420329] ieee80211 phy0: brcmf_cfg80211_stop_ap: setting AP mode failed -52
[  406.785752] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[  420.841323] ieee80211 phy0: brcmf_vif_set_mgmt_ie: vndr ie set error : -52
[  420.841340] ieee80211 phy0: brcmf_cfg80211_scan: scan error (-52)

Were you by chance trying to make an upstream wifi connection? Lets cut to the chase here, what are you trying to setup?
This is usually caused by trying to use the wifi as a client while hostapd is running, both can run together just can't scan as a client wifi but known connections connect ok while the hostapd "server" is running. Since I see a desktop installed with a terminal open do this:
`sudo iiab-hotspot-off`
configure your upstream connection via rasp-config or the gui, ensure the connection works
`sudo iiab-hotspot-on` 
`reboot`
Don't bother testing without a reboot.

> iiab_network_mode_applied = LanController

I don't think you want that. Once rebooted ensure your gateway works and re-run iiab-network to change the above to "Gateway"

holta, 2020-08-30:
> http://wiki.laptop.org/go/IIAB/FAQ#What_are_the_best_places_for_community_support.3F

Recap: If a call is easier...don't hesitate (:

pumpkinpi85, 2020-08-30:
> > http://wiki.laptop.org/go/IIAB/FAQ#What_are_the_best_places_for_community_support.3F
> 
> Recap: If a call is easier...don't hesitate to email (:

Message on 'Unleash Kids' gmail Group?

pumpkinpi85, 2020-08-30:
> > http://wiki.laptop.org/go/IIAB/FAQ#What_are_the_best_places_for_community_support.3F
> 
> Recap: If a call is easier...don't hesitate to email (:

@holta I sent you an email

holta, 2020-08-30:
@pumpkinpi85 no email arrived, try again if poss!

pumpkinpi85, 2020-08-30:
Sent to holt@unleashkids.org from Nathan.koops@gmail.com

pumpkinpi85, 2020-08-31:
To date the issue lies with the existing kernel. <sudo rpi-update> to fix the matter of the RPi dropping packets. Once complete you should be able to connect to the network without issue. Open browser input <http://172.18.96.1> and away you go. Thanks to @holta & @jvonau for plugging our way though this one. Good work guys!

jvonau, 2020-08-31:
"Linux iiab.lan 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l" has this issue
http://box.lan should work in place of http:/172.18.96.1

jvonau, 2020-08-31:
5.4.51-v8+ in the subject line is just speculation as the problem is with the booted armv7l kernel, the kernels are not the same but come from the same deb package.

jvonau, 2020-09-01:
kernel 1333 issue in a nutshell client wifi is flaky out of the box.
pi@iiab:~ $ ping google.ca
PING google.ca (172.217.14.195) 56(84) bytes of data.
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=1 ttl=118 time=43.6 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=2 ttl=118 time=40.0 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=3 ttl=118 time=34.9 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=4 ttl=118 time=37.7 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=5 ttl=118 time=36.6 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=6 ttl=118 time=35.7 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=7 ttl=118 time=36.5 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=8 ttl=118 time=35.8 ms
ping: sendmsg: Network is unreachable
ping: sendmsg: Network is unreachable
ping: sendmsg: Network is unreachable
ping: sendmsg: Network is unreachable
ping: sendmsg: Network is unreachable
ping: sendmsg: Network is unreachable
ping: sendmsg: Network is unreachable
ping: sendmsg: Network is unreachable
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=21 ttl=118 time=43.2 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=22 ttl=118 time=38.4 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=23 ttl=118 time=36.3 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=24 ttl=118 time=37.2 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=25 ttl=118 time=40.5 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=26 ttl=118 time=36.1 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=27 ttl=118 time=36.3 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=28 ttl=118 time=36.6 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=29 ttl=118 time=37.7 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=30 ttl=118 time=36.1 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=31 ttl=118 time=43.8 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=32 ttl=118 time=36.9 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=33 ttl=118 time=35.7 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=34 ttl=118 time=36.1 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=35 ttl=118 time=36.7 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=36 ttl=118 time=35.6 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=37 ttl=118 time=36.1 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=38 ttl=118 time=37.1 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=39 ttl=118 time=35.3 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=40 ttl=118 time=36.3 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=41 ttl=118 time=36.6 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=42 ttl=118 time=35.7 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=43 ttl=118 time=35.4 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=44 ttl=118 time=36.2 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=45 ttl=118 time=35.6 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=46 ttl=118 time=37.9 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=47 ttl=118 time=36.0 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=48 ttl=118 time=35.5 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=49 ttl=118 time=35.10 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=50 ttl=118 time=36.4 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=51 ttl=118 time=35.7 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=52 ttl=118 time=36.4 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=53 ttl=118 time=36.3 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=54 ttl=118 time=36.7 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=55 ttl=118 time=35.8 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=56 ttl=118 time=36.7 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=57 ttl=118 time=37.9 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=58 ttl=118 time=36.8 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=59 ttl=118 time=41.5 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=60 ttl=118 time=39.0 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=61 ttl=118 time=39.1 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=62 ttl=118 time=35.7 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=63 ttl=118 time=36.8 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=64 ttl=118 time=35.4 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=65 ttl=118 time=36.0 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=66 ttl=118 time=35.2 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=67 ttl=118 time=36.0 ms
ping: sendmsg: Network is unreachable
ping: sendmsg: Network is unreachable
[... cut ...]
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=147 ttl=118 time=36.4 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=148 ttl=118 time=35.4 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=149 ttl=118 time=37.1 ms
64 bytes from sea30s01-in-f3.1e100.net (172.217.14.195): icmp_seq=150 ttl=118 time=35.3 ms
^C
--- google.ca ping statistics ---
150 packets transmitted, 112 received, 25.3333% packet loss, time 910ms
rtt min/avg/max/mdev = 34.946/36.794/43.825/1.711 ms

holta, 2020-09-03:
Somewhat related:

#2497 "IIAB's 1-line installer should *require* a recent OS e.g. (a recent version of) Raspbian Buster"

holta, 2020-09-08:
1) This Raspberry Pi 3 B+'s Ethernet was failing due to inadequate electrical power (that part is now solved, by using a strong power supply, rather than PoE).

2) Its WiFi also has ongoing issues.  On @jvonau's suggestion, @pumpkinpi85 is looking into turning off or renaming the 5 GHz WiFi AP that appears to be emanating from his home WiFi router -- so that WiFi client devices (like the 3 B+ IIAB here) don't connect "randomly each time" to either 2.4 GHz or 5 GHz AP's.

   i.e. this non-deterministic behavior creates many problems when IIAB must create its own AP/hotspot...which by necessity MUST use the same frequency (2.4 GHz or 5 GHz) as the upstream WiFi-to-Internet.

holta, 2020-09-10:
@pumpkinpi85 can we close this ticket?

Or if you have any remaining WiFi issues, please summarize including details so others can fully reproduce the situation on their own Raspberry Pis.

Many Thanks!

holta, 2020-09-12:
FYI @pumpkinpi85 is currently trying a new install of http://download.iiab.io (IIAB 7.2 pre-release) onto 'RaspiOS with desktop' on RPi 3 B+ &mdash; and he's running into a WiFi issue again (intermittent failures of the WiFi, even if not as often as last time a week ago?)

It would (seem) this is the RaspiOS kernel #1333 WiFi issue again...and if so, apparently this WiFi issue completely unrelated to IIAB.

So he'll look into 'rpi-update' as a possible workaround, thanks to the new 5.4.64 kernel just pushed today by https://github.com/Hexxeh/rpi-firmware/commits/master ?

ASIDE: Not *directly* unrelated, but just fyi the July 2020 research into RaspiOS WiFi firmwares is here: https://github.com/iiab/iiab/issues/823#issuecomment-662285202

holta, 2020-09-12:
2 questions for @pumpkinpi85 if you have time!

1. Can you please post the URL arising from `sudo iiab-diagnostics` on your new fresh install, so that others might understand/diagnose/assist?

2. Perhaps the most key info of all: can you reproduce any of these WiFi failures on a vanilla install of RaspiOS (e.g. are there any failures of 'apt update' and 'ping 8.8.8.8' ...while of course verifying that you're really running RaspiOS kernel #1333 ...prior to running 'rpi-update' AND most important, **prior to install IIAB?**

   _(If you possibly get to testing that in coming hours/days...that would be insanely valuable!)_

jvonau, 2020-09-12:
With me shining up my remote viewing crystal ball, would @pumpkinpi85 by change have both eth0 and wlan0 active(connected to a router) at the same time prior to starting the install?

holta, 2020-09-12:
@pumpkinpi85 can you tell if PR #2515 ("detect secondary gateway on wifi interface") might...or might not...be relevant to any of your WiFi challenges?

Thanks much for clarifying!

holta, 2020-09-16:
Until we know more, I've added this warning in [IIAB 7.2 Release Notes > Known Issues](https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes#known-issues):

> Raspberry Pi OS kernel `#1333` (as shown by `uname -a` output) can cause WiFi outages every minute or so, e.g. with shaw.ca routers. While this bug is unrelated to IIAB, in September 2020 it can cause sporadic havoc (with your IIAB learning hotspot, and also depending on what WiFi router you use to connect to the Internet?) The short-term workaround is to run [rpi-update](https://github.com/Hexxeh/rpi-update) which will quickly upgrade your IIAB kernel (if you dare!) #2497

holta, 2020-09-18:
Just for comparison, the current kernel used by Ubuntu (20.04.1) on RPi is:

```
root@box:~# uname -a
Linux box.lan 5.4.0-1018-raspi #20-Ubuntu SMP Sun Sep 6 05:11:16 UTC 2020 aarch64 aarch64 aarch64 GNU/Linux
```

I'd assume the above OS/kernel is another option for people trying to work around RaspiOS's `#1333` WiFi problems (but I might be wrong!)

holta, 2020-09-20:
Preliminary evidence from @pumpkinpi85 indicates the flaky WiFi issue is solved by RaspiOS's new `#1333` kernel.

(Thanks @pumpkinpi85 for confirming & hopefully closing this tkt in coming days!)

jvonau, 2020-09-27:
The saga continues, the new replacement install, now without upstream WiFi, is having hostap|dnsmasq connection issues like issues without having run raspi-update.

holta, 2020-09-27:
1) Profound thanks to @pumpkinpi85 & @jvonau who have continued working on this.

2) @jvonau theorizes that VNC (might) be interfering: i.e. is it possible VNC's "2nd desktop" is disrupting the WiFi connection(s) ?

3) IIAB 7.2 will be released within about a week, so worst case let's gather the facts (shining the brightest light we can) to improve our writeup (of proven cause+effect patterns) for the "Known Issues" section at the bottom of https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes#known-issues

holta, 2020-09-27:
FYI @pumpkinpi85 is "still not seeing the access point" (IIAB's hotspot for "student" WiFi devices) on his RPi 3 B+, where a BIG-sized IIAB was installed a week ago.

Its [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) output is here: http://sprunge.us/WFWhzr?en

Thanks to @jvonau who connected over ssh, reviewed the machine's logs, and is asking @pumpkinpi85 to look into these 2 lines in his /etc/iiab/local_vars.yml :

> hostapd_secure: true
> hostapd_password: [REDACTED]

i.e. it's conceivable that the above password (Line 56 of local_vars.yml) might need to be quoted and/or contain less punctuation?

Thanks @pumpkinpi85 for persisting in the hunt, so we can hopefully explain this to all soon.

holta, 2020-09-27:
@pumpkinpi85 reverted to `hostapd_secure: False` in /etc/iiab/local_vars.yml then ran `./iiab-network` and believes his RPi 3B+ hotspot is still failing.

This is confusing as `systemctl status hostapd` (claims!) the hotspot is working.

Here are some questions that might help us get to the bottom of this: (when @pumpkinpi85 has time!)
- Specifically how is hotspot being tested to determine it's not working?
- When was the hotspot last working?
- What action appears to have caused the hotspot to break?
- Is this repeatable when moving the microSD card to his other RPi's?
- Is this repeatable when (quickly) installing a new (MIN-sized) IIAB onto another microSD card, in a different Raspberry Pi?

jvonau, 2020-09-27:
> FYI @pumpkinpi85 is "still not seeing the access point" (IIAB's hotspot for "student" WiFi devices) on his RPi 3 B+, where a BIG-sized IIAB was installed a week ago.
> > hostapd_secure: true
> > hostapd_password: [REDACTED]
> 
> i.e. it's conceivable that the above password (Line 56 of local_vars.yml) might need to be quoted and/or contain less punctuation?

You tested the change to using what I detected the presence of and the connection still worked, you left out:
Adam:
>What's unique about Nathan's 10.8.0.46 RPi 3 B+ that is different from this 10.8.0.42 RPi 4 right in front of me? I don't know :/ -- so far everything seems to work on this end.

Jerry:
>odd man out is the unique hardware addons. Can't recall, does this device use an overlay that you have to declare in /boot/config.txt?
or maybe power issue?

jvonau, 2020-09-28:
>     * Is this repeatable when moving the microSD/disk to other Raspberry Pis?
This is a usb-booted rpi3 with a "hat" installed and a 1TB drive, the last go around it took raspi-update, which I suspect has other usb/sdio improvement applied that 1333 does not have, to get wifi fully working with that hardware combo.
On a side note might want to set "iiab_usb_lib_show_all" to False in local_vars.yml otherwise the the root filesystem is exposed to viewing at http://box/usb via usbmount.

holta, 2020-09-28:
@pumpkinpi85 can you schedule time for a call this week, to try to work thru this?  (If you can!)

I'm sure @jvonau can solve this, if time allows.  If you don't suddenly solve it alone ;)

holta, 2020-09-28:
> the last go around it took raspi-update, which I suspect has other usb/sdio improvement applied that 1333 does not have, to get wifi fully working with that hardware combo

While generally a risky activity, if there's a need to experiment with [rpi-update](https://github.com/Hexxeh/rpi-update) as in early September, note that it just pushed a new 5.4.68 kernel: https://github.com/Hexxeh/rpi-firmware/commits/master

jvonau, 2020-09-28:
>Concerning 1 line install on Raspberry pi 3b+ [kolibri.service failed to start due to Ansible 2.9.12; 'ansible_userspace_architecture' is undefined; WiFi flaky due to RaspiOS kernel 5.4.51-v7+ #1333 ?] 

while equipped with 3'rd party 'hat hardware' to provide upstream OS supported usb-boot feature.

The upper software stack may never work correctly should the lower level hardware be configured incorrectly. To make this 'work | no work' distinction this needs to be stripped back to the just the new hardware and a stock install of the OS, with a hand installation and configuration of hostapd. I'll help guide that part to get to the bottom of this issue, at that point if hostapd doesn't work, file an issue somewhere upstream, bug the hardware vendor for better instructions, or give rpi-update a shot. I'm done putting any more effort into debugging this as overlays are not my specialty and don't want to lead anybody astray.

jvonau, 2020-09-28:
>     2\. i.e. this non-deterministic behavior creates many problems when IIAB creates its internal AP/hotspot...which by necessity MUST use the same frequencyor 5 GHz) as the upstream WiFi-to-Internet.

I don't recall if I corrected this earlier, hostapd is setup to uses 2.4 GHz ONLY, so the upstream WiFi-to-Internet connection **has** to be 2.4GHz

holta, 2020-09-28:
> I don't recall if I corrected this earlier, hostapd is setup to uses 2.4 GHz ONLY, so the upstream WiFi-to-Internet connection **has** to be 2.4GHz

Good to know.

While these warnings might not be 100% perfect poetry, but I think they're sufficient for now:
- PRs #2508 -> #2509 -> #2510
- PR iiab/iiab-factory#139

holta, 2020-11-15:
NEW WARNING (re: 2.4GHz WiFi interference with USB 3.0 boot devices) ADDED TO:

- http://FAQ.IIAB.IO > "44 Can I install IIAB onto a USB drive?"

(@jvonau & others please help me refine this warning if new info arises, Thanks!)

jvonau, 2021-07-21:
close

closed 2021-07-21 by holta