iiab/iiab #2502: Key Error when reindexing Kiwix content

ganastor, 2020-09-02:
### Expected behavior and actual behavior:
Expected: rebuild library.xml with new zim
Actual: key error on zim

### Steps to reproduce the problem:
Downloaded wikipedia_en_all_nopic_2020-08.zim (also tried 2020-07) to /library/zims/content, and getting this error. As a test I also downloaded wikipedia_an_all_nopic_2020-08.zim and it worked fine

### IIAB version, OS version, hardware details:
IIAB version - master/latest, on Raspberry Pi OS, Raspberry Pi 3 B+
### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

iiab-diagnostics: https://pastebin.com/LDqixbHg

Actual error message:

```
File "/usr/bin/iiab-make-kiwix-lib.py", line 85, in
main()
File "/usr/bin/iiab-make-kiwix-lib.py", line 70, in main
adm.write_zim_versions_idx(zim_versions, kiwix_library_xml, zim_version_idx_dir, zim_menu_defs)
File "/usr/lib/python3/dist-packages/iiab/adm_lib.py", line 38, in write_zim_versions_idx
get_substitution_data(perma_ref, zim_versions, zims_installed, path_to_id_map)
File "/usr/lib/python3/dist-packages/iiab/adm_lib.py", line 97, in get_substitution_data
zim_id = path_to_id_map[path]
KeyError: 'content/wikipedia_en_all_nopic_2020-08.zim'
Finished making Kiwix library.xml
```
Contents of content directory:
```
$ ls -alh /library/zims/content

total 38G
drwxrwxrwx 2 pi         pi         4.0K Sep  1 14:39 .
drwxrwxr-x 3 pi         pi         4.0K Aug 31 16:46 ..
-rwxrwxrwx 1 pi         pi          53K Jun 13 02:27 test.zim
-rwxrwxrwx 1 iiab-admin iiab-admin  38G Aug  7 07:24 wikipedia_en_all_nopic_2020-08.zim
-rwxrwxrwx 1 iiab-admin iiab-admin 224M Aug  5 13:12 wikipedia_en_astronomy_nopic_2020-08.zim
```

holta, 2020-09-02:
@georgejhunt & @jvonau do you recognize this error?

jvonau, 2020-09-02:
Think the meta data in the zim file contains an error, @tim-moody would be the best person to ask.

jvonau, 2020-09-02:
Things to note:
>=IIAB==========================================================================
-rw-r--r-- 1 root root 56 Jul 12 22:21 /etc/iiab/openvpn_handle
iiab-7.1-200712-ub20-medium-content-ready-g54406367.img

That is a bit confusing, this is not Ubuntu.

>=IIAB==========================================================================
-rw-r--r-- 1 root root 1286 Jul 12 22:21 /.iiab-image
iiab-7.1-200712-ub20-medium-content-ready-g54406367.img
\* master
commit 5440636746c4c0d13230f1691e2dd854d84a3044
Merge: c4df8efd 3870df77
Author: A Holt <holta@users.noreply.github.com>
Date:   Sat Jul 11 14:41:08 2020 -0400
    Merge pull request #2467 from holta/master
    Gitea 1.12.1 -> 1.12.2
commit 3870df77852ed89376a286d682f128d99a58d524
Author: A Holt <holta@users.noreply.github.com>
Date:   Sat Jul 11 14:39:17 2020 -0400
    Gitea 1.12.1 -> 1.12.2
commit c4df8efddcdb9cb7bed1611148166bbc615fa142
Merge: 4804b414 e88fec33
Author: A Holt <holta@users.noreply.github.com>
Date:   Sat Jul 11 13:52:14 2020 -0400
    Merge pull request #2466 from holta/calibre-web-master-branch
    Calibre-Web 0.6.8 -> master
commit e88fec334c678acb4079c2acb5c94eaecbfa7a9a
Author: A Holt <holta@users.noreply.github.com>
Date:   Sat Jul 11 13:48:08 2020 -0400
    Calibre-Web 0.6.8 -> master
commit 10fc1291ff1807a7a413ac1c5d29022d0d0631ad
Merge: 0a108960 4804b414
Author: A Holt <holta@users.noreply.github.com>
Date:   Sat Jul 11 13:46:40 2020 -0400
    Merge pull request #409 from iiab/master
    Sync from iiab:master
Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage2
 =IIAB==========================================================================

It's one of Tim's pre-canned images
>runtime_branch = master
runtime_commit = 17253d4d1477e200480ae8eff95daaf0e49d04ef  
runtime_date = 2020-06-13T13:56:13Z
ansible_version = 2.9.9

https://github.com/iiab/iiab/commit/17253d4d1477e200480ae8eff95daaf0e49d04ef is from June 11, no changes to iiab-make-kiwix-lib3.py in 9 months but kiwix-tools was bumped a couple of months ago but I doubt that is related.

jvonau, 2020-09-02:
>File "/usr/lib/python3/dist-packages/iiab/adm_lib.py", line 97, in get_substitution_data
zim_id = path_to_id_map[path]
KeyError: 'content/wikipedia_en_all_nopic_2020-08.zim'

adm_lib.py is supplied by [iiab-admin-console](https://github.com/iiab/iiab-admin-console) with [no changes](https://github.com/iiab/iiab-admin-console/commits/master/roles/common/templates) since Apr 26

holta, 2020-09-02:
> 17253d4 is from June 11, no changes to iiab-make-kiwix-lib3.py in 9 months but kiwix-tools was bumped a couple of months ago but I doubt that is related.

Several people experienced similar problems in June, so I strongly suspect this is related.

IIAB 7.2 pre-releases (http://download.iiab.io with the newer version of kiwix-tools, that works with newer ZIM files) very likely solves this problem.

jvonau, 2020-09-02:
Links to reported issues?

holta, 2020-09-02:
> Links to reported issues?

@shanti-bhardwa might remember more details, but the summary is here: https://github.com/iiab/iiab/pull/2460

jvonau, 2020-09-03:
So if one were inclined edit your /etc/iiab/local_vars.yml add to the bottom:
>kiwix_version_armhf: kiwix-tools_linux-armhf-3.1.2
kiwix_version_linux64: kiwix-tools_linux-x86_64-3.1.2
kiwix_version_i686: kiwix-tools_linux-i586-3.1.2

`cd /opt/iiab/iiab`
`sudo ./runrole --reinstall kiwix`

jvonau, 2020-09-03:
well 17253d4d1477e200480ae8eff95daaf0e49d04ef should have 2460 applied so the above fix would do nothing but I would check the contents of /opt/iiab/iiab/roles/kiwix/defaults/main.yml to be sure what is installed.

jvonau, 2020-09-03:
I still think there is an issue with zim I'll assume the command runs for a while then fails as the other zim is reported to work.

ganastor, 2020-09-03:
> 
> 
> I would check the contents of /opt/iiab/iiab/roles/kiwix/defaults/main.yml to be sure what was installed before the above fix is tried

From that file it appears that 3.1.2 is installed.

```
iiab-admin@box:~$ cat /opt/iiab/iiab/roles/kiwix/defaults/main.yml
# kiwix_install: True
# kiwix_enabled: True

# kiwix_port: 3000
# iiab_zim_path: "{{ content_base }}/zims"    # /library/zims

# All above are set in: github.com/iiab/iiab/blob/master/vars/default_vars.yml
# If nec, change them by editing /etc/iiab/local_vars.yml prior to installing!

# INSTRUCTIONS TO REINSTALL Kiwix:
# (1) VERIFY THESE VARS IN /etc/iiab/local_vars.yml
#     kiwix_install: True
#     kiwix_enabled: True
# (2) RUN: cd /opt/iiab/iiab; ./runrole --reinstall kiwix

# FYI /library/zims contains 3 important things:
# - library.xml
# - content = directory for all *.zim's
# - index = directory for legacy *.zim.idx's
kiwix_library_xml: "{{ iiab_zim_path }}/library.xml"

# 3 lines below specify which version(s) of kiwix-tools to download from...
# http://download.iiab.io/packages/ ...as originally obtained from...
# http://download.kiwix.org/release/kiwix-tools/ ...or sometimes...
# http://download.kiwix.org/nightly/

kiwix_version_armhf: kiwix-tools_linux-armhf-3.1.2
kiwix_version_linux64: kiwix-tools_linux-x86_64-3.1.2
kiwix_version_i686: kiwix-tools_linux-i586-3.1.2

# kiwix_src_file_i686: "kiwix-linux-i686.tar.bz2"
# v0.9 for i686 published May 2014 ("use it to test legacy ZIM content")
# v0.10 for i686 published Oct 2016 ("experimental") REPLACED IN EARLY 2018,
# thx to @mgautierfr / Matthieu Gautier:
# https://github.com/kiwix/kiwix-build/issues/94
# https://github.com/kiwix/kiwix-tools/issues/170

# Used for Kiwix proxy http://box/kiwix/
kiwix_url: /kiwix
kiwix_url_plus_slash: "{{ kiwix_url }}/"    # /kiwix/
kiwix_path: "{{ iiab_base }}/kiwix"         # /opt/iiab/kiwix

kiwix_nginx_timeout: 600
kiwix_threads: 4
```

holta, 2020-09-03:
> From that file it appears that 3.1.2 is installed.

You can verify this by running:

```
cd /opt/iiab/kiwix/bin/
./kiwix-serve --version
```

ganastor, 2020-09-03:
> You can verify this by running:

Thanks, I was just about to start looking for that. 3.1.1 is installed.

Should I grab kiwix-tools_linux-armhf-3.1.2.tar.gz and get it updated and retest?

```
iiab-admin@box:/opt/iiab/kiwix/bin$ ./kiwix-serve --version
3.1.1
```

holta, 2020-09-03:
> Should I grab kiwix-tools_linux-armhf-3.1.2.tar.gz and get it updated and retest?

Yes!  Follow @jvonau's instructions @ https://github.com/iiab/iiab/issues/2502#issuecomment-686182095

ganastor, 2020-09-03:
> Yes! Follow @jvonau's instructions @ [#2502 (comment)](https://github.com/iiab/iiab/issues/2502#issuecomment-686182095)

Thanks, and sorry, I completely missed that comment. I tried and it failed:

```
failed: [127.0.0.1] (item=mediawiki) => {"ansible_loop_var": "item", "assertion": "mediawiki_install or not mediawiki_enabled", "changed": false, "evaluated_to": false, "item": "mediawiki", "msg": "DISALLOWED: 'mediawiki_install: False' WITH 'mediawiki_enabled: True' -- IIAB DOES NOT SUPPORT UNINSTALLS -- please verify those 2 variable values e.g. in /etc/iiab/local_vars.yml, and other places variables are defined?"}
```

I've a good hunch of where to go from here, but I'll let you experts advise me before I start hacking away and breaking things. I was just going to replace the executables in /opt/iiab/kiwix/bin, will that not work? Otherwise I assume we need to do some other temporary edits in /etc/iiab/local_vars.yml

ganastor, 2020-09-03:
Downloaded kiwix-tools_linux-armhf-3.1.2.tar.gz, replaced with 3.1.1 files in /opt/iiab/kiwix/bin. #2460 is correct in that 3.1.2 is needed for the large Wikipedia zim files.

closed 2020-09-03 by ganastor

holta, 2020-09-03:
Thank you @ganastor for persisting.

I'd also encourage you to look very closely into IIAB 7.2 pre-releases as well, which offers a lot more, and can generally be installed in less than an hour using http://download.iiab.io

Here are the Linux OS's you can choose from: https://github.com/iiab/iiab/wiki/IIAB-Platforms

Keep in touch!

ganastor, 2020-09-03:
Thanks @holta I'll try and keep in touch, I'll definitely be following this project.

I will be doing the latest versions, I just used the image for brevity's sake so I could check out the project before diving in all the way. I fully intend on incorporating packet radio/SDR for transmitting capabilities with no internet as well, so hopefully I can come back and share at some point.

holta, 2020-09-03:
> I fully intend on incorporating packet radio/SDR for transmitting capabilities with no internet as well

We can put you in touch with @m-anish in India (and others in South Africa etc) doing similar work with Internet-in-a-Box, as nec while you make progress here?

holta, 2020-09-03:
> I'm thinking either a laptop or RPi4 with more ram

Raspberry Pi 4 with 2GB RAM is our general recommendation!

OS choices are up to you (:

jvonau, 2020-09-11:
> Thanks, and sorry, I completely missed that comment. I tried and it failed:
>
> ```
> failed: [127.0.0.1] (item=mediawiki) => {"ansible_loop_var": "item", "assertion": "mediawiki_install or not [...]

The failure is pretty clear "DISALLOWED: 'mediawiki_install: False' WITH 'mediawiki_enabled: True' " in local_vars, maybe the message could include the full path to local_vars.yml? The local edit to enable kiwix should of also set kiwix_install to 'True'