iiab/iiab #2517: Medical IIAB does not boot if wifi_hotspot_capacity_rpi_fix: True

tim-moody, 2020-09-13:
latest raspios with latest master on rpi w.
hangs at line random: crng init done
I saw this before on rpi 4, but the install was not clean in that wifi_hotspot_capacity_rpi_fix: True was added to local vars after initial install.
Also, before this occurred on an image newly written to disk, which had not yet resized.
This image was already resized before moving to W.

tim-moody, 2020-09-13:
I moved the card back to rpi 4 and now it hangs after line:
hub 1-1:1.0: 4 ports detected, which is where it had stopped before.

tim-moody, 2020-09-13:
mounted the card on a nuc and removed wifi_hotspot_capacity_rpi_fix line from local vars
then cp brcmfmac43455-sdio.bin.orig brcmfmac43455-sdio.bin
attempted to boot in rpi 4:
hangs at the same spot ?!

tim-moody, 2020-09-13:
https://www.raspberrypi.org/forums/viewtopic.php?t=209093

holta, 2020-09-14:
@tim-moody out of curiosity are you running RaspiOS kernel #1333 [*] ?

Consider experimenting with a better / more recent kernel if so, e.g. by running `rpi-update` ?

[*] As seen by the output of 'uname -a' ...and just to clarify: kernel #1333 has Nothing To Do with ticket #1333 !  Background: @jvonau uncovered problems with kernel #1333 on ticket #2497

jvonau, 2020-09-14:
Seems more like a corrupt image than anything related to an option that was never present on first boot "I saw this before on rpi 4, but the install was not clean..."

I'd also make sure `blkid`, /etc/fstab, and /boot/cmdline.txt all have the correct UUIDs. Anything to do with 2465?

Funny, the above url mentions havegd  I had that in one of my previous PR from January 3 #2381 3d1df1241e430e112ed3454b1d072ec7661764aa & 0248cdaf15d2be7b5a795bb61290c92a1c78a988 but that was just glossed over by the power brokers.

tim-moody, 2020-09-14:
So I think I now understand this a much as I care to, 

>Seems more like a corrupt image ("I saw this before on rpi 4, but the install was not clean..." referred to the fact that when I wrote that I thought it was improper to add wifi_hotspot_capacity_rpi_fix: True to local vars after the ansible run as a fix, but in fact that is an acceptable way of configuring it.)

Yes, but how did it become corrupt? There is indeed a bug in growpart https://bugs.launchpad.net/cloud-utils/+bug/1881014 where it fails to parse the kernel version from uname -r. Surprisingly, and part of what fooled me, is that this happens on W but not on 4, even though they are the same kernel (because the output of uname -r is different).

If the boot on which resizing takes place is the same as the wifi_hotspot_capacity_rpi_fix: True first boot, I think you can get corruption on the W perhaps because the file system is not big enough, but I'm not certain.

In any event if you resize on a 4 and then set wifi_hotspot_capacity_rpi_fix: True in local vars and do the requisite reboots everything seems to work fine, even after you move the card to a W, This is no real burden as we have already warned in the past that resizing is very slow on the W and until the upstream bug is fixed it doesn't even work.

jvonau, 2020-09-15:
Have a look at #2522, think there is a timeout issue on the W's and it might be better to resize before the system mounts the partition read-write?

jvonau, 2020-09-15:
>pi@box:/opt/iiab/iiab $ sudo apt info cloud-guest-utils
Package: cloud-guest-utils
Version: 0.29-1

I see, the fix is in [cloud-utils 0.31-22](https://github.com/canonical/cloud-utils/pull/11), [Ubuntu released it](https://launchpad.net/ubuntu/groovy/armhf/cloud-guest-utils/0.31-22-g37d4e32a-0ubuntu1), could try the [ubuntu .deb](http://launchpadlibrarian.net/489493335/cloud-guest-utils_0.31-22-g37d4e32a-0ubuntu1_all.deb)

tim-moody, 2020-09-15:
yes. I also found the link. the question is whether to use the deb or just wait for it to get included. As far as I know there are not many people who try images directly on W, but I don't object if there is an easy way to install it with ansible. It would also allow for timing tests on W.

jvonau, 2020-09-15:
[The patch](https://github.com/canonical/cloud-utils/pull/11/commits/c764723c49d6c447a8b3670f1c553a9f9fe46b73) looks pretty straight forward, perhaps just patch growpart after installation?

jvonau, 2020-09-15:
Ported the patch to 29.1 that comes with RaspOS [growpart.diff.txt](https://github.com/iiab/iiab/files/5228028/growpart.diff.txt)
download then rename growpart.diff.txt to growpart.diff
`sudo patch -u -i growpart.diff`
>can't find file to patch at input line 3
Perhaps you should have used the -p or --strip option?
The text leading up to this was:
-/usr/bin/growpart.org	2020-09-15 20:47:57.971618614 +0100
+/usr/bin/growpart	2020-09-15 20:53:05.582260594 +0100
File to patch:

It will prompt you for the path to the file
/usr/bin/growpart [enter]
then you will see
> patching file /usr/bin/growpart

tim-moody, 2020-09-15:
what about
https://github.com/canonical/cloud-utils/commit/7a442f68cc4fe0e777cd10caecd22231e80ecfd2.patch
https://docs.ansible.com/ansible/latest/modules/patch_module.html

jvonau, 2020-09-16:
> what about
> https://github.com/canonical/cloud-utils/commit/7a442f68cc4fe0e777cd10caecd22231e80ecfd2.patch

Yes that is the source but does not apply cleanly(note the line numbers between the two) as is, but is the source of the above attached patch

> https://docs.ansible.com/ansible/latest/modules/patch_module.html

Don't think I'm going to pursue automation until somebody tries the above patch by hand and reports success or failure.

jvonau, 2020-09-16:
The other alternative would be to change /usr/sbin/iiab-rpi-max-rootfs.sh back to using parted after writing the image to a sdcard. You can then re-roll the image for distribution with the fix included if you want. No matter how you look at it there will be some fiddling with files prior to booting the image on a W, or just boot the card on something that is not a W.

holta, 2020-09-16:
A warning has been added to the Release Notes of the imminent IIAB 7.2:
https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes#known-issues

For now it reads:
> The 1st boot of an IIAB microSD card should NOT occur in a Raspberry Pi Zero W, as this can freeze or worse corrupt the microSD card, failing to expand (resize) the main partition. If you do need to resize: make sure the 1st boot of your IIAB microSD card happens in a Raspberry Pi 3, 3 B+ or 4. #2517

tim-moody, 2020-09-16:
>Don't think I'm going to pursue automation until somebody tries the above patch by hand and reports success or failure.

success. and was done on W before I could login

I wrote iiab-PRE-7.2-200914-medical-content-ready-g01d418e.img to 128G sdcare
manually modified growpart
booted in W

Did a second test by running 
./min-sd test /dev/sdb2
against the same sdcard mounted on a nuc
booted in W
this time was able to login and watch resizing with df

tim-moody, 2020-09-16:
I don't think a big patch effort is warranted since we expect the fix to roll out eventually. How dangerous would it be just to copy the patch growpart when building an image? Might miss a future, unrelated fix.

tim-moody, 2020-09-16:
I will patch growpart on any images I produce manually until such time as the fix is rolled out. Others targetting rpi W will have to do the same.

closed 2020-09-16 by tim-moody

jvonau, 2020-09-21:
Can somebody edit the title to this issue as it has been determined to be regression brought on by #2465 and not anything to do with wifi_hotspot_capacity_rpi_fix at all.

jvonau, 2020-09-27:
No (simply closing was not enough), it is about the accuracy of the bug description, origin, resolution and simply not pointing the finger at an unrelated change given there are now hidden comments connected to this thread.