iiab/iiab #2525: validate vars errors non viewable?

jvonau, 2020-09-15:
### Steps to reproduce the problem:
Use invalid combinations in local_vars.yml, when the error hits ref: #2502
The error message goes flying by, have to scroll back to view, and is not clearly visible at the end of the logfile. 
I do all of my installs over ssh, so no issue for me, but scrolling back at a keyboard with a stage2 install while "iiab" was typed after a bad edit to local_vars.yml can be problematic.

jvonau, 2021-04-14:
Here is one example using admin-console, note one cannot scroll the feedback. 
[image] induced this by selecting (Squid Cache) in
[image]
Being able to "enable" something that is not "installed" is a bug. Should I file an issue for this if so where here or iiab-admin?

tim-moody, 2021-04-14:
>Being able to "enable" something that is not "installed" is a bug. Should I file an issue for this if so where here or iiab-admin?

was install True? I admit to not checking the installed state, but if install is false a service should not appear, but not sure about some of these old ones.

tim-moody, 2021-04-14:
>Here is one example using admin-console, note one cannot scroll the feedback.

are you saying the scroll bar under results does not work?

jvonau, 2021-04-14:
> > Here is one example using admin-console, note one cannot scroll the feedback.
> 
> are you saying the scroll bar under results does not work?

Gives very little past history, same for completed jobs

tim-moody, 2021-04-14:
I'll have to check. I vaguely remember truncating it to improve performance, but I thought that was only before the job terminates.

tim-moody, 2021-04-14:
Really needs reworking to make it expandable on demand rather than pulling a huge history every time.

jvonau, 2021-04-14:
> > Being able to "enable" something that is not "installed" is a bug. Should I file an issue for this if so where here or iiab-admin?
> 
> was install True? I admit to not checking the installed state, but if install is false a service should not appear, but not sure about some of these old ones.
>
Install was False
 
> was the inconsistency edited manually or through Admin Console?

Editing strictly by the admin-console only.

I have a mockup to expose the *_install variables for this pane
[image]

Diff https://github.com/iiab/iiab-admin-console/compare/master...jvonau:squid_proto
result of change
[image]

tim-moody, 2021-04-14:
the hide if install false logic was likely only applied to the Services Enabled panel. file a bug on Adm Cons

jvonau, 2021-04-14:
Is your intention to only offer the ability turn off|on installed services and never offer the ability to install services that are not currently installed?

tim-moody, 2021-04-14:
That has been the situation up until now, but I have been reconsidering. The install presets functionality could benefit from this as you might select a present that requires a service that you didn't install. The project for that is https://github.com/iiab/iiab-admin-console/projects/6

jvonau, 2021-04-14:
From the POV of ansible and the validation, as long as both _install and _enable are True the new service will get installed (internet is needed of course), I just exposed both boxes in the above demo. The form just needs to set both variables when the tickbox is checked, and just toggle _enable when unchecked. The service_id list based on * _service is being suppressed, to hide services that are no longer supported I presume. https://github.com/iiab/iiab-admin-console/compare/master...jvonau:dead_wood

jvonau, 2021-04-14:
The code to bail on a new install without internet was PR #2505

holta, 2021-04-16:
FYI @tim-moody's PR https://github.com/iiab/iiab-admin-console/pull/392 appears to have solved some of the above issue.

jvonau, 2021-04-16:
That solves(?) the secondary issue, the primary issue, reading an error at a cli console (think keyboard and monitor, only what is on the screen with no way to scroll back) remains and now also appearing in admin-console.

holta, 2021-04-17:
> What is interesting is that squid was finally disabled in all the installs in Jan 2019, after all this is an off-line project, as squid's iptables redirect rule was interfering with captive-portal.

Great to know.

@georgejhunt was this on purpose?

Either way, that's critical info we need to clearly state in http://FAQ.IIAB.IO and https://github.com/iiab/iiab/wiki/IIAB-Networking for implementers who are always asking for cleaner/simpler choices during the deployment process &mdash; which is always happening under intense time pressure &mdash; for better or for worse.

jvonau, 2021-04-17:
> > What is interesting is that squid was finally disabled in all the installs in Jan 2019, after all this is an off-line project, as squid's iptables redirect rule was interfering with captive-portal.
> 
> Great to know.
> 
> @georgejhunt was this on purpose?
>

The github search function works very well for this kind of stuff https://github.com/iiab/iiab/search?o=asc&q=squid&s=created&type=issues, you should know, you pushed the final change in https://github.com/iiab/iiab/pull/1369 because of https://github.com/iiab/iiab/issues/1323

Other refs: https://github.com/iiab/iiab/pull/152 https://github.com/iiab/iiab/pull/242 https://github.com/iiab/iiab/pull/278 https://github.com/iiab/iiab/issues/512 https://github.com/iiab/iiab/issues/623 https://github.com/iiab/iiab/issues/822 https://github.com/iiab/iiab/pull/1374

holta, 2021-06-18:
@tim-moody @jvonau to summarize the situation quickly, is there outstanding work required here?

tim-moody, 2021-06-18:
Currently the output of jobs is truncated (only the tail) to cut down on the traffic from the server. There could be an enhancement to change this.
Currently services can only be enabled, not marked for install. (Except using presets, which can do both). There could be an enhancement such that checking a service would imply both install and enable.

jvonau, 2021-12-11:
Scroll-back not available to show error
[image]

Selection that is the cause of the error, then "Save Configuration" "**Install** Configured Options"
[image]

tim-moody, 2021-12-11:
https://github.com/iiab/iiab-admin-console/issues/393 is the ticket.

tim-moody, 2021-12-11:
This issue mostly comes up for ansible runs, so I hate to have the whole log of downloads in order to get the ansible log. One idea would be to add Display Ansible Log to Utilities. Could be used even if ansible was not run with Adm Cons

jvonau, 2022-04-05:
#2739

tim-moody, 2023-06-16:
https://github.com/iiab/iiab-admin-console/pull/545 may help with this as it now does not truncate log for selected commands, including ansible.

jvonau, 2025-01-27:
I'll take it as fixed.

closed 2025-01-27 by jvonau