iiab/iiab #2526: Lokole delays IIAB shutdown: "A stop job is running for Celery service (1min 34s / 1min 34s)"

holta, 2020-09-15:
### Expected behavior and actual behavior:

Teachers are prevented from shutting down their IIAB servers efficiently due to this 90+ second delay, which they do many times per day in countries where electricity is lacking and expensive (sometimes at the end of every lesson!)  So IIAB's shutdown needs to be reliable and rapid/clean (an IIAB that is working well almost always powers off within about 10-20 seconds).

This Lokole (Celery service) failure to shut down gracefully seems to happen almost every time one tries to shut down IIAB (about 80% or 90% of the time).

Whereas very occasionally, IIAB seems to shut down properly/rapidly within about 10-to-20 seconds (as teachers/schools expect and need it to do).  There's a theory (highly unconfirmed, I'm not sure I believe it) that the Celery service (and hence IIAB) shut down more cleanly if it hasn't been on more many minutes.

### Steps to reproduce the problem:

Install IIAB (http://download.iiab.io) making sure `lokole_install: True` and `lokole_enabled: True` are both set in [/etc/iiab/local_vars.yml](http://wiki.laptop.org/go/IIAB/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it.3F)

I chose option 3. (BIG-sized) during IIAB's install process, to make this very easy.

Once IIAB is installed, try to shut it down by running `sudo poweroff` at the command-line.

### IIAB version, OS version, hardware details:

Ubuntu Server 20.04.1 VM (VirtualBox) with 2GB RAM.

FYI the IIAB VM is not constrained by RAM: it uses about ~1.2 GB RAM in normal circumstances.

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

/usr/bin/iiab-diagnostics output is here: http://sprunge.us/U8mFAB?en

Thank you @adamsclafani @c-w @jvonau if you have any ideas to bring this under control?

If you need direct ssh access to this VM (u20041-srv-vm-BIG-ans210 = 10.8.0.66) let me know!

Related: PRs #2507 #2513 #2514

jvonau, 2020-09-15:
Post install reboot is fine or does it hang there? When rebooted does a restart of the service work correctly?

holta, 2020-09-15:
> Post install reboot is fine or does it hang there?

Boot is rapid, every time.

> When rebooted does a restart of the service work correctly?

Yes, the Celery service starts reliably every time on boot.  Example:

```
root@box:~# systemctl status celery
● celery.service - Celery service
     Loaded: loaded (/etc/systemd/system/celery.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2020-09-15 12:44:29 EDT; 3h 54min ago
   Main PID: 700 (celery)
      Tasks: 7 (limit: 2267)
     Memory: 87.1M
     CGroup: /system.slice/celery.service
             ├─ 700 /library/lokole/venv/bin/python3 /library/lokole/venv/bin/celery --app=opwen_email_client.webapp.tasks worker --loglevel=info --concurrency=2
             ├─1612 /library/lokole/venv/bin/python3 /library/lokole/venv/bin/celery --app=opwen_email_client.webapp.tasks worker --loglevel=info --concurrency=2
             └─1615 /library/lokole/venv/bin/python3 /library/lokole/venv/bin/celery --app=opwen_email_client.webapp.tasks worker --loglevel=info --concurrency=2

Sep 15 12:44:29 box.lan systemd[1]: Started Celery service.
Sep 15 12:44:54 box.lan celery[700]: /library/lokole/venv/lib/python3.8/site-packages/celery/platforms.py:800: RuntimeWarning: You're running the worker with superuser privileges: this is
Sep 15 12:44:54 box.lan celery[700]: absolutely not recommended!
Sep 15 12:44:54 box.lan celery[700]: Please specify a different user using the --uid option.
Sep 15 12:44:54 box.lan celery[700]: User information: uid=0 euid=0 gid=0 egid=0
Sep 15 12:44:54 box.lan celery[700]:   warnings.warn(RuntimeWarning(ROOT_DISCOURAGED.format(
Sep 15 12:44:54 box.lan celery[700]: [2020-09-15 12:44:54,126: INFO/MainProcess] Connected to sqlalchemy+sqlite:////tmp/celery.sqlite3
Sep 15 12:44:55 box.lan celery[700]: [2020-09-15 12:44:55,223: INFO/MainProcess] celery@box.lan ready.
```

jvonau, 2020-09-15:
now restart the service, snappy or takes a long time?

c-w, 2020-09-15:
@adamsclafani @aidan-fitz

holta, 2020-09-15:
> now restart the service, snappy or takes a long time?

`systemctl restart celery` takes 90+ seconds to return to the prompt...almost every time...just like during system shutdown...something is definitely wrong (:

jvonau, 2020-09-15:
RaspOS appears unaffected, python3.8 related maybe?

holta, 2020-09-15:
If anyone needs to ssh into to this 10.8.0.38 (formerly 10.8.0.66) in order to diagnose, let me know.

adamsclafani, 2020-09-16:
I'm currently trying to set up a vm with the dev environment but my ansible setup keeps failing... 

On another note, @holta I can ssh into your vm maybe so I can mess around with this?

adamsclafani, 2020-09-16:
I was thinking that maybe a custom shutdown script for celery might do the trick? I'm totally guessing though

holta, 2020-09-16:
> I'm currently trying to set up a vm with the dev environment but the ansible setup script keeps failing on me 🙃
> It was working fine earlier when I was working on adding the new services...

Are you running the latest https://github.com/iiab/iiab/blob/master/scripts/ansible (/opt/iiab/iiab/scripts/Ansible as used by http://download.iiab.io) to install Ansible 2.10.1 ?

Where is it failing exactly?

> On another note, @holta I can ssh into your vm maybe so I can mess around with this?

Yes!  Please email or text me and I'll get you in.

holta, 2020-09-17:
@adamsclafani didn't hear from you yesterday.  (Please text me or email me so I can provide you ssh access if possible!)

adamsclafani, 2020-09-17:
@holta Just sent a message your way.

On another note, this is the error I keep running into. I am running the latest script in the repo. It only started happening after I pulled updates to keep my repo up to date. I cloned a new separate dev environment and it happens there too. Here is the output I get:

[image]

holta, 2020-09-18:
> @holta Just sent a message your way.

Hopefully you now have ssh access to the VM.  But let me know if not.

> On another note, this is the error I keep running into. I am running the latest script in the repo. It only started happening after I pulled updates to keep my repo up to date. I cloned a new separate dev environment and it happens there too.

@nzola & @jvonau experienced a similar-but-different failure here: #2532 

Your error (when trying to install Ansible Collections) was:

```
The 'packaging' distribution was not found and is required by ansible-base.
```

@adamsclafani I notice `stretch` in your above screenshot.  Note that we no longer support Debian 9 (Stretch).  Could you consider running Debian 10 (Buster) or Ubuntu 20.04.1 or Linux Mint 20?

Supported OS's are listed here: https://github.com/iiab/iiab/wiki/IIAB-Platforms#operating-systems

If nothing works, can you please run `sudo iiab-diagnostics` and share the resulting URL here?  (Thanks!)

adamsclafani, 2020-09-18:
Ok interesting! I'll go ahead and try one of the suggested OS's

adamsclafani, 2020-09-19:
@holta @jvonau After messing with things a little bit on the VM, the systemctl restart celery took a long time for the first run, but subsequent runs are snappy. I then tried systemctl stop celery, the followed up with the start command. The first run of the stop command took a long time as well, but re-runs are quick. All startups are snappy.

adamsclafani, 2020-09-19:
Also using buster did the trick! Ansible install went smoothly

jvonau, 2020-09-19:
> @holta @jvonau After messing with things a little bit on the VM, the systemctl restart celery took a long time for the first run, but subsequent runs are snappy. I then tried systemctl stop celery, the followed up with the start command. The first run of the stop command took a long time as well, but re-runs of stop in the stop/start sequence are quick. All startups are snappy.

> > now restart the service, snappy or takes a long time?
> 
> `systemctl restart celery` takes 90+ seconds to return to the prompt...almost every time...just like during system shutdown...something is definitely wrong (:

The 90 seconds is the stock systemd timeout, the service is forcibly being killed at that point. Remember the issue reported against Ubuntu-20. I had no issues, 5-10 sec restarts for something I presume to be attached to a database I would consider normal, on RaspOS based on buster. Like I said above I suspect something python3.8 related or maybe systemd on Ubuntu.

holta, 2020-09-19:
@adamsclafani we're closer:

1) celery.service (systemd service) failed to stop 4 times when I rebooted the VM about 40 times.  In short, the 90-second delays happen less often, but are still happening.

   What did you change on this VM to bring about the improvement?

   Got ideas towards doing even better?!

2) If you continue to ssh to the VM via NGROK, the port has now changed to: **14227**

3) ASIDE: I changed VirtualBox > (actual VM) > Properties > System > Processor from 1 CPU to 2 CPU's so that you can reboot it remotely now, since almost ~50% of bootups had been freezing completely.  Based on these suggestions:
   - https://www.maketecheasier.com/fix-ubuntu-freezing-virtualbox/
   - https://askubuntu.com/questions/1030181/ubuntu-18-04-lts-freezes-while-in-boot-in-wndows-virtual-box-solution

   CAVEAT: This Ubuntu Server 20.04.1 VM still very occasionally fails to boot due to cloud-init, udev and/or swap.img and associated inode problems (despite the underlying disk being an SSD!)  So I set `tune2fs -c 10 /dev/mapper/ubuntu--vg-ubuntu--lv` (mountpoint /) and `tune2fs -c 20 /dev/sda2` (mountpoint /boot) to force fsck (i.e. a disk check) to run every 10-or-20 boots, which seems to have solved the problem.

jvonau, 2020-09-20:
To isolate the reproducer to VMs I suggest a Ubuntu-20 server install on Rpi hardware and note the restart times as we know RaspOS is not affected on that hardware platform. That would rule out python3.8 as an issue.

>  3\. Just FYI the filesystem errors...that had been occurring about once a minute...appeared as follows:
> > ext4-fs error (device dm-0) ext4_lookup inode [one of ~6 numbers, that ended up corrupted] comm [various program/service names] iget: checksum invalid

Maybe the underling disk of the parent OS is starting to go away? I'd check the filesystem there too, that is where the image lives just to make sure.

holta, 2020-09-20:
`systemctl restart celery` sometimes took 90+ seconds on both of these Raspberry Pi 4 OS's, showing that the problem arises on Python 2.7 and 2.8 both:

1) Celery slowness occurred 3 times out of 10 tests on Python 3.8.5 & systemd 246 on [Ubuntu 20.10 Groovy Gorilla](http://cdimage.ubuntu.com/ubuntu-server/daily-preinstalled/current/) @ rpi4-2g-u2010-0919-BIG = 10.8.0.62

   FYI this BIG-sized IIAB (http://download.iiab.io) was installed onto the 2020-09-19 pre-release of the "Hard-Float" version (not the 64-bit ARM version) of the OS.

   iiab-diagnostics output: http://sprunge.us/PE2Kfh?en

2) Celery slowness occurred 1 time out of 10 tests on Python 2.7.16 & systemd 241 on [64-bit RaspiOS 2020-08-24](https://downloads.raspberrypi.org/raspios_lite_arm64/images/) @ 183-rpi4-2g-lite-64bit-BIG-2487 = 10.8.0.54

   iiab-diagnostics output: http://sprunge.us/fQJIIF?en

@adamsclafani: let me know if you need help to ssh into either above machine.

adamsclafani, 2020-09-21:
>    What did you change on this VM to bring about the improvement?
Didn't change anything. Just ssh'ed in, ran some commands and recorded what I saw.

@holta @jvonau, this is getting more interesting by the moment. On another note, after scanning thorough some [celery docs](https://docs.celeryproject.org/en/stable/userguide/workers.html#restarting-the-worker), I noticed that it says utilizing the HUP signal isn't the recommended form of restarting, but instead to use the TERM signal, then start a new worker. I don't think this is the issue though because the [restart command](which uses HUP) is showing the same behavior (mostly quick execution, some slow) as the [stop command](https://github.com/iiab/iiab/blob/819bf92b74e96dfe8cfdb121952f95ed4e0ad99e/roles/lokole/templates/celery.service.j2#L11) (which uses TERM)

holta, 2020-09-21:
> HUP signal isn't the recommended form of restarting, but instead to use the TERM signal, then start a new worker. I don't think this is the issue though because the restart command (which uses HUP) [...]

@adamsclafani I would agree as `systemctl stop celery` (TERM) took 90+ seconds to complete 3 times out of 10, when I just now tested this on the VM (Ubuntu Server 20.04.1) you have access to (NGROK port 14227).

It would seem...there is little-or-no difference from `systemctl restart celery` (HUP).

jvonau, 2020-09-29:
> `systemctl restart celery` sometimes took 90+ seconds on both of these Raspberry Pi 4 OS's, showing that the problem arises on Python 2.7 and 2.8 both:
> 
>     1. Celery slowness occurred 3 times out of 10 tests on Python 3.8.5 & systemd 246 on [Ubuntu 20.10 Groovy Gorilla](http://cdimage.ubuntu.com/ubuntu-server/daily-preinstalled/current/) @ 182-rpi4-2g-u2010-0919-BIG = 10.8.0.62
>        FYI this BIG-sized IIAB (http://download.iiab.io) was installed onto the 2020-09-19 pre-release of the "Hard-Float" version (not the "64-bit ARM" version) of the OS.
>        iiab-diagnostics output: http://sprunge.us/PE2Kfh?en
> 
>     2. Celery slowness occurred 1 time out of 10 tests on Python 2.7.16 & systemd 241 on [64-bit RaspiOS Lite 2020-08-24](https://downloads.raspberrypi.org/raspios_lite_arm64/images/) @ 183-rpi4-2g-lite-64bit-BIG-2487 = 10.8.0.54
>        iiab-diagnostics output: http://sprunge.us/fQJIIF?en
> 
> 
> @adamsclafani: let me know if you need help to ssh into either above machine.

The reported python versions are inaccurate, Lokole is running in a [python3 virtual environment,](https://github.com/iiab/iiab/blob/master/roles/lokole/tasks/install.yml#L43-L65) the version on Ubuntu used 3.8.5 while RaspiOS used 3.7.3 when the virtual environment was created.

holta, 2021-05-18:
Quite likely this is solved @jvonau's PR #2782 just merged?

@deldesir: keep an eye on it, if you're able to look at IIAB shutdown times in coming weeks!

holta, 2021-05-27:
@jvonau recommends closing this as the new roles/lokole is running a completely different process manager.

closed 2021-05-27 by holta