iiab/iiab #2528: Unable to retrieve content moved to a USB drive in Admin Console

mihalkaengineering, 2020-09-16:
###Expected behavior:
Followed instructions on IIAB/FAQ - Item 43.  Instructions for saving content to a permanently mounted USB Drive.  Copied content (Zim files to be exact) from internal storage to attached HDD.  Later I remove duplicate Zim files from internal storage as the whole point of having content on an attached HDD was to free up internal storage.  I created the necessary symbolic link to external drive (at lease I believe I did this correctly).  I would have expected the Zim files on the external drive to remain present under "Manage Content" within the "Admin Console".

###Actual behavior:
None of the content moved to the external drive is present in the "Admin Console" nor is it accessible via the web interface.

### Steps to reproduce the problem:
Follow the instructions on IIAB/FAQ.  Move a Zim file from internal storage to external storage.  "Refresh Display" under "Manage Content" within the Admin Console. 

### IIAB version, OS version, hardware details:
IIAB Version 7.1, RPi OS Version 4.19, RPI3+

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)
Received the following when running iiab-diagnositcs:
Unable to read or parse the result page, it could be a server timeout or a change server side, try with another pastebin.

Thanks Mike

holta, 2020-09-16:
> Received the following when running iiab-diagnositcs:
Unable to read or parse the result page, it could be a server timeout or a change server side, try with another pastebin.

Hi Mike!

Can you run this:

```
sudo su -    # i.e. become root
cd /usr/bin
mv iiab-diagnostics iiab-diagnostics.old
wget https://raw.githubusercontent.com/iiab/iiab/master/scripts/iiab-diagnostics
chmod a+x iiab-diagnostics
iiab-diagnostics
```

Thanks for replying with the resulting URL!

mihalkaengineering, 2020-09-16:
Here you go.  Thanks!

http://sprunge.us/soBvh6

holta, 2020-09-16:
http://sprunge.us/soBvh6?en is extremely helpful, thanks @mihalkaengineering 

I'm hoping @tim-moody can review this and respond to you later today.

Asides:
- It looks like you upgraded kiwix-tools to 3.1.2 which was a good move!
- FYI [IIAB 7.2](https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes) is in near-final release at the moment, installable using http://download.iiab.io if you need it.

tim-moody, 2020-09-16:
http://wiki.laptop.org/go/IIAB/FAQ#Can_I_permanently_attach_an_external_USB_drive.2C_to_add_more_content.3F

is a bit ambiguous, but does not explicitly mention zims.

In fact the key line is  ln -s /library-ext/www/html/modules/ /library/www/html/modules.

zims do not live in modules, they live in /library/zims/content and that is where admin console looks for them, so you would need a link for that as well or map the entirety of /library-ext to /library, but then you have to copy everything there.

mihalkaengineering, 2020-09-16:
> http://sprunge.us/soBvh6?en is extremely helpful, thanks @mihalkaengineering
>
> I'm hoping @tim-moody can review this and respond to you later today.
>
> Asides:
>
> * It looks like you upgraded kiwix-tools [...]

Holta,

Thanks for the reply.  I had to update kiwix-tools because I kept getting a KeyError notification when I would try to add certain zim files.  Specifically Wikipedia content. Updating to 3.1.2 resolved this issue.  And  If 7.2 is nearing final release then maybe I will install it and see if my problem is resolved.

-Mike

> http://sprunge.us/soBvh6?en is extremely helpful, thanks @mihalkaengineering
>
> I'm hoping @tim-moody can review this and respond to you later today.
>
> Asides:
>
> * It looks like you upgraded kiwix-tools [...]

Holta,

Thanks for the reply.  I had to update kiwix-tools because I kept getting a KeyError when I would try to add certain zim files, specifically the Wikipedia content.  Updating to 3.1.2 was the only way to resolve this.  I am mentioning this here incase anyone else runs into a similar issue.  The solution was presented by @jvonau in the comment thread for issue #2502

If 7.2 is nearing final release then maybe it would be worthwhile to update to that now.  Do you think there is a possibility it would resolve my issue?

-Mike

holta, 2020-09-16:
> If 7.2 is nearing final release then maybe it would be worthwhile to update to that now. Do you think there is a possibility it would resolve my issue?

There are many reasons to install a fresh copy of IIAB 7.2!  So do please consider that.  _But First: was @tim-moody comment below helpful to you?_

> In fact the key line is `ln -s /library-ext/www/html/modules/ /library/www/html/modules`
>
> zims do not live in modules, they live in `/library/zims/content` and that is where admin console looks for [...]

mihalkaengineering, 2020-09-16:
> http://wiki.laptop.org/go/IIAB/FAQ#Can_I_permanently_attach_an_external_USB_drive.2C_to_add_more_content.3F
>
> is a bit ambiguous, but does not explicitly mention zims.
>
> In fact the key line is ln -s [...]

Tim,

Thanks so much for taking a look at my issue.  I know you mentioned the ambiguity of the instructions, which at first caught me out, but as I read and learned along the way I was able to understand how this needed to be done.

Hopefully I can add a little context.  Here is where I currently am.
-At the root I have /library/zims/content/  (this shows a capacity of 14.3GiB)
    -In this folder I have two files: test.zim and I file called "content" (which has a file type listed as a symbolic link but it lacks an actual extension)
-Also at the root I have /library-ext/zims/content/  (this shows a capacity of 867GiB)
    -In this folder I have placed one file as a test case: wikipedia_en_simple_all_maxi_2020-09.zim

A few notes:
* I created the Symbolic Link by running the following command
sudo ln -s /library-ext/zims/content/ /library/zims/content/
* I mention the files sizes because through this process it was the only point of reference which made me feel like I was doing this whole process correctly.
*The symbolic link file has the "Target File:" listed as "/library-ext/zims/content/"

Once I got the folders setup as described above my expectation was that the zim files located on the external drive would show up as available content to manage within the Admin Console (Install Content> Manage Content).  Instead the only content is "Test.zim"

Is there any nuance to how the user must get the content onto the external drive in the first place?  For example, is it required that all content added to the external drive must first be downloaded to the internal storage and then moved or copied over to the external drive.  Or should it be that I can copy zim files from say a thumb drive to the external and the content will be "discovered" within the admin console?

Thanks in advance for your input.
-Mike

mihalkaengineering, 2020-09-16:
> > If 7.2 is nearing final release then maybe it would be worthwhile to update to that now. Do you think there is a possibility it would resolve my issue?
> 
> There are many reasons to install a fresh copy of IIAB 7.2 rapidly approaching final release. So do please consider that... _But First: was @tim-moody comment below helpful to you?_
> 
> > In fact the key line is `ln -s /library-ext/www/html/modules/ /library/www/html/modules`
> > zims do not live in modules, they live in `/library/zims/content` and that is where admin console looks for them, so you would need a link for that as well or map the entirety of `/library-ext` to `/library`, but then you have to copy everything there.

Hello Hota,

Yes, his reply was helpful in that it confirmed my thought process and reassured me that I was dong it correctly but maybe just missed a small detail.  Please see my reply.

That being said as I write this reply something occurred to me that I may have overlooked in his response.  If I just do a single symbolic link to the entirety of 'library-ext/zims/content/' then is he saying I cannot have any content in 'library/zims/content' out side of the symbolic link file?  If that's the case maybe that's what is causing my issue because I have left "test.zim" in that original directory.

Thanks,
Mike

tim-moody, 2020-09-16:
so what you did might well work, but there is a step missing. First, I would prefer
sudo ln -s /library-ext/zims /library/zims
so we don't have the zims on a separate device from the config file library.xml
I'd also mkdir /library-ext/zims/content and index (no longer used but could still be hanging around)
then put your zims in /library-ext/zims/content (including test.zim if you like) and run
iiab-make-kiwix-lib (can also be run from admin console)

also, did you create a link to /library-ext/www/html/modules/?
personally I would have mapped /library-ext/modules/ to /library/www/html/modules/

mihalkaengineering, 2020-09-16:
Solved!

Tim, 

With your additional input it became clear the error I was making with my Symbolic Link.  Based on the command I had run I essentially ended up with a directory that looked like this:
/library/zims/content/content/
Where the second the second content folder was my Symbolic Link.  Once I discovered this I recreated the Symbolic Link one level up in the folder hierarchy and removed the extra content folder.  I did also rebuild the library file.  Once completed the Admin Console was able to recognize zim files stored on the external. 

So thank you for your help!

With that sorted out I did implement your recommendation and moved the entirety of the zim directory to the external drive so my library.xml file, content folder, index folder, and my zim files are stored in the same place.   I will do the same with the OER2Go files.

I am appreciative of your time and I have a couple questions before I wrap up but if you don't have time to answer them I completely understand.
First, now that I have an established path to my external drive will the admin console reject my request for files which require more space than I actually have on my internal drive.  For example, project gutenberg is ~56GB and my internal drive has 15GB free while I have 900+GB available on the external available.
Second, should the Admin Console be able to recognize the content which is copied to my external from a secondary source (for example a USB Flash drive).

Thanks again for all your help!
-Mike

tim-moody, 2020-09-16:
your symlinks should be transparent to admin console and it should do space calculations as before. (let me know if it doesn't)

>should the Admin Console be able to recognize the content which is copied to my external from a secondary source (for example a USB Flash drive)

anything in the linked directories should be recognized, but a zim or module elsewhere won't be. if you copy a zim, don't forget iiab-make-kiiwix-lib.

admin cons can do copies from usb, but they are expected to be under /media and to have the standard directory structure of /library. it can copy in both directions, in and out.

mihalkaengineering, 2020-09-16:
The console is definitely not recognizing the additional space.  The space available doesn't change after new files are added but it never exceeds what is actually available internally.

Do you think if I migrated the entirety of '/library/' to the external drive and put my symlink at the root it would then recognize the additional space.  As of right now the only files being added to the external are the zim files as that is the only directory so far which I have migrated and created a symlink for.

holta, 2020-09-16:
Thanks @mihalkaengineering for persisting!

I'm closing this ticket for now, but feel free to re-open it (or a new one) as nec!

holta, 2020-09-16:
Oops, am re-opening this ticket so the follow-up question can be answer better.

tim-moody, 2020-09-17:
Sorry, I misspoke as I did not remember the code correctly. When adm cons does a storage check it checks the root device and also checks to see if /library is a separate device in which case it shows that storage, but the linux command to get total storage probably doesn't follow links. I was focused on whether the zims would be visible, which they should be.

On other platforms, such as NUCs, we often make /library a separate device. We modify /etc/fstab to mount the device as /library at boot time. If you do that adm cons should see the space on /library. Doing this with a removable drive could cause problems, but I used to have a thumbdrive that I mounted as /library and it worked fine. I don't recall the setting, but you can tell fstab not to require the device in order to boot, so you can still log into the server if it is missing.

So if you use symlinks the space available will have to be estimated manually, but you have a simpler setup. If you feel confident that the drive will be present then I'd mount it as /library

Here is an fstab from a NUC

```
root@iiab-devel:~# cat /etc/fstab
UUID=84dffd80-2977-4a54-b7ce-fbd12c0e9ebe / ext4 defaults 0 0
UUID=c81c6cee-ca33-4c61-bdbd-de7ffae9b07e /library ext4 defaults 0 0
UUID=7210-DABC /boot/efi vfat defaults 0 0
/swap.img       none    swap    sw      0       0

```

jvonau, 2020-09-17:
> I don't recall the setting, but you can tell fstab not to require the device in order to boot, so you can still log into the server if it is missing.

Think you are referring to the "nofail" option.

holta, 2020-09-17:
@mihalkaengineering Thanks Much for shining light on the use of "permanent" external storage.

Please don't hesitate to open a new ticket if you have further questions (or further clarify here!)

closed 2020-09-17 by holta

mihalkaengineering, 2020-09-18:
Thank you so much for all your help.  Much appreciated.