iiab/iiab #2536: Design decisions made by committee, later warnings ignored.

jvonau, 2020-09-21:
### Expected behavior and actual behavior:
Not have roles install when the local_vars.yml is set to False
### Steps to reproduce the problem:
grep samba /etc/iiab/local_vars.yml 
samba_install: False
samba_enabled: False
sudo ./runrole samba or sudo ./runroles samba

>TASK [Set 'samba_installed: True'] ******************************************************
ok: [127.0.0.1]

>TASK [Add 'samba' variable values to /etc/iiab/iiab.ini] ********************************
changed: [127.0.0.1] => (item={'option': 'name', 'value': 'Samba'})
changed: [127.0.0.1] => (item={'option': 'description', 'value': '"Samba is a Microsoft-compatible network file system that re-implements SMB/CIFS (Common Internet File System)."'})
changed: [127.0.0.1] => (item={'option': 'enabled', 'value': False})

Now this has set the user up to fail at the next run of ansible when validate.yml catches the local_vars|iiab_state mismatch of samba_install{ed) in this case. Breaks the basic presumption of False is false.

layout of meta agreed to by committee  https://github.com/iiab/iiab/pull/2052
lockout relation change by committee https://github.com/iiab/iiab/pull/2126
highjack of PR removing installed check from main,yml https://github.com/iiab/iiab/pull/2134
moving of installed from main.yml  ***warning***  https://github.com/iiab/iiab/pull/2142
moving of installed to meta https://github.com/iiab/iiab/pull/2190/files

jvonau, 2020-09-21:
main.yml that need the same treatment as 2537

- azuracast
- bluetooth
- cups
- firmware
- iiab-admin
- minetest
- monit
- mosquitto
- phpmyadmin
- sshd
- transmission
- vnstat
- www_options

jvonau, 2020-09-22:
1d3627510573f343fa140b96a166669bf47e87d0 introduces a "Structural merges at the end of a release cycle" and should be out side of the scope of catching up the lessor used roles.

jvonau, 2020-09-22:
#2539 #2540 #2533 should the release cycle be open as there are structural change going on.

jvonau, 2020-09-22:
Proposed merged order #2541, 2505, 2540, 2539, 2533 with an eye towards getting 2503 in also.

holta, 2020-09-22:
PR #2541 is good enough for now, having solved most all of these runrole-like bugs.

closed 2020-09-22 by holta

holta, 2020-09-22:
> Proposed merged order #2541, 2505, 2540, 2539, 2533 with an eye towards getting 2503 in also.

Good info.  Let's open a new tkt with planning/design info just like this.

jvonau, 2020-09-22:
This **is** the tracking ticket for 'planning/design info' Why would you close this issue?

jvonau, 2020-09-22:
The question has become:
> [1d36275](https://github.com/iiab/iiab/commit/1d3627510573f343fa140b96a166669bf47e87d0) introduces a "Structural merge at the end of a release cycle" #2531 and should be outside of the scope of catching up the lesser used roles.

This is now up for discussion at the next meeting and should not be closed.

tim-moody, 2020-09-24:
No idea what the title means. Who is the committee? Design by committee is worse than better than design by individuals?