iiab/iiab #2542: Design flaw in mysql database password changing

jvonau, 2020-09-22:
Required reading:
https://docs.ansible.com/ansible/latest/modules/mysql_db_module.html
https://docs.ansible.com/ansible/latest/modules/mysql_user_module.html

The setup.yml in a role is where the passwords should be changed but setup.yml are now being called from within install.yml but were not prior to: 
1. f5180d99a2ebefff64e30e3976cbe360fcc0f372#diff-a45913a4ffde64b946e0db1c03591a25 for nextcloud Jan 29, 2020
2. 027832889be31438306dbf6353f85a31e4699a33#diff-1460b7c549aa9ce1a2a72a38aed3cd65 for elgg  Feb 4. 2020 
3. mediawiki lacks setup.yml 

I believe these efforts above were pushed at that time to purposely suppress or attempt to sabotage my #1937 pi-gen branch work opened on Aug 25, 2019 then continuity updated with #2254 -> #2381 -> #2503 to stay current with master.

There is no easy way to update the passwords post-install without "reinstalling" the role while wordpress still has setup.yml in main.yml. Now #2503 attempts to bring the same standardization of operations that the *_installed effort has obtaining by:
1. - Using setup.yml in the same way across the 4 popular apps above.
2. - Ensure mysql is running and records the current running state of mysql.
3. - Introduce *_provisioned for use with the setup.yml routines, like install.yml currently has.
4. - Introduce the optional ability to suppress the running of setup.yml.

Now the stage is set to be able to more easily change the mysql passwords in the future, and allows any image builder an easy way to accommodate user changing passwords in local_vars on prefabricated images build with 'provision_active True' 
at a later point in time.

holta, 2020-09-22:
If there is a serious bug with our treatment of MySQL / MariaDB passwords, there aren't many but there still there are a few days left before IIAB 7.2 is released, if folks can confirm that some kind of emergency fix is needed.

jvonau, 2020-09-22:
As a matter of fact as of today it is not possible to change the default mysql passwords without editing vars/default_vars.yml, but a crafty implementer could populate local_vars before the initial install. Post install, changing in the field, prior to deployment is not possible.

jvonau, 2020-09-23:
http://wiki.laptop.org/go/IIAB/FAQ#What_are_the_default_passwords.3F
>[*] While the Admin Console can change iiab-admin's password instantly, it's important to understand that many Admin Console changes (within its Configure menu especially) require you click "Save Configuration" then "Install Configured Options" and then wait for this to complete. Monitor for Status "SUCCEEDED" under Utilities menu -> Display Job Status (takes about 20 min on a Raspberry Pi 3, 3 B+ or 4, depending what changes you've requested!)

>Finally, certain install-time passwords are listed in /etc/iiab/local_vars.yml which overrides passwords listed in default_vars.yml. If you change any of these install-time passwords (by modifying local_vars.yml) don't forget to then run Ansible, e.g. using "Install Configured Options" above

closed 2025-04-06 by jvonau