iiab/iiab #2552: Usb booted Rpi hardware can display the contents of "/"

jvonau, 2020-09-29:
### Expected behavior and actual behavior:
Boot Rpi hardware capable of usb boot from an usb drive and not be able to see contents of '/' at http://box/usb

from: #2497 
>981 =IIAB==========================================================================
 982 COMMAND: /bin/df -h    # Disk usage
 983 
 984 Filesystem      Size  Used Avail Use% Mounted on
 985 /dev/root       917G   11G  860G   2% /
 991 /dev/sda1       253M   54M  199M  22% /boot

> 995 =IIAB==========================================================================
 996 COMMAND: /bin/lsblk    # Partition mount points
 997 
 998 NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
 999 sda      8:0    0 931.5G  0 disk 
1000 ├─sda1   8:1    0   256M  0 part /media/usb0
1001 └─sda2   8:2    0 931.3G  0 part /media/usb1

### Steps to reproduce the problem:
see above and note the boot device is mounted at the usb share point

### IIAB version, OS version, hardware details:
master as of 2497

Suggestions:
1. toggle local_vars iiab_usb_lib_show_all: to False, and document the effects of True
or
2. add code to override the selection of True when usb-boot is detected, and note why the override is present

holta, 2020-09-29:
@pumpkinpi85 have you too observed this?

CONTEXT: http://FAQ.IIAB.IO > "4 Can teachers display their own content?"

While USB boot of IIAB machines is not common, this is definitely undesirable behavior in those cases!

holta, 2020-09-29:
For now I've added this as an [IIAB 7.2 Known Issue](https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes#known-issues) : 

- If you're booting from a USB device, please set `iiab_usb_lib_show_all: to False` in [/etc/iiab/local_vars.yml](http://wiki.laptop.org/go/IIAB/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it.3F) so that your filesystem is not viewable at http://box/usb  <sub><sub>[#2552](https://github.com/iiab/iiab/issues/2552)</sub></sub>

For IIAB 8.0 let's think about what cleaner solution(s) might be better.

jvonau, 2020-09-29:
The usbboot is not a new concept, NextCloudPi supports it, with mentions of [usb boot and iiab issues](http://lists..laptop.org/pipermail/server-devel/2019-March/008307.html) prior to usb booted devices support being allowed https://github.com/iiab/iiab-factory/issues/103 in April. 

Given the whole install is driven via local_vars.yml sensible defaults should employed to prevent newer users who are not fully versed in the inner working of iiab yet to have a reasonably secure install out of the box without having to fiddle with the defaults. Power users who are generating images for re-deployment should set this value according to their individual needs and not force a new user to be fully aware of what is in the release notes.

tim-moody, 2020-10-21:
I have long thought that iiab_usb_lib_show_all should be false by default. A naïve user is just as likely to expose grades or other sensitive material on the usb even when not booting from it.

jvonau, 2020-10-21:
> I have long thought that iiab_usb_lib_show_all should be false by default. A naïve user is just as likely to expose grades or other sensitive material on the usb even when not booting from it.

Valid point but that does't tickle 'the leadership's' fancy, who are too busy trying to market themselves rather than the project as a whole.

holta, 2020-10-21:
> I have long thought that iiab_usb_lib_show_all should be false by default. A naïve user is just as likely to expose grades or other sensitive material on the usb even when not booting from it.

Teachers don't want that the bureaucratic complication of a subdirectory, as many Mexican school visits reconfirmed in February and March of this year.

They again made clear they want their content displayed immediately, without delay.

holta, 2020-10-21:
> Valid point but that does't tickle 'the leadership's' fancy, who are too busy trying to market themselves rather than the project as a whole.

Personal attacks against teachers and others are not appropriate.

They need to end.

jvonau, 2020-10-21:
> > I have long thought that iiab_usb_lib_show_all should be false by default. A naïve user is just as likely to expose grades or other sensitive material on the usb even when not booting from it.
> 
> Teachers don't want the bureaucratic complication of a subdirectory to hold learning on content USB sticks, as many Mexican school visits reconfirmed in February and March of this year.
> 
The question is what should be the most suitable defaults to protect the 'naïve user', the defaults should provide an opt-in opportunity not an opt-out. The distinction needs to be made between a "drive-by first time curl user's" point of view and that of an install being prepared for a specific deployment. The former user is more likely to just select 1-3 at the local_vars selection point and skip editing altogether in 'iiab' without understanding any of the underlying defaults is what I believe Tim and myself are trying to bring to the forefront to discuss.

> They again made clear they want content displayed immediately from their USB sticks, without delay.

The latter user could/(should?) be using a pre-customized local_vars.yml with the deployment's specific wishes taken into account and dropped into place before starting 'iiab' or stopping iiab at the editing prompt and copying a custom local_vars.yml into place and/or editing then continuing with iiab.

holta, 2020-10-21:
Personally I disagree.

I believe teachers are not being listened to nearly enough, and this is a systemic problem with IIAB, just like it was with OLPC.

holta, 2020-10-22:
> Suggestions:
> 
> 1. toggle local_vars iiab_usb_lib_show_all: to False, and document the effects of True
>    or
> 2. add code to override the selection of True when usb-boot is detected, and note why the override is present

One of several other options beings discussed on this week's http://minutes.iiab.io call:

Manually disable using Admin Console (http://box.lan/admin) > Content > Services Enabled > USB based content libraries ?

(Question is if that's sufficient: would that possibly not be quite enough to FULLY protect folks who boot from USB devices...as it removes the web link but doesn't prevent that secondary mounting of the main/OS filesystem e.g. /media/usb1 with a symlink from /library/www/html/local_content ?)

holta, 2020-10-22:
These 2 prominent warnings will evolve along any technical improvments (currently being discussed on http://minutes.iiab.io call) we come with:

1. https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes#known-issues
2. http://FAQ.IIAB.IO > "45 Can I install IIAB onto a USB drive?"

holta, 2020-10-22:
fyi @jvonau is looking into how safe this Admin Console workaround is at the moment:

> Admin Console (http://box.lan/admin) > Content > Services Enabled > USB based content libraries

holta, 2020-10-22:
Good News: @jvonau confirmed that toggling `usb_lib_enabled: True` to `usb_lib_enabled: False` and then running `./runrole usb_lib` (and then rebooting!) works to hide the contents of the USB (mount) disk from http://box/usb

holta, 2020-10-22:
I've beefed up these 2 warnings, explaining how to work around the security/privacy exposure both before and after installing IIAB:

> These 2 prominent warnings will evolve along with any technical improvments (currently being discussed on http://minutes.iiab.io call) we come with:
> 
> 1. https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes#known-issues
> 2. http://FAQ.IIAB.IO > "45 Can I install IIAB onto a USB drive?"

Good enough for now.

In future these 2 options would be even better, if either or both turns out to be achievable?
1. find a way to prevent such USB-booted filesystems from mounting themselves twice (e.g. to /media/usb1)
2. if IIAB could be made to prevent the softlink from being created from /library/www/html/local_content

holta, 2020-10-22:
This can later be moved to IIAB 8.0's Milestone as nec, e.g. if a design is possible for a more proper fix in future, that would allow IIAB's run from an external/USB booted disk to also live-display the content on teachers' USB sticks (at http://box/usb).

holta, 2020-11-03:
We mitigated this 2 weeks ago with a now-well-documented Admin Console workaround.

Should we now close this issue, thanks to @jvonau's more proper fix PR #2607 ?

_(Or is PR #2607 not significant here?)_

jvonau, 2020-11-04:
> We mitigated this 2 weeks ago with a now-well-documented Admin Console workaround.
> 

It's not an exclusively an Admin Console workaround but rather a local_vars workaround that can be applied with the Admin Console.
>Should we now close this issue, thanks to @jvonau's more proper fix PR #2607 ?
> 
> _(Or is PR #2607 not significant here?)_

That was culling of dead wood, not related. Once a routine that auto detects the situation and correctly deals with it automatically without forcing users to change the setting on their own should this issue be considered for closing.

jvonau, 2021-04-28:
#2739

jvonau, 2022-06-17:
Closing should be resolved with 3254

closed 2022-06-17 by jvonau

holta, 2022-06-17:
> Closing should be resolved with 3254

Great!  Here's the direct link to @tim-moody's:

- PR #3254

jvonau, 2022-06-17:
Don't clutter up the issues with a bunch of re-posts, there is a direct link to 3254 just above, from when you mentioned this issue in 3254.

holta, 2022-06-17:
Thanking people for their efforts is not clutter.  Otherwise we're all in trouble ;-)