iiab/iiab #2560: references to admin console

tim-moody, 2020-10-06:
I don't understand 4-server-options/tasks/main.yml 59ff:
```
# 2020-02-17: what was roles/homepage lives in roles/www_options.  Eventually
# softcoding of iiab_home_url should happen everywhere (incl Admin Console) for
# more field options, e.g. changing /library/www/html/home even when offline...
```
Is this still needed, nginx/tasks/install.yml 19ff?

```
- name: Add user '{{ apache_user }}' to shadow group, so it can authenticate Admin Console
  user:
    name: "{{ apache_user }}"    # www-data or apache, per /opt/iiab/iiab/vars/<OS>.yml
    groups: shadow

```
captiveportal/tasks/enable-or-disable.yml and elsewhere. If desired admin console can install uwsgi: it already installs the config files for admin console. (btw it only uses 1 child process, though with 4 threads)

```
# 2020-01-22: Package 'uwsgi' is installed by nginx/tasks/install.yml for
# Captive Portal (responds to browsers trying to detect a Captive Portal) and
# also for Admin Console.

# IIAB makes the 'uwsgi' service mandatory for now (see Line 45) but this could
# perhaps be auto-disabled in future (or never installed?)  e.g. if Captive
# Portal + Admin Console are both unneeded...and impact of uWSGI is too high ?

- name: Stop 'uwsgi' systemd service (may include 3 child processes for Captive Portal, and 3 child processes for Admin Console)
  systemd:
    name: uwsgi
    state: stopped

```

holta, 2020-10-08:
@jvonau helped clarify the 3 questions above with @tim-moody during our call today (http://minutes.iiab.io)

1. Comment should be freshened up (I'll work on that)
2. Likely that stanza should be removed, @jvonau & @tim-moody agree
3. @tim-moody wonders if Admin Console should install uWSGI, just to be safe.  @jvonau is not sure that necessary, but perhaps no harm.

holta, 2020-10-08:
> 1. Comment should be freshened up (I'll work on that).

Done.  This is part of PR #2565:

https://github.com/holta/iiab/commit/fff2829fd83c1b4d2ebdf81141f126fb7d3fbbe0

holta, 2020-10-17:
Appears resolved.  @tim-moody please re-open as nec.

closed 2020-10-17 by holta