iiab/iiab #2561: TK: Installing IIAB disables graphical login to Ubuntu Desktop 20.04 accounts (administrative accounts only?)

holta, 2020-10-07:
@jvonau is monitoring TK Kang's progress in understanding the problem, which has been reproduced ~3 times now on Ubuntu 20.04 VM's and PC's.

jvonau, 2020-10-07:
#2562 for the wired slave detection as somebody actually used that code path, thanks TK as I don't have a U-20 amd64 reproducer handy.

> which has been reproduced ~3 times now on Ubuntu 20.04 VM's and PC's.

At least 2 different people should be tripped up not being able to login before panicking. Think this would be showing up on Mint 20 as it's close cousin to Ubuntu. From Chat

>To narrow this down I would do a stock desktop install on hardware or VM and note if you are forced to login at the select user dialog box after rebooting several times. Once the stock behavior is established, run just the part 'iiab' to the point where the iiab-admin user is created and `ctrl c`after the password is set. Reboot and note if the select user login box is present or not.

jvonau, 2020-10-07:
More context
>On Tue, Oct 6, 2020 at 11:54 PM T.K. Kang wrote:
            OK... if I login with iiab-admin the desktop will reappear. Hence the issues are account that are created prior to IIAB or after IIAB install.
            Cheers
    On Wed, Oct 7, 2020 at 1:30 PM Jerry Vonau wrote:
        Is this a case of having 2 possible logins and now the display manager can't or refuses to auto login?
On Wed, Oct 7, 2020 at 6:47 PM T.K. wrote:
    Only the iiab-admin account can login in via local desktop to get the ubuntu interface. My admin account will just display a blank page. However can ssh from another terminal.

holta, 2020-10-08:
More context from TK Kang:
> > Can you clarify the exact URL of the Ubuntu 20.04 installer you are using?
> >
> > e.g. if https://ubuntu.com/download/desktop (or any other!)

> ubuntu-20.04-desktop-amd64
>
> Will try the latest ubuntu-20.04.1-desktop-amd64 for another install again

shanti-bhardwa, 2020-10-08:
did 7.2 build on Ubuntu-20.04.1-desktop-amd64.iso . Medium build, cannot loging using my original userid, but can do so using iiab-admin. see http://spruge.us/JBOgXV

looks like my profile as original user is being changed by iiab-admin

holta, 2020-10-08:
A big thank you to @shanti-bhardwa who confirmed @jvonau's suspicion that deleting `/etc/profile.d/sshpwd-profile-iiab.sh` appears to solve this problem!

We should identify which of the 3 main stanzas here is the culprit:
https://github.com/iiab/iiab/blob/master/roles/iiab-admin/templates/sshpwd-profile-iiab.sh

Does anyone have a hunch?

Tangentially related: #1537

holta, 2020-10-09:
As suggested by PR#1551, this is likely related:
https://github.com/iiab/iiab/blob/master/roles/iiab-admin/tasks/main.yml#L38-L44

jvonau, 2020-10-11:
Given somebody has to kickoff the install with an account with sudo access, here is a novel idea, don't create iiab-admin at all in script `iiab` and don't force iiab-admin to be created with a default install. Those who want iiab-admin can toggle the local_vars file prior to install, or install iiab-admin on demand with iiab-support at any time. I personally don't like a second admin account present and can view iiab-admin as a backdoor that needs to be removed post install. Might want to canvass other users on their feelings and opinions on the subject.

tim-moody, 2020-10-11:
>Given somebody has to kickoff the install with an account with sudo access

can be root

jvonau, 2020-10-11:
Sure sudo grants root like privileges to the user once verified, but sure running the install from a root shell works also. There are really two parts to original usage of the iiab-admin user, first is the login account and second is iiab_admin_user that is now only used for the password changing function in the admin-console, prior usage required logging in as iiab-admin. Now that authentication can be any valid sudo user in the admin-console the need for iiab-admin for access is really non-existent now and the defaults should reflect that going forward. I find the stock OS proved admin user account enough for day to day usage, less is better is my maxum, just one less thing to maintain.

tim-moody, 2020-10-11:
Personally I don't like using the stock admin user as it varies by OS, but if you do just set iiab_admin_user equal to it. Now you can change its password without having to ssh in.

jvonau, 2020-10-11:
I set the iiab_admin_user as needed and don't create the iiab-admin login user. The varies by OS part I agree with, but for the most popular arm OSs, pi and ubuntu is pretty standard its the X86_64 installs from iso that might force the creation of a new admin account from scratch and force one to come up with a name. Can't recall, it's been awhile.

tim-moody, 2020-10-11:
>I set the iiab_admin_user as needed and don't create the iiab-admin login user.

To be clear, I think that if you set iiab_admin_user to 'pi' for example, the iiab-admin login will not be created.

jvonau, 2020-10-11:
> > I set the iiab_admin_user as needed and don't create the iiab-admin login user.
> 
> To be clear, I think that if you set iiab_admin_user to 'pi' for example, the iiab-admin login will not be created.

No, you need to explicitly set iiab_admin_user_install to False in local_vars.yml to suppress the iiab-admin login account creation. As you said iiab_admin_user is the variable used for admin-console's password changing function, but any valid sudo user may login. This is what I do on RaspOS:
>iiab_admin_user_install: False
iiab_admin_user: pi

Note for others that may try this, 'iiab_admin_user' must match a local preexisting account with sudo privileges(ubuntu, sysop, or whatever somebody came up with) for the password changing to work. I have not noted any ill effects setting iiab_admin_user to a dummy value upon admin-console's operation. I just wound not try to run the password utility knowing the underling account does not exist and not sure if there are any bad effects but suspect just a minor account not found error is passed back.

tim-moody, 2020-10-11:
>No, you need to explicitly set iiab_admin_user_install to False in local_vars.yml to suppress the iiab-admin login account creation

we disagree on the facts

jvonau, 2020-10-11:
I'll agree that we disagree on the facts.
[jerry@box iiab]$ grep admin_user_install vars/*
>vars/default_vars.yml:# Set iiab_admin_user_install: False if you don't want iiab_admin_user & wheel
vars/default_vars.yml:iiab_admin_user_install: True
vars/default_vars.yml:# If iiab_admin_user_install: False, set iiab_admin_user (below) to an existing
vars/default_vars.yml:# e.g. iiab_admin_user_install, iiab_admin_user, iiab_admin_pwd_hash
vars/local_vars_big.yml:# Set iiab_admin_user_install: False if you don't want iiab_admin_user & wheel
vars/local_vars_big.yml:iiab_admin_user_install: True
vars/local_vars_big.yml:# If iiab_admin_user_install: False, set iiab_admin_user (below) to an existing
vars/local_vars_big.yml:# e.g. iiab_admin_user_install, iiab_admin_user, iiab_admin_pwd_hash
vars/local_vars_medium.yml:# Set iiab_admin_user_install: False if you don't want iiab_admin_user & wheel
vars/local_vars_medium.yml:iiab_admin_user_install: True
vars/local_vars_medium.yml:# If iiab_admin_user_install: False, set iiab_admin_user (below) to an existing
vars/local_vars_medium.yml:# e.g. iiab_admin_user_install, iiab_admin_user, iiab_admin_pwd_hash
vars/local_vars_min.yml:# Set iiab_admin_user_install: False if you don't want iiab_admin_user & wheel
vars/local_vars_min.yml:iiab_admin_user_install: True
vars/local_vars_min.yml:# If iiab_admin_user_install: False, set iiab_admin_user (below) to an existing
vars/local_vars_min.yml:# e.g. iiab_admin_user_install, iiab_admin_user, iiab_admin_pwd_hash

[jerry@box iiab]$ cat roles/1-prep/tasks/main.yml 
>- name: IIAB-ADMIN
  include_role:
    name: iiab-admin
  #when: iiab_admin_install | bool    # Flag might be created in future?

[jerry@box iiab]$ cat roles/iiab-admin/tasks/main.yml 
>- include_tasks: admin-user.yml
  when: iiab_admin_user_install | bool

[jerry@box iiab]$ cat roles/iiab-admin/tasks/admin-user.yml 
>- name: Create user {{ iiab_admin_user }} for Admin Console; set password from iiab_admin_pwd_hash if newly creating account
  user:
    name: "{{ iiab_admin_user }}"
    password: "{{ iiab_admin_pwd_hash }}"
    update_password: on_create
    shell: /bin/bash

tim-moody, 2020-10-11:
excellent. now set iiab_admin_user : pi in local vars and tell me where the user iiab-admin will get created.

jvonau, 2020-10-11:
Sure, 'iiab-admin' is first created by '/usr/sbin/iiab', which neither of us uses when building. The user creation is then duplicated and sudo is doctored up in roles/iiab-admin/tasks/admin-user.yml only when iiab_admin_user_install is set to True based on what iiab_admin_user is populated with from 1-Prep's use of the iiab-admin's role is my understanding. What is your understanding?

closed 2020-10-11 by holta

tim-moody, 2020-10-11:
>Sure, 'iiab-admin' is first created by '/usr/sbin/iiab'

which is to say none of the 'facts' you cited was relevant, only

```
# C. Create user 'iiab-admin' as nec, with default password
if ! id -u iiab-admin > /dev/null 2> /dev/null; then
    useradd iiab-admin
    echo iiab-admin:g0adm1n | chpasswd
fi

```
I would say this is a bug in /usr/sbin/iiab in that it does not honor iiab_admin_user or iiab_admin_pwd_hash. (another potential use for vars-util)

I don't object to either fixing or removing it. I guess I'd prefer the latter, but expect it is here so that a prompt to change the password can be issued. Could still use the variables.

tim-moody, 2020-10-11:
@jvonau Jeez, I finally understand what you were saying as I assumed you meant created via ansible when running iiab and not in the script itself. So turns out we do agree.

jvonau, 2020-10-11:
> > Sure, 'iiab-admin' is first created by '/usr/sbin/iiab'
> 
> which is to say none of the 'facts' you cited was relevant, only

> >which neither of us uses when building... 

The issue is the unsuspecting curl newbie who doesn't get a choice up front, others use scripts or just setup though ansible. iiab-install, install to achieve the end result, there are many paths to success.

> # C. Create user 'iiab-admin' as nec, with default password
> if ! id -u iiab-admin > /dev/null 2> /dev/null; then
>     useradd iiab-admin
>     echo iiab-admin:g0adm1n | chpasswd
> fi
> ```
>
> I would say [...]

I tend to agree with the bug aspect given the necessary workarounds noted above.  The user created and password is currently set before looking for a local_vars.yml file to source. Perhaps for the best of both worlds with little effort the iiab_user/password routine can come after checking in the local_vars.yml file or only when a local_vars.yml needs to be retrieved because one was not suppled?

PR #2566 (2020-10-11): Fix for Graphical and TTY Logins [clean up auto-checking for iiab-admin's published password]
Resolves #2561 and its very serious `sudo` problem.

By greatly simplifying `/etc/profile.d/sshpwd-profile-iiab.sh` since the prior version was quite broken, for those with graphical desktops especially, as TK & @shanti-bhardwa confirmed.

Needs more testing on more OS's than just Ubuntu Desktop 20.04.1 (where it was smoke-tested & functionality tested) so this important fix can hopefully be merged in short order.

Refs: #1252, PR #1253, #1537, PR #1551