iiab/iiab #2563: Using local vars outside of ansible yields uncertain results

tim-moody, 2020-10-07:
Inside of ansible absence in local vars wll cause the value of default vars to be used. Outside of ansible absence in local vars will lead to uncertain results. 

If a runtime config is required then a new file could be used. For example cmdsrv has a config file that gets defaults from default or local vars at install time, but can later be edited.

Otherwise all tests for values in local vars should also test default vars.

tim-moody, 2020-10-07:
OK. I have an idea.

closed 2020-10-07 by tim-moody

holta, 2020-10-07:
> OK. I have an idea.

It is a general challenge for sure.

e.g. @jvonau's [iiab-check-firmware](https://github.com/iiab/iiab/blob/master/roles/firmware/templates/iiab-check-firmware) is just one of several examples needing rapid runtime configuration changes w/o Ansible bureaucracy, while also respecting implementers/operators wishes as best we can anyway (as they've encoded into /etc/iiab/local_vars.yml).