iiab/iiab #2567: Map Pack install fails

holta, 2020-10-09:
@shanti-bhardwa how are you trying to download/install the Africa Map Pack?  It looks like you might be using the older Admin Console approach?

Have you tried the NEW approach at http://box > "Install IIAB Maps" ?

Your "404: Not Found" error at https://raw.githubusercontent.com/iiab/maps/master/osm-source/regional-base/build/main.js "404: Not Found" (apparently caused by /opt/admin/cmdsrv/scripts/osm-vect_install_step3.sh ?) does not look good, for sure!

( i.e. if you fell in this trap, it's guaranteed many others will too :O )

Long-form instructions should not be needed by most folk -- and certainly do no good if people can't find them -- but FYI here they are just in case:

"How do I install Map Pack(s) and Satellite Photo Region(s) on IIAB 7.2+ ?"
https://github.com/iiab/iiab/wiki/IIAB-Maps#how-do-i-install-map-packs-and-satellite-photo-regions-on-iiab-72-

holta, 2020-10-09:
@shanti-bhardwa's trace pasted in here:

```
scripts/osm-vect_install_step3.sh /library/working/maps/en-osm-omt_africa_2017-07-03_v0.23 /library/www/osm-vector-maps /library/downloads/maps/en-osm-omt_africa_2017-07-03_v0.23.zip

Swapping 3MB /library/working/maps/en-osm-omt_africa_2017-07-03_v0.23/main.js for a newer one, to help with low-memory client devices

--2020-10-08 22:23:35-- https://raw.githubusercontent.com/iiab/maps/master/osm-source/regional-base/build/main.js

Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 199.232.52.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|199.232.52.133|:443... connected.

HTTP request sent, awaiting response... 404 Not Found

2020-10-08 22:23:35 ERROR 404: Not Found.

Download of main.js FAILED
FAILED10-08 15:52 06:31:18  

/usr/bin/unzip -uo /library/downloads/maps/en-osm-omt_africa_2017-07-03_v0.23.zip -d /library/working/maps/

inflating: /library/working/maps/en-osm-omt_africa_2017-07-03_v0.23/src/index.html
inflating: /library/working/maps/en-osm-omt_africa_2017-07-03_v0.23/main.js
inflating: /library/working/maps/en-osm-omt_africa_2017-07-03_v0.23/regions.json
inflating: /library/working/maps/en-osm-omt_africa_2017-07-03_v0.23/init.json
inflating: /library/working/maps/en-osm-omt_africa_2017-07-03_v0.23/searchapi.php
inflating: /library/working/maps/en-osm-omt_africa_2017-07-03_v0.23/index.html
linking: /library/working/maps/en-osm-omt_africa_2017-07-03_v0.23/satellite.mbtiles -> satellite_z0-z9_v3.mbtiles
finishing deferred symbolic links:
/library/working/maps/en-osm-omt_africa_2017-07-03_v0.23/detail.mbtiles -> ./africa.mbtiles
/library/working/maps/en-osm-omt_africa_2017-07-03_v0.23/satellite.mbtiles -> satellite_z0-z9_v3.mbtiles

SUCCEEDED10-08 15:52 06:31:14  

usr/bin/wget -c --progress=dot:giga https://archive.org/download/en-osm-omt_africa_2017-07-03_v0.23/en-osm-omt_africa_2017-07-03_v0.23.zip -O /library/downloads/maps/en-osm-omt_africa_2017-07-03_v0.23.zip

8257536K ........ ........ ........ ........ 97% 392K 8m0s
8290304K ........ ........ ........ ........ 98% 476K 6m28s
8323072K ........ ........ ........ ........ 98% 475K 4m57s
8355840K ........ ........ ........ ........ 99% 364K 3m27s
8388608K ........ ........ ........ ........ 99% 391K 1m56s
8421376K ........ ........ ........ ........ 99% 693K 25s
8454144K ........ 100% 727K=6h29m

2020-10-08 22:21:36 (362 KB/s) - ?/library/downloads/maps/en-osm-omt_africa_2017-07-03_v0.23.zip? saved [8666403119/8666403119]
SUCCEEDED10-08 15:52 06:29:18
```

tim-moody, 2020-10-09:
The 404 is on main.js. The map pack actually downloaded according to the attached output. But it is the older version map pack, so it will not work with the new osm vector maps because it will be in the wrong directory, besides which it is an older set of tiles. Current Admin Console does not know about the new map structure.

holta, 2020-10-09:
Let's try find the best place(s) to link to https://github.com/iiab/iiab/wiki/IIAB-Maps#how-do-i-install-map-packs-and-satellite-photo-regions-on-iiab-72- so people like @shanti-bhardwa don't get lost, in Admin Console and elsewhere?

tim-moody, 2020-10-09:
A short term approach would be to modify the existing UI to use the new catalog as a phase I. This would allow maps to be downloaded properly but none of the other new functionality.

tim-moody, 2020-10-09:
Not as short as I thought

holta, 2020-10-09:
Possible workaround / fix?

https://github.com/iiab/iiab-admin-console/pull/344

holta, 2020-10-11:
Short-term mitigation looks good here:

https://github.com/iiab/iiab-admin-console/compare/0.4.4.1-iiab-7.2

holta, 2020-10-16:
PR iiab/iiab-admin-console#349 (merged) helps public testers here.

closed 2020-10-16 by holta