iiab/iiab #2574: TK Question: RPi 3 isn't connecting reliably to home WiFi router

holta, 2020-10-15:
TK's report:

> "My IP is 192.168.0.123  172.18.96.1"  together the IIAB will work normally, Most of the time it display only one IP (either a 192.168.0.123 or just 172.18.96.1) and things do not work.
>
> There is issue of stability .. How do I get it to work all the time?’

@jvonau has requested the output of [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) ...to try to investigate if his Raspberry Pi 3 hardware ("that has a a broken Ethernet port") and/or router are otherwiseok...or whether this problem is independent of IIAB possibly?

jvonau, 2020-10-16:
> T.K: I added LXDE desktop on this RaspOs lite
You:  so your roll your own desktop installed 'wicd' as a network controller... it's occasional scanning might upset the normal course of events..
T.K. The apt install task-lxde-desktop must have installed it.
T.K. How to delete it .. so life back to nornal
You: most likely, I just try to stop it from starting...
You: sudo systemctl disable wicd
You: and reboot

holta, 2020-10-16:
Huge thanks to @jvonau for solving this with TK.

Question: can people like TK (who replace the guts of their networking by accident, installing NetworkManager-or-whatever by accident) ...can then run `cd opt/iiab/iiab` then `sudo ./iiab-network` to rebuild their network layout in such situations?

(Or is that asking too much of ./iiab-network ?)

jvonau, 2020-10-16:
> Huge thanks to @jvonau for solving this with TK.
> 
> Question: can people like TK (who replace the guts of their networking by accident when installing a desktop, e.g. by installing NetworkManager-or-whatever by accident) ...can they then run `cd opt/iiab/iiab` and `sudo ./iiab-network` to recover, by rebuilding their network layout in such situations?

Read the network code, dealing with the much widely used NetworkManager is supported, but as noted in https://github.com/iiab/iiab/pull/2562#issuecomment-708479142 manual intervention is needed.
> (Or is that asking way too much of ./iiab-network ?)

I'm not interested in automating that step given the runaround I have been subjected to regarding any PR submitted.

holta, 2020-10-16:
Hopefully TK is grateful for all the support you've provided him.

And as a result more people will realize that:

1. It's unrealistic to believe [./iiab-network](https://github.com/iiab/iiab/tree/master/roles/network) to solve all their never-ending networking particulars.
2. Running `cd /opt/iiab/iiab` then `./iiab-network` (and rebooting) can solve _many-but-not-all_ networking issues!

closed 2020-10-16 by holta

holta, 2021-06-26:
Thank you @jvonau & TK for having tracked this down in October 2020.  In summary:

- Very unfortunately Wicd (Wireless Interface Connection Daemon) causes problems problems with IIAB networking.

- Sadly Wicd appears unmaintained for 5.5 years now:

  https://en.wikipedia.org/wiki/Wicd

- There are no Wicd announcements for 7 years now, since 2014:

  https://launchpad.net/wicd