iiab/iiab #2584: Sierra Leone: "can no longer log into KA Lite and Admin Console"

holta, 2020-10-21:
I'm trying to understand the situation that (might) have been caused by the implementer/operator installing additional packages after everything was working?

Output of [sudo iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) is here: http://sprunge.us/HdPJ80?en

jvonau, 2020-10-22:
>1014 /dev/root       217G  216G     0 100% /

That needs attention

holta, 2020-10-22:
>I was able to delete some KA-Lite videos to free up some space and that solved the problem . 
I plan to deploy this installation in a school in Sierra Leone.

Where should/could IIAB make a "DISK FULL" warning much more visible/colorful at http://box and/or on the Admin Console page (http://box.lan/admin) _prior to login_ ?

i.e. to give operators/educators a head's up when this relatively common situation occurs...

jvonau, 2020-10-23:
If the disk is that full you will never be able to login, or as reported access the webserver, that is why there was the 5% fudge factor for root normally present to help to prevent that kind of issue, but was tossed to the wayside in favor of 1% later raised to 2% on RPi hardware.

closed 2021-05-21 by holta