iiab/iiab #2585: Testing the new Ubuntu Desktop 20.10 for RPi 4 (rumor confirmed!)

holta, 2020-10-22:
Can someone help evaluate and test install http://download.iiab.io on this just-released hot-off-the-press new Ubuntu Desktop 20.10?

There's only 1 graphical desktop version available.  But the almost-7-month-old gossip (that Ubuntu would mainline a graphical desktop for Raspberry Pi !) turns out to be true in the end:

https://ubuntu.com/download/raspberry-pi

Noting that Ubuntu Server 20.10 is released in the usual array of 5 versions available for RPi 2, 3 and 4 (32-bit and 64-bit).  I don't expect any problems with the 64-bit server version for RPi 4 as its pre-releases have already been quite heavily tested in recent weeks.  But a final test here would also be very useful to sanity check https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes &mdash; and also to confirm which WiFi drivers they're officially shipping within `/lib/firmware/brcm` per https://github.com/iiab/iiab/issues/823#issuecomment-662285202

shanti-bhardwa, 2020-10-23:
/boot/firmware$ cat config.txt
[pi4]
max_framebuffers=2

[all]
arm_64bit=1
kernel=vmlinuz
cmdline=cmdline.txt
initramfs initrd.img followkernel

# Enable the audio output, I2C and SPI interfaces on the GPIO header
dtparam=audio=on
dtparam=i2c_arm=on
dtparam=spi=on

# Enable the FKMS ("Fake" KMS) graphics overlay, enable the camera firmware
# and allocate 128Mb to the GPU memory
dtoverlay=vc4-fkms-v3d
gpu_mem=128
start_x=1

# Comment out the following line if the edges of the desktop appear outside
# the edges of your display
disable_overscan=1

# If you have issues with audio, you may try uncommenting the following line
# which forces the HDMI output into HDMI mode instead of DVI (which doesn't
# support audio output)
#hdmi_drive=2

# If you have a CM4, uncomment the following line to enable the USB2 outputs
# on the IO board (assuming your CM4 is plugged into such a board)
#dtoverlay=dwc2,dr_mode=host

holta, 2020-10-23:
Here is a bit more context on @shanti-bhardwa 's Bluetooth issue of Ubuntu Desktop 20.10: (Groovy Gorilla, released yesterday)

```
Error: TASK [bluetooth : Disable built-in serial uart in /boot/firmware/usercfg.txt on ubuntu]
**************************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Destination /boot/firmware/usercfg.txt does not exist !", "rc": 257.
```

> Blue tooth is working on the desktop as I can pair my phone with PI4. Here is the diagnostics http://sprunge.us/WqHpvW

jvonau, 2020-10-23:
comment out https://github.com/iiab/iiab/blob/master/roles/bluetooth/tasks/install.yml#L1-L6 as something changed with the ubuntu file layout.

jvonau, 2020-10-23:
For context the contents of /boot/firmware/*,txt files from U-20.04 for RPi would be useful also.

holta, 2020-10-23:
What kind of Ansible `when:` clause should be considered if Ubuntu 20.10 is different from 20.04 ?

Obviously `when: is_ubuntu_20` (which applies to both) doesn't quite suffice.

jvonau, 2020-10-23:
My crystal ball is fuzzy, re-ask the question once more info is gathered.

shanti-bhardwa, 2020-10-23:
shanti@shanti-desktop:/boot/firmware$ cat cmdline.txt

dwc_otg.lpm_enable=0 console=tty1 root=LABEL=writable rootfstype=ext4 elevator=deadline rootwait fixrtc quiet splash

jvonau, 2020-10-23:
> What kind of Ansible `when:` clause should be considered if Ubuntu 20.10 is different from 20.04 ?
> 
> Obviously `when: is_ubuntu_20` (which applies to both) doesn't quite suffice.

What do you get with `grep distribution iiab-install.log` on an Ubuntu install?
Ref: https://github.com/ansible/ansible/issues/34511
{{ ansible_distribution_major_version }} should show the broad release number while {{ ansible_distribution_minor_version }} if that variable exists for the given OS should show the other part of the release numbering scheme.

shanti-bhardwa, 2020-10-23:
shanti@shanti-desktop:~$ grep distribution  /opt/iiab/iiab/iiab-install.log
        "ansible_distribution": "Ubuntu",
        "ansible_distribution_file_parsed": true,
        "ansible_distribution_file_path": "/etc/os-release",
        "ansible_distribution_file_variety": "Debian",
        "ansible_distribution_major_version": "20",
        "ansible_distribution_release": "groovy",
        "ansible_distribution_version": "20.10",
        "ansible_distribution": "Ubuntu",
        "ansible_distribution_file_parsed": true,
        "ansible_distribution_file_path": "/etc/os-release",
        "ansible_distribution_file_variety": "Debian",
        "ansible_distribution_major_version": "20",
        "ansible_distribution_release": "groovy",
        "ansible_distribution_version": "20.10",
2020-10-22 23:32:59,132 p=52143 u=root n=ansible | changed: [127.0.0.1] => (item={'option': 'distribution', 'value': 'Ubuntu'})

jvonau, 2020-10-23:
"ansible_distribution_release": "groovy",
"ansible_distribution_version": "20.10",

Either would be specific enough to make the distinction.

holta, 2020-10-26:
Testing ongoing thanks to @floydianslips who is Ethernet-installing BIG-sized IIAB (PR #2191) onto RPi 4 with 8GB RAM.

At the same time I'm WiFi-installing MIN-sized IIAB (PR #2191) onto RPi 4 with 1GB RAM.

(We're presuming ./iiab-install will crash in Stage 4 [soon] when trying to install Bluetooth!)

holta, 2020-10-26:
As a temp workaround I'm tryinig to change Line 6 of [/opt/iiab/iiab/roles/bluetooth/tasks/install.yml](https://github.com/iiab/iiab/blob/master/roles/bluetooth/tasks/install.yml#L1-L6) to:

```
  when: is_ubuntu and rpi_model != "none" and (ansible_distribution_version is version('20.10', '<'))
```

holta, 2020-10-26:
FYI the WiFi-install of IIAB failed here:

```
2020-10-26 13:14:05,536 p=83637 u=root n=ansible | TASK [network : Copy the bridge script] ***********************************************************
2020-10-26 13:14:06,775 p=83637 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": false, "checksum": "0ea624d7abc690111be93223f50426b8f03a5d53", "msg": "Destination directory /etc/network/interfaces.d does not exist"}
```

jvonau, 2020-10-26:
Useless without context, I'm not chasing the information on the latest shinny new thing to come along, How about being proactive with posting iiab-diagnosics on a *brand new OS* without having to be asked? I can review success and/or failures while mobile and prepare a response for a tighter feedback loop if you are truly interest in the reply and a resolution.

holta, 2020-10-26:
I don't have access to either machine but FYI @floydianslips experienced this MongoDB problem with his BIG-sized install:

```
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Failed to update apt cache: E:The repository 'https://repo.mongodb.org/apt/ubuntu groovy/mongodb-org/4.4 Release' does not have a Release file."}

jvonau, 2020-10-27:
> As a temp workaround I changed Line 6 of [/opt/iiab/iiab/roles/bluetooth/tasks/install.yml](https://github.com/iiab/iiab/blob/master/roles/bluetooth/tasks/install.yml#L1-L6) to:
> 
> ```
>   when: is_ubuntu and rpi_model != "none" and (ansible_distribution_version is version('20.10', '<'))
> ```

Ugly looking syntax IMHO, but if it works. Future proofing, U-20 is now the minimum and 20.04 is only place where that files exists think ansible_distribution_release ==  "focal" reads easier.

holta, 2020-10-27:
@floydianslips cc: @deldesir

1) Keep an eye on your 8GB RPi 4 if possible, to make sure the above 3 fixes are enough and the BIG-sized IIAB install completes?  (Currently `/usr/sbin/iiab` is running PR #2595 in a screen session, as root, working through Stage ~9.)

2) If it does complete successfully, take a look at look at recent changes with Admin Console (e.g. map & satellite photo installs) and let us know your thoughts the overall progress!

holta, 2020-10-27:
PR #2596 fixes the Mint issue properly, thank @jvonau for catching that.

Hopefully this tkt can be closed soon with others' confirmation(s), @shanti-bhardwa ?

holta, 2020-10-29:
Ubuntu Desktop 20.10 testing appears far better a week later, thanks to PRs #2588, #2593, #2594, #2595, #2596, #2597, #2599 &mdash; and indirectly related PR's #2600, #2602 more related to Debian 10 and 11 Bullseye (Debian's "testing" branch).

@deldesir @floydianslips please shout if any other flies hit the windscreen?  Otherwise let's close this ticket in the coming week!

fyi our experimental support for Ubuntu 20.10's on RPi[*] is now documented here:
https://github.com/iiab/iiab/wiki/IIAB-Platforms

[*] Let's NOT support all 11 images here.  But refine which of these Ubuntu-on-RPi deserve MORE/LESS of our attention (:
https://ubuntu.com/download/raspberry-pi

shanti-bhardwa, 2020-11-03:
I followed these instructions and desltop is not fast, but useable

 First download Ubuntu Desktop 20.10 from here. Use  64bit for PI4 hardware.

https://ubuntu.com/download/raspberry-pi

2. Follow the Tutorial  here:

https://ubuntu.com/tutorials/how-to-install-ubuntu-desktop-on-raspberry-pi-4#1-overview

shanti-bhardwa, 2020-11-04:
Notes on Installing Ubuntu Desktop on PI4 and Booting from USB/SATA SSD. Shanti Bhardwa. 4 Nov 2020

1. These notes describe how you can build a Desktop version of Ubuntu Groovy Gorilla release  and boot from an SSD drive in USB/SATA Caddy connected to USB3 port on PI4.

2. My hardware for testing is

            ▪ PI4 with 4 GB of RAM, Official PI4 Power Supply and Case
            ▪ UGREEN USB/SATA Caddy from Amazon
            ▪ SUNBOW 120GB SSD from Amazon
              
            ▪ DELL Laptop running LINUX Mint  for downloading and preparing the image file
            ▪ balena Etcher for burning the image file to  creating a bootable drive using the SSD

3. First download Ubuntu Desktop 20.10 Image from here. Use  64bit for PI4 hardware. Extract the compressed image file from the downloaded file.

https://ubuntu.com/download/raspberry-pi

4. The files in question are

ubuntu-20.10-preinstalled-desktop-arm64+raspi.img.xz (1.7GB)
ubuntu-20.10-preinstalled-desktop-arm64+raspi.img (8.8GB)

5. Follow the Tutorial  here  to see how Desktop is built. In this case, you are using Etcher to write the image file to SSD drive rather than the normal SD card.

https://ubuntu.com/tutorials/how-to-install-ubuntu-desktop-on-raspberry-pi-4#1-overview

6.  The instructions for changing the boot order are described here.

https://ubuntu.com/tutorials/how-to-install-ubuntu-desktop-on-raspberry-pi-4#4-optional-usb-boot

7. I  found that by following the first two steps,  the BOOT_ORDER option is already set to 0xf41 ( meaning attempt SD card, then USB mass-storage device) is already set up correctly in this release.

8.  When you type

 sudo apt install rpi-eeprom 

the system will tell you that 

rpi-eeprom is already the newest version (9.0-1ubuntu1).

9. When you type

sudo vcgencmd bootloader_config  

you will get the following showing that BOOT_ORDER set correctly.

[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
SD_BOOT_MAX_RETRIES=1
USB_MSD_BOOT_MAX_RETRIES=1
BOOT_ORDER=0xf41

10. So there is no need to  follow rest of the steps.

11. Connect the Caddy with the SSD drive to PI4 USB3 port and the system will boot up. You will need the PI4 to be connected to the Internet using Ethernet Cable.

12. Follow the on-screem instructions to complete the Desktop Build.

13. Once the build is complete, set up WIFI and then you can use the desktop.

14. Run 1 line installer for iiab 7.2 and it installs OK.

15. See here  http://sprunge.us/c0tdyF

jvonau, 2020-11-04:
@shanti-bhardwa In reference to #2552, can you see the contents of your usbdisk at http://box/usb ?
>sda      8:0    0 111.8G  0 disk 
├─sda1   8:1    0   256M  0 part /media/usb0
└─sda2   8:2    0 111.5G  0 part /media/usb1

Nasty traceback in systemd-dispatcher, think [this](https://bugs.launchpad.net/ubuntu/+source/networkd-dispatcher/+bug/1884248) upstream issue is related.
>-- Logs begin at Thu 2020-09-24 20:27:11 BST, end at Wed 2020-11-04 12:30:32 GMT. --
Nov 04 12:21:09 box.lan systemd[1]: Starting Dispatcher daemon for systemd-networkd...
Nov 04 12:21:11 box.lan networkd-dispatcher[691]: ERROR:Error handling notification for interface 'lo' entering operational state carrier
Nov 04 12:21:11 box.lan networkd-dispatcher[691]: Traceback (most recent call last):
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:   File "/usr/bin/networkd-dispatcher", line 290, in _handle_one_state
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:     self.run_hooks_for_state(new_iface, state)
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:   File "/usr/bin/networkd-dispatcher", line 328, in run_hooks_for_state
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:     'json': json.dumps(data, sort_keys=True),
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:   File "/usr/lib/python3.8/json/__init__.py", line 234, in dumps
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:     return cls(
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:   File "/usr/lib/python3.8/json/encoder.py", line 199, in encode
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:     chunks = self.iterencode(o, _one_shot=True)
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:   File "/usr/lib/python3.8/json/encoder.py", line 257, in iterencode
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:     return _iterencode(o, 0)
Nov 04 12:21:11 box.lan networkd-dispatcher[691]: TypeError: '<' not supported between instances of 'NoneType' and 'str'
Nov 04 12:21:11 box.lan networkd-dispatcher[691]: ERROR:Error handling notification for interface 'eth0' entering operational state off
Nov 04 12:21:11 box.lan networkd-dispatcher[691]: Traceback (most recent call last):
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:   File "/usr/bin/networkd-dispatcher", line 290, in _handle_one_state
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:     self.run_hooks_for_state(new_iface, state)
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:   File "/usr/bin/networkd-dispatcher", line 328, in run_hooks_for_state
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:     'json': json.dumps(data, sort_keys=True),
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:   File "/usr/lib/python3.8/json/__init__.py", line 234, in dumps
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:     return cls(
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:   File "/usr/lib/python3.8/json/encoder.py", line 199, in encode
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:     chunks = self.iterencode(o, _one_shot=True)
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:   File "/usr/lib/python3.8/json/encoder.py", line 257, in iterencode
Nov 04 12:21:11 box.lan networkd-dispatcher[691]:     return _iterencode(o, 0)
Nov 04 12:21:11 box.lan networkd-dispatcher[691]: TypeError: '<' not supported between instances of 'NoneType' and 'str'
Nov 04 12:21:11 box.lan networkd-dispatcher[691]: ERROR:Error handling notification for interface 'wlan0' entering operational state off

On a side note can iiab-diagnostics scrape the output of `iw reg get`
Anybody brave enough to try a wifi only install from the keyboard/monitor?

jvonau, 2020-11-04:
How to deal with unattended updates blocking apt?
https://github.com/iiab/iiab/pull/2512#discussion_r48605515
https://github.com/iiab/iiab/pull/2609#issuecomment-720910267

holta, 2020-11-05:
Sorry I missed the comments yesterday at the bottom of the post:

> On a side note can iiab-diagnostics scrape the output of `iw reg get`

Ok.  What 2-6 words English description should accompany that?

> Anybody brave enough to try a wifi only install from the keyboard/monitor?

@shanti-bhardwa can you consider?

jvonau, 2020-11-13:
Ubuntu-20.10 appears to have an [issue with networkd-dispatcher version 2.0.1-1](https://gitlab.com/craftyguy/networkd-dispatcher/-/issues/51) that is currently shipped, should be resolved with an update when released. Temporary workaround is to edit /etc/default/networkd-dispatcher to read just networkd_dispatcher_args="".

holta, 2020-11-13:
Ref: https://github.com/iiab/iiab/issues/2624#issuecomment-726857269 "Who can help test IIAB on Ubuntu Server 20.04.1 & 20.10 on RPi 4 ? Can WiFi firmware 7.45.18.0 work reliably?"

holta, 2020-11-15:
fyi Ubuntu Desktop 20.10's[*] graphical desktop freezes[**] after a certain number of hours (of inactivity?) such that you can no longer use the mouse and keyboard.

[*] This might not affect (a lot of?) other Ubuntu's as well, and not just on RaspberryPi!

[**] The workaround is to run `systemctl restart display-manager` from an ssh session (if one is available!)  The USB Keyboard and USB Mouse will then come back to life.

jvonau, 2020-11-18:
There is [MATE](https://releases.ubuntu-mate.org/groovy/arm64/ubuntu-mate-20.10-desktop-arm64+raspi.img.xz) desktop version that should be a lighter than the "preinstalled-desktop" version based on gnome.

holta, 2020-11-18:
> There is [MATE](https://releases.ubuntu-mate.org/groovy/arm64/ubuntu-mate-20.10-desktop-arm64+raspi.img.xz) desktop version that should be a lighter than the "preinstalled-desktop" version based on gnome.

Interesting (1.2GB download).  They describe their work here: https://ubuntu-mate.org/ports/raspberry-pi/

Is this work presumably being done/supported by ubuntu-mate.org, as a rather distinct effort/org from Canonical?  I have no idea.  But it seems Ubuntu itself is distributing their work: https://cdimages.ubuntu.com/ubuntu-mate/

FYI their community support site is here: https://ubuntu-mate.community/c/support/raspberry-pi/19

holta, 2020-11-18:
@shanti-bhardwa & @georgejhunt: would you possibly have time to try evaluating an IIAB install (http://download.iiab.io) onto this alternative Groovy Gorilla OS (["Ubuntu 20.10 + Mate"](https://releases.ubuntu-mate.org/groovy/arm64/)) that @jvonau uncovered?

georgejhunt, 2020-11-19:
The iso download is 1GB. When installed on an SD card that expanded to
3GB+. But what gets installed is a GUI netinst image which feels a little
like a regular ubuntu installer.  It takes about 30 minutes to finish the
installer. The iiab-install went without error (after the curl error)
install.txt fails because u20.10 out of the box does not have curl
installed.
apt install curl -- fixed that
I like the mint desktop version. It might be a good candidate for creation
of an image for rpi-imager -- the whole process feels pretty long.

jvonau, 2020-11-19:
Test post from ubuntu-mate-desktop-arm64 linked image above usb-booted via wifi only.  Notes: 

- quick first reboot like raspbian, that is the partition resizing
- the longer delay in the GUI is the filesystem resizing and system updating
- once complete you can login without rebooting first.

holta, 2020-11-19:
Several people are having quite good luck with Ubuntu MATE 20.10 on RPi 4.  But keep in mind this 2-step patch is still needed for now:

1) Edit `/etc/default/networkd-dispatcher` — make it read just:

   ```
   networkd_dispatcher_args=""
   ```

2) Run:

   ```
   sudo ip link set ap0 up
   ```

jvonau, 2020-11-23:
>     1. Edit `/etc/default/networkd-dispatcher` — make it read just:
>        ```
>        networkd_dispatcher_args=""
>

Needs a restart of networkd-dispacher to be effective, however the error reappears with a reboot but restarting networkd-dispacher restores a working state with the edit in place.
> 
>     2. Run:
>        ```
>        sudo ip link set ap0 up
> 

That is the foolproof method but may not persist over a client dhcp lease renewal

Nothing gets fixed unless reported upstream, the bug was found at the launch of 20.10, why do **I** have to file bugs on behalf of this organization? Want a prompt resolution of upstream issues, file bugs. Upstream may not even be aware there maybe an issue unless informed. https://bugs.launchpad.net/ubuntu/+source/networkd-dispatcher/+bug/1905283

holta, 2021-05-21:
Please test Ubuntu 21.04 instead, following up on #2624 or #2744.

closed 2021-05-21 by holta