iiab/iiab #2587: Admin Console fails to set Static IP Address: 'wan_namserver' is undefined [TK Kang's IIAB 7.2 pre-release on x86_64 Ubuntu Server 20.04.1]

holta, 2020-10-23:
> Installation went well and DHCP Internet and AP work.
Issue [arises] when I [want] to change to static address using the admin

```
TASK [Supply static WAN template when network_manager_active is False] *********
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "AnsibleUndefinedVariable: 'wan_namserver' is undefined"}

PLAY RECAP *********************************************************************
127.0.0.1 : ok=229 changed=55 unreachable=0 failed=1 skipped=173 rescued=0 ignored=0
```

iiab-diagnostics output:

http://sprunge.us/j19lOm

jvonau, 2020-10-23:
typo in https://github.com/iiab/iiab/blob/master/roles/network/templates/network/systemd-static-net.j2#L9

shanti-bhardwa, 2020-10-23:
Testing with new Ubuntu 20.10 "goriila" desktop, get Blue tooth Error: TASK [bluetooth : Disable built-in serial uart in /boot/firmware/usercfg.txt on ubuntu] **************************************************************************************************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Destination /boot/firmware/usercfg.txt does not exist !", "rc": 257}. Blue tooth is working on the desktop as I can pair my phone with PI4. Here is the diagnostics http://sprunge.us/WqHpvW

tim-moody, 2020-10-23:
This has nothing to do with Admin Console. Please change the title of the issue. The problem is an untested commit
https://github.com/iiab/iiab/commit/7d0cfc0f93dccb70dd7528d6544eaf461ed5d344#diff-4f278f09238906184aa319b2f2ab6985af0a9fb193fb209d8aa6fd9a038d7e88

Also hard to think it has anything to do with Blue Tooth.

jvonau, 2020-10-23:
> This has nothing to do with Admin Console. Please change the title of the issue. The problem is an untested commit
> [7d0cfc0#diff-4f278f09238906184aa319b2f2ab6985af0a9fb193fb209d8aa6fd9a038d7e88](https://github.com/iiab/iiab/commit/7d0cfc0f93dccb70dd7528d6544eaf461ed5d344#diff-4f278f09238906184aa319b2f2ab6985af0a9fb193fb209d8aa6fd9a038d7e88)
> 

The above commit above from May 2018 during Ubuntu's shift to netplan and the resulting juggling I contemplated not supporting static ip addressess at all given it's the user's job to create a netplan file to gain internet access prior to installing, the network code reacts to which interface has the default gateway and does not alter the user supplied files. My bad yup its a typo that just points out the usage and testing on Ubuntu since that time delaying surfacing of the typo in the commit till now.

> Also hard to think it has anything to do with Blue Tooth.

It doesn't, misplaced post.

holta, 2020-10-23:
I don't know what situations trigger this problem, FWIW I had no problems setting a Static IP Address using these instructions in 2018 and 2019:

http://FAQ.IIAB.IO > "22 How do I set a static IP address?"

tim-moody, 2020-10-23:
when: wan_ip != "dhcp" and not is_ubuntu_18 probably only happened now with ub 20.x. This used to work for me too, but probably on ub18. I'm guessing it is conditioned on being a variant of debian higher up given the name of the file.

holta, 2020-10-24:
Is there anything else people like TK need beyond PR #2588 fixing the typo?

holta, 2020-10-25:
TK responds that he's got things under control.

closed 2020-10-25 by holta