iiab/iiab #2603: transmission seeding needs sysctl settings

tim-moody, 2020-10-29:
In order to seed on ubuntu 20.04.1 transmission needs the following settings in /etc/sysctl.conf

```
net.core.rmem_max = 4194304
net.core.wmem_max = 1048576

```

Could be true of other OSes.

jvonau, 2020-10-30:
Most likely across all OS but the values may vary, are you looking at adding an option to turn an installed machine into a seeder? Or a quick howto maybe?

tim-moody, 2020-10-30:
It's @georgejhunt 's project. This is just a statement of fact (and not a question). 

>the values may vary

but not the property name? I think the values have to correspond to configuration settings in transmission for the down and up buffer sizes, but I couldn't find them.