iiab/iiab #2610: Who can help test on Ubuntu 20.10 (Server &/or Desktop version!) on x86_64 ?

holta, 2020-11-01:
Somewhat related:

- #1737 "Who can help test IIAB 7.2 on Ubuntu 20.04+, Debian 10.6+, Linux Mint 20 & Raspberry Pi ? [Kolibri 502 Bad Gateway?]"
- #2582 "Overseas request for immediate testing of IIAB on Debian 11 Bullseye + similarly clean out long-deprecated apache_config_dir from high-level <OS>.yml vars files [ASIDE: dnsmasq & dhclient issues? host_country_code MUST match PC/HW, or hostapd won't start]"
- #2585 "Testing the new Ubuntu Desktop 20.10 for RPi 4 (rumor confirmed!) [Bluetooth, MongoDB, NetworkManager]"

shanti-bhardwa, 2020-11-03:
installed 20.10 Ubuntu Groovy Gorilla desktop on PI 4 with 4GB of RAM. Desktop works fine.  This version requires installation of curl first before running the 1 line install script. for testing  IIAB 7.2.
Doing a small build gives this error first:

W: GPG error: http://ppa.launchpad.net/ansible/ansible/ubuntu bionic InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 93C4A3FD7BB9C367
E: The repository 'http://ppa.launchpad.net/ansible/ansible/ubuntu bionic InRelease' is not signed.

have to run

sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 93C4A3FD7BB9C367"

then sudo iiab again to restart.

Install reboots the desktop and doing sudo iiab completes the install

shanti-bhardwa, 2020-11-03:
http://sprunge.us/rYBwZr

holta, 2020-11-03:
@shanti-bhardwa thanks!  Noting your report is really more appropriate for #2585 !  (This ticket [#2610] is about PC and VM uses of Ubuntu 20.10)

But since we're here (:

> have to run
> 
> sudo apt-key adv --keyserver hkp://keyserver.ubuntu.com:80 --recv-keys 93C4A3FD7BB9C367"

That's (likely!) unrelated to Ubuntu Desktop 20.10 as I and @floydianslips did not have that problem (when installing onto 2 different RPi 4's).

Finally: @georgejhunt has some thoughts about Ubuntu Desktop 20.10 just not being ready for primetime, certainly as of 2020 (: https://github.com/iiab/iiab/pull/2609#issuecomment-720910267

georgejhunt, 2020-11-03:
> of 2020 (: #2609 (comment)

I'm remembering that for IIAB 7.1, I loaded Ubuntu server, and then LXDE on
top, to create an image (when we thought that only Ubuntu would support 32
clients). That combination had the look/feel of raspbian (which is also
based on LXDE) and was equally responsive. I also remember that
Tim expressed preference for Ubuntu over RaspiOS. But I think we need to
incorporate Ubuntu into the one-line installer, or just support Ubuntu as
an image, for the Rpi.

holta, 2020-11-14:
Can anybody confirm that IIAB (http://download.iiab.io) runs decently on [Ubuntu Server 20.10 and/or Ubuntu Desktop 20.10](https://releases.ubuntu.com/20.10/) on a regular PC, laptop or VM?

(i.e. Intel x86_64 / amd64 instead of Raspberry Pi!)

holta, 2020-11-14:
Tangentially Related:

Ubuntu Server 21.04 (Hirsute Hippo) Daily Builds:
https://cdimage.ubuntu.com/ubuntu-server/daily-live/current/

deldesir, 2020-11-26:
Gitea download failed on Ubuntu Desktop 20.10. The error was: socket.timeout: The read operation timeout.
Can someone confirm this?

To download it, I added "timeout: 60" in gitea install task (Download section) and runrole it.

holta, 2020-11-26:
> Gitea download failed on Ubuntu Desktop 20.10. The error was: socket.timeout: The read operation timeout.
> Can someone confirm this?
> 
> To download it, I added "timeout: 60" in gitea install task (Download section) and runrole it.

@deldesir Can you mention the specific/full download URL that failed?

And if possible the exact line number you patched within https://github.com/iiab/iiab/blob/master/roles/gitea/tasks/install.yml ?

CC: @aidan-fitz

deldesir, 2020-11-26:
I added it in Line 51 in /opt/iiab/iiab/roles/gitea/tasks/install.yml under mode: '0775'.

holta, 2020-11-26:
> TASK [Download Gitea binary https://dl.gitea.io/gitea/1.12.6/gitea-1.12.6-linux-amd64 to /library/gitea/bin/gitea-1.12.6] ***
> An exception occurred during task execution. To see the full traceback, use -vvv. The error was: socket.timeout: The read operation timed out
> fatal: [127.0.0.1]: FAILED! => {"changed": false, "elapsed": 0, "msg": "failed to create temporary content file: The read operation timed out"}

Thanks!  Is this repeatable every time?  How many times did this happen?

deldesir, 2020-11-26:
> Thanks! Is this repeatable every time? How many times did this happen?

To test it again, it would require to uninstall it and retry. It failed 3 times. I checked the link https://dl.gitea.io/gitea/1.12.6/gitea-1.12.6-linux-amd64 and it was ok for a manual download.

holta, 2020-11-26:
> It failed 3 times. I checked the link https://dl.gitea.io/gitea/1.12.6/gitea-1.12.6-linux-amd64 and it was ok for a manual download.

Thanks for the information!

That's enough info for others (e.g. @aidan-fitz, who wrote this code) to keep an eye on.

Note: this download occurred over a 3G/4G mobile connection, which is essentially the only option in Haiti here.

fisherbryan, 2021-02-19:
Hey Guys, I have been trying to install iiab on a digital ocean droplet since Tuesday this week and still trying...
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.2 LTS
Release:	20.04
Codename:	focal
Please;
http://sprunge.us/A0BCbs

Your assistance will be greatly appreciated please. Thanks so much.

georgejhunt, 2021-02-19:
I remember installing IIAB on a droplet.  You will need to disable the
network role, and deal with connectivity yourself. But we do have success
on U20.04.
I remember that I lost control, because I just let-r rip without disabling
network role, and needed to use the vnc console from the droplet admin page
to re-enable connectivity.
As I remember it installed very fast.

holta, 2021-02-19:
> You will need to disable the network role, and deal with connectivity yourself.

@georgejhunt are you saying to comment out the last 3 lines of https://github.com/iiab/iiab/blob/master/iiab-stages.yml#L63-L65 ?

Note the errors at the bottom of http://sprunge.us/A0BCbs?en are very serious here: (Lines 1490-1493, suggesting this crashed due to lack of RAM/memory long before Ansible reached IIAB's [network role](https://github.com/iiab/iiab/tree/master/roles/network) !?)

```
Feb 18 23:48:32 box vmunix: [16897.754937] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/system.slice/gitea.service,task=gitea-1.13.2,pid=14008,uid=1002
Feb 18 23:48:32 box vmunix: [16897.754959] Out of memory: Killed process 14008 (gitea-1.13.2) total-vm:1523300kB, anon-rss:133280kB, file-rss:0kB, shmem-rss:0kB, UID:1002 pgtables:684kB oom_score_adj:0
Feb 18 23:48:32 box vmunix: [16897.769298] oom_reaper: reaped process 14008 (gitea-1.13.2), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB
Feb 18 23:49:06 box vmunix: [16932.152597] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.
```

cc: @jvonau, @tim-moody

holta, 2021-02-19:
Side Question for @fisherbryan:

This ticket was created to test IIAB with Ubuntu 20.10 not 20.04 -- is there any particular reason you chose this ticket in particular to post to (rather than opening a new ticket?)

holta, 2021-02-19:
> Out of memory: Killed process

@fisherbryan the error at the bottom of your [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) output (Line 1491 of http://sprunge.us/A0BCbs?en) confirms that you are trying to install BIG-sized IIAB (Line 193) with insufficient memory (Line 778 shows only 1GB RAM).

Why are you installing BIG-sized?  I would suggest customizing your `/etc/iiab/local_vars.yml` immediately after you select a MEDIUM-sized IIAB install.

In this way you can add the essentials (e.g. Kolibri and/or Moodle?) without "polluting" (i.e. overwhelming) your installation with many unnecessary services -- overtaxing your machine's resource.

PS @georgejhunt's recommendation might well still/also apply!?

jvonau, 2021-02-19:
>COMMAND: /usr/sbin/ip route    # Routing table
default via 138.197.176.1 dev eth0 proto static 
10.19.0.0/16 dev eth0 proto kernel scope link src 10.19.0.5 
10.114.0.0/20 dev eth1 proto kernel scope link src 10.114.0.2 
138.197.176.0/20 dev eth0 proto kernel scope link src 138.197.182.215 

The eth1 device could be problematic without a gateway present, along with the whole br0 infrastructure to support to the WiFi hotspot. As an alternative to completely disabling the network role one could actually use the built-in options from default_vars.yml but are absent in local_vars.yml try adding:
iiab_lan_enabled:  False
or
user_lan_iface: none
or
is_VM: True

holta, 2021-02-19:
@fisherbryan I suppose I'd go with @jvonau's top recommendation (of the 3 choices).

In any case, here's some quick background on all 3 of these variables:

> iiab_lan_enabled: False

...instead of `True` by default:
https://github.com/iiab/iiab/blob/master/vars/default_vars.yml#L130

> user_lan_iface: none

...instead of `auto` by default:
https://github.com/iiab/iiab/blob/master/vars/default_vars.yml#L191-L194

> is_VM: True

...instead of being undefined by default.  If defined (e.g. in [/etc/iiab/local_vars.yml](http://wiki.laptop.org/go/IIAB/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it.3F)) it will be acted upon here:
https://github.com/iiab/iiab/blob/master/roles/network/tasks/detected_network.yml#L209-L213

jvonau, 2021-02-19:
There is reserved_device that also could be used as this variable should be subtracted from the list of possible LAN devices, that should disable br0 from being created also. Now to deal with iptables, the stock detection with any of the 4 options above would write IIAB_LAN_DEVICE=none into iiab.env where iiab-gen-iptables would open up the interface (--state NEW -i $lan). Once installed one could manually set IIAB_LAN_DEVICE=eth1 and run `sudo iiab-gen-iptables` to pick up the new setting. I might entertain extending reserved_device to cover this situation automatically.

holta, 2021-02-19:
@fisherbryan, if you have time also take a look at the ongoing design question about `reserved_device` and `reserved_wifi` at PR #2682.

holta, 2021-02-24:
Thanks to @jvonau who solved this very serious problem with Raspberry Pi's WiFi internal hotspot &mdash; apparently resulting from the recent RaspiOS apt update from Linux kernel 5.4 to 5.10:

PR #2686

Thanks also to @fisherbryan and @tim-moody for helping to discover the problem, and for noticing that a temporary workaround for all these problems was to run `systemctl restart hostapd`.

holta, 2021-02-25:
> comment out the last 3 lines of https://github.com/iiab/iiab/blob/master/iiab-stages.yml#L63-L65

@georgejhunt confirms this strategy worked on DigitalOcean for him!  @jvonau confirms that may well be the easiest way on DigitalOcean-like VM/cloud environments for now.

jvonau, 2021-02-25:
[ref](https://www.reddit.com/r/pihole/comments/g86xo8/after_ubuntu_2004_upgrade_unable_to_start_dns/) dnsmasq error see: https://github.com/iiab/iiab/issues/1737#issuecomment-702648023

holta, 2021-04-28:
A BIG-sized IIAB install just completed on the newer 2021-04-09 **64-bit** RaspiOS Lite on RPi 4 at 10.8.0.38

FWIW I used the released version of ansibe-core 2.11.0 installing as follows:

```
curl d.iiab.io/install.txt | sudo bash -s 2743
```

holta, 2021-05-21:
Please test Ubuntu 21.04 instead, following up on #2624 or #2744.

closed 2021-05-21 by holta