iiab/iiab #2612: Moodle doesn't install cleanly

jvonau, 2020-11-02:
### Expected behavior and actual behavior:
Working moodle

> TASK [Install /library/pgsql-iiab/postgresql.conf owned by postgres:postgres, from template] **********
changed: [127.0.0.1]
TASK [Disable & Stop stock 'postgresql' (parent) systemd service] *************************************
changed: [127.0.0.1]
TASK [Set 'postgresql_installed: True'] ***************************************************************
ok: [127.0.0.1]
TASK [Add 'postgresql_installed: True' to /etc/iiab/iiab_state.yml] ***********************************
changed: [127.0.0.1]
TASK [Enable & Start 'postgresql-iiab' systemd service, if postgresql_enabled] ************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Unable to start service postgresql-iiab: Job for postgresql-iiab.service failed because the control process exited with error code.\nSee \"systemctl status postgresql-iiab.service\" and \"journalctl -xe\" for details.\n"}
PLAY RECAP ********************************************************************************************
127.0.0.1                  : ok=55   changed=16   unreachable=0    failed=1    skipped=14   rescued=0    ignored=0   

### Steps to reproduce the problem:
enable moodle  in local_vars.yml  
sudo ./runrole moodle
### IIAB version, OS version, hardware details:
RaspOS-10 server
### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)
Doesn't provide the information needed
>pi@box:/opt/iiab/iiab $ journalctl  -u postgresql-iiab -l --no-pager > ~/dump.txt
pi@box:/opt/iiab/iiab $ cat  ~/dump.txt
-- Logs begin at Thu 2019-02-14 10:11:59 GMT, end at Mon 2020-11-02 12:59:09 GMT. --
Nov 02 12:55:40 box.lan systemd[1]: Starting PostgreSQL database server...
Nov 02 12:55:40 box.lan pg_ctl[13135]: 2020-11-02 12:55:40.949 GMT [13138] LOG:  could not bind IPv4 address "127.0.0.1": Address already in use
Nov 02 12:55:40 box.lan pg_ctl[13135]: 2020-11-02 12:55:40.949 GMT [13138] HINT:  Is another postmaster already running on port 5432? If not, wait a few seconds and retry.
Nov 02 12:55:40 box.lan pg_ctl[13135]: 2020-11-02 12:55:40.951 GMT [13138] LOG:  could not bind IPv6 address "::1": Cannot assign requested address
Nov 02 12:55:40 box.lan pg_ctl[13135]: 2020-11-02 12:55:40.951 GMT [13138] HINT:  Is another postmaster already running on port 5432? If not, wait a few seconds and retry.
Nov 02 12:55:40 box.lan pg_ctl[13135]: 2020-11-02 12:55:40.951 GMT [13138] WARNING:  could not create listen socket for "localhost"
Nov 02 12:55:40 box.lan pg_ctl[13135]: 2020-11-02 12:55:40.951 GMT [13138] FATAL:  could not create any TCP/IP sockets
Nov 02 12:55:40 box.lan pg_ctl[13135]: 2020-11-02 12:55:40.951 GMT [13138] LOG:  database system is shut down
Nov 02 12:55:40 box.lan pg_ctl[13135]: pg_ctl: could not start server
Nov 02 12:55:40 box.lan pg_ctl[13135]: Examine the log output.
Nov 02 12:55:40 box.lan systemd[1]: postgresql-iiab.service: Control process exited, code=exited, status=1/FAILURE
Nov 02 12:55:40 box.lan systemd[1]: postgresql-iiab.service: Failed with result 'exit-code'.
Nov 02 12:55:40 box.lan systemd[1]: Failed to start PostgreSQL database server.
pi@box:/opt/iiab/iiab $ sudo netstat -natp
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      1188/nginx: master  
tcp        0      0 127.0.0.1:53            0.0.0.0:*               LISTEN      979/dnsmasq         
tcp        0      0 172.18.96.1:53          0.0.0.0:*               LISTEN      979/dnsmasq         
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      983/sshd            
tcp        0      0 0.0.0.0:3000            0.0.0.0:*               LISTEN      5942/kiwix-serve    
tcp        0      0 127.0.0.1:8090          0.0.0.0:*               LISTEN      1079/apache2        
tcp        0      0 0.0.0.0:8008            0.0.0.0:*               LISTEN      1249/python2.7      
tcp        0      0 127.0.0.1:27017         0.0.0.0:*               LISTEN      4204/mongod         
tcp        0      0 127.0.0.1:3306          0.0.0.0:*               LISTEN      976/mysqld          
tcp        0     92 172.18.96.1:22          172.18.121.236:52056    ESTABLISHED 1353/sshd: pi [priv 
tcp6       0      0 :::22                   :::*                    LISTEN      983/sshd            

I suspect the original postmaster process was slow to shut down.

holta, 2020-11-03:
Is this bug likely arising after (but not during) IIAB installations?

Do we have an idea which OS's and/or which HW are most affected and/or always affected?

(Conversely which OS's / which HW is most immune?)

jvonau, 2020-11-03:
> Is this bug likely arising after (but not during) IIAB installations?
>

reported the use of runrole above
> Do we have an idea which OS's and/or which HW are most affected and/or always affected?
> 

reported RaspiOS-32bit server above.
> (Conversely which OS's / which HW appear most immune?)

Hardware is Rpi4B+ with 4GB ram.

jvonau, 2020-11-05:
Systemclt has '--no-block Do not wait until operation finished' while https://docs.ansible.com/ansible/2.10/collections/ansible/builtin/systemd_module.html 'no-block no|yes Do not synchronously wait for the requested operation to finish. Enqueued job will continue without Ansible blocking on its completion'  Badly named option in ansible for 'no_block' IMHO but looks like adding no_block: yes to https://github.com/iiab/iiab/blob/master/roles/postgresql/tasks/install.yml#L71-L75 should cause ansible to wait while the task is completed before moving to the next task.

holta, 2020-11-05:
> Systemclt has '--no-block Do not wait until operation finished' while https://docs.ansible.com/ansible/2.10/collections/ansible/builtin/systemd_module.html 'no-block no|yes Do not synchronously wait [...]

Thanks @jvonau and @georgejhunt if you can propose a solution?

holta, 2021-02-01:
Seem very rare.  Possibly we should close this?

jvonau, 2021-02-06:
Could show up during an install, I just hit this using runrole. If I recall correctly it's actually a race condition caused by postgresql.service unit file layout and ansible not waiting for the port to be fully released while postgresql@"version-cluster" service was taking its time to shut down before the postgresql-iiab service was started. 2020-10-28 https://github.com/iiab/iiab/commit/6a5536001779151e8f3e68b27d1063100ce53a88 aggravated the situation as there was an explicit call to the "child" "postgresql@version-cluster", the 2 stanzas removed may not of really done anything other than introduce a delay allowing the first service to fully stop before starting the replacement service.

holta, 2021-07-22:
Close?

If this still happens, it's extremely rare.

jvonau, 2021-07-27:
You could close if you want, 'no_block: yes' should be employed anywhere the underling service might takes awhile to shutdown to avoid a noted race condition.

holta, 2021-07-27:
> You could close if you want, 'no_block: yes' should be employed anywhere the underling service might takes awhile to shutdown to avoid a noted race condition.

Where should `no_block: yes` be considered?

jvonau, 2021-07-27:
> > You could close if you want, 'no_block: yes' should be employed anywhere the underling service might takes awhile to shutdown to avoid a noted race condition.
> 
> Where should `no_block: yes` be considered?

I'd say anywhere a service is being shutdown because another service of a different name is going to start using the same port.

jvonau, 2025-04-06:
'no_block: yes' would of been interesting to pursue with #3965

closed 2025-04-06 by jvonau