iiab/iiab #2614: Live content on external storage -- "create the symbolic link(s) you need" is pretty user-unfriendly.

myself248, 2020-11-03:
Expected: I'd like to plug in USB sticks as storage for content. Keep the OS.

Actual: I can move content on and off USB sticks, but not keep my live ZIMs or KA content there, for instance. At least not with the admin console.

I found this: http://wiki.laptop.org/go/IIAB/FAQ#Can_I_permanently_attach_an_external_USB_drive.2C_to_add_more_content.3F 

But it's a little intimidating, to say the least. (How do I know what links I need?) Can this be in the admin console somehow? Ideally, I'd just like a checkbox for "Make the content on usb0 part of the live system". (And usb1, and usb2, and.... because no single stick is large enough for all the content I want.) 

Also, be able to download things directly to USB -- my SD card isn't large enough for a single map pack, so I need to put those straight onto the USB. Already discussed in #2205 to some extent.

myself248, 2020-11-03:
Related to:
#828 
#857

holta, 2020-11-03:
Thank you @myself248

There's a lot to digest here.  To get this right.

If you have time Thursday 10AM NYC Time (EST) please do join our IIAB weekly community/design call:

http://minutes.iiab.io

(Ideally prior to the call email me [holt @ unleashkids DOT org ] your Skype username if you have one, but if not let's keep in touch in others ways, to get this progressively cleaner across the many diverse IIAB implementation scenarios that so many different folks around the world are all attempting!)

holta, 2020-11-04:
@shanti-bhardwa just posted this somewhat related result:

https://github.com/iiab/iiab/issues/2585#issuecomment-721711281

holta, 2020-11-05:
Huge Thanks to @myself248 for joining IIAB's weekly call earlier today — with @jvonau, @georgejhunt, @tim-moody and others.

Quick summary as excerpted from http://minutes.iiab.io :

> i. FAQ.IIAB.IO > “43 Can I permanently attach an external USB drive, to add more content?”
> ii. NB: UnionFS / overlayfs / similar approaches?
>   1. A checkbox for “newly downloaded content goes here” could switch which FS is mounted write-thru, the others are read-only.
>
> iii. JV: bind-mount used by Snaps, so Content Packs can be elsewhere and accessed from within webroot. SquashFS can be used to reduce the storage space as an extension of using bind-mounting.
> iv. Goal is content on multiple removable devices
>   1. e.g. Auto-display of ZIM files when Teachers insert USB sticks/drives  #857
>     a. Running ‘iiab-make-kiwix-lib’ is fast (much less than a minute?) so could we consider that on auto-mount possibly?
>   2. It’s been pointed out that pure-html content simply appears on http://box/usb & works seamlessly already, so changes are only needed to support zim/osm/etc content.

jvonau, 2020-11-06:
@myself248 I might have an alpha quality solution to try out if you are interested.

myself248, 2020-11-06:
@jvonau Absolutely, that's probably still better quality than anything I'd cook up. How's it work?

jvonau, 2020-11-07:
With 'iii' above, pass me your contact info via email and we can have a quick chat on how it works and what you can and can't do with this approach.

jvonau, 2020-11-08:
Good related info in #2528

holta, 2020-11-08:
> Related to:
> #828 Copy Content Packs to/from USB sticks/storage
> #857 Auto-display of ZIM files when Teachers insert USB sticks/drives

Tangentially Related:
- #2626 "Copying zims manually without running iiab-make-kiwix-lib will cause Admin Console installation to fail (and kiwix not to work)"
- PR #2627 Kiwix [role improvements for pre-positioning of ZIM files into /library/zims/content, etc]