iiab/iiab #2618: Satellite downloading several regions at once fails, sqlite3 database locked

myself248, 2020-11-05:
### Expected behavior and actual behavior:

Expected: Click a region, click "add satellite photos", click another region, click "add satellite photos", repeat until I've inhaled all the satellite imagery I'm interested in. 

Actual: The first region succeeds, but any subsequent downloads started while the first one is still downloading, fail with:

File "/usr/bin/iiab-extend-sat.py", line 770, in do_downloads
download_tiles(src,args.lat,args.lon,zoom,args.radius)
File "/usr/bin/iiab-extend-sat.py", line 713, in download_tiles
replace_tile(src,zoom,tileX,tileY)
File "/usr/bin/iiab-extend-sat.py", line 684, in replace_tile
mbTiles.SetTile(zoom, tileX, tileY, r.data)
File "/usr/bin/iiab-extend-sat.py", line 256, in SetTile
self.conn.commit()
sqlite3.OperationalError: database is locked

### Steps to reproduce the problem:

As above.

### IIAB version, OS version, hardware details:

commit 05b61e867291256d7c3c70a60fa1722acfd18237 
Raspbian 10 on a Pi 4B with 2GB RAM, on a 128GB microSD card

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

http://sprunge.us/090UU0

tim-moody, 2020-11-05:
if @georgejhunt is willing to make iiab-extend-sat.py support db concurrency that would probably solve this, otherwise I can try to make multiple downloads single threaded by making the start of each download dependent on the completion of the previous in Admin Console.

holta, 2020-11-05:
Huge Thanks to @myself248 for joining IIAB's weekly call earlier today &mdash; with @tim-moody, @georgejhunt @jvonau and others.

Quick summary as excerpted from http://minutes.iiab.io :

> i. Possible approaches: 
>  1. Coalesce regions into a single download (hard, saves on bandwidth/time by not double-fetching the overlaps)
>  2. Make simultaneous downloads interleave (likely to still double-fetch tiles in the overlapping regions)
>  3. Make downloads single-threaded (less hard)
>    a. check if tile exists before fetching (saves time/bandwidth)

holta, 2020-11-12:
@myself248 do you have time to retest this now that @georgejhunt's PR iiab/maps#42 has been merged?

(If so start easy, with just 2 simultaneous downloads of e.g. overlapping Satellite Photo squares?!)

jvonau, 2020-11-13:
How does somebody back fit the fix? runrole --reinstall?

georgejhunt, 2020-11-13:
I think ./runroles osm-vector-maps will test the new code