iiab/iiab #2625: http://box/modules/en-algebra2go/ won't load over WiFi

holta, 2020-11-07:
...but it _can_ be loaded over any Ethernet connection to the same IIAB's.

Restated: Full URL http://box/modules/en-algebra2go/pagefiles/ does not load over WiFi connections to any IIAB...but it _does_ load over Ethernet connections to these same IIAB's (built on various 64-bit and 32-bit OS's).

FYI the above page _should_ be equivalent to this online page:
http://oer2go.org/mods/en-algebra2go/prealgebra/index.html

- What Happens?  The Chrome browser permanently stalls with a green background and permanently stuck progress bar (showing about "2/3" of the page downloaded).  Presumably the intro page to this RACHEL/OER2Go module is just broken whenever truly offline folk try to access in their IIAB ??
- Can anybody tell if it's attempting to download some graphic-or-javascript-or-some-such that's live on the Internet...or (presumably) some such thing in the page...that always blocks it from appearing every time...when you try to access it from the LAN(offline)-not-WAN(online) side of these IIAB's...in the Chrome browser anyway ?

tim-moody, 2020-11-07:
can you check the javascript console for errors. hard to imagine wifi getting different results than wired except for speed or reliability

so the supposition is that if a wired connection is used, the user has access to the internet?

jvonau, 2020-11-07:
connection the the source page
[image]

tim-moody, 2020-11-07:
so a file that is missing online only causes a problem when you are offline?

tim-moody, 2020-11-07:
on the demo machine modules/en-algebra2go/pagefiles/
yields

```
(index):3 GET https://www.saddleback.edu/INC/mm_menu.js net::ERR_ABORTED 404 (Not Found)
(index):5 Uncaught ReferenceError: mmLoadMenus is not defined
    at (index):5

```

don't know when it was last updated

holta, 2020-11-07:
Should I try a quick recursive search+replace across all of `/library/www/html/modules/en-algebra2go` so every online URL (e.g. those that contain saddleback.edu ?) are auto-replaced with...something else...more harmless?

p.s. some say Algebra2Go might not be the greatest content every created, as PDF's really aren't the best way to learn in the 2020's, but still these might still have some hope+purpose possibly?

holta, 2020-11-07:
@floydianslips you understand JavaScript and how to hack it way better than I do...can you possibly run the following on any IIAB:

```
cd /library/www/html/modules
rsync -Pavz rsync://dev.worldpossible.org/rachelmods/en-algebra2go ./
```

If so...10min later you can probably see exactly why http://box/modules/en-alegbra2go/ fails to load...when you access it from the OFFLINE SIDE (LAN side) of IIAB ??

(i.e. connect to IIAB using a regular student WiFi device, and be sure there's no internet access, making sure `iiab_gateway_enabled: False
` is set in [/etc/iiab/local_vars.yml](http://wiki.laptop.org/go/IIAB/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it.3F) !)

So we know how to quick patch up this RACHEL/OER2Go module ideally~

holta, 2020-11-10:
@darkenvy can you tell why the page http://oer2go.org/mods/en-algebra2go/prealgebra/index.html fails to load when it's offline?

Is this an easy fix or not?

The entire 1.2GB package can be downloaded over {http, ftp or rsync} here if nec: http://oer2go.org/viewmod/en-algebra2go

jvonau, 2022-04-05:
is this still relevant? seems like an upstream bug

holta, 2022-04-05:
> is this still relevant? seems like an upstream bug

Good question.

It seems folks inevitably end up trying out this Content Pack, and unhappy.

IIAB has a role in (trying!) to help people quickly/efficient navigate these kinds of messes &mdash; somehow anyway, possibly with augmented/amplified Content Pack metadata etc?