iiab/iiab #2626: Admin Console failed to install on 32-bit RaspiOS with desktop

holta, 2020-11-08:
@tim-moody did this Admin Console fresh install fail possibly because I pre-positioned a few ZIM files into `/library/zims/content` ?

(Is pre-positioning ZIM files not allowed?)

FYI I'm trying to install IIAB (http://download.iiab.io) onto 32-bit 'RaspiOS with desktop' on RPi 4 with 4GB RAM:

> TASK [js-menu : Add any IIAB roles that are installed to home/menu.js] *********
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "/opt/admin/cmdsrv/scripts/iiab_update_menus.py", "delta": "0:00:00.921648", "end": "2020-11-07 21:35:55.677129", "msg": "non-zero return code", "rc": 1, "start": "2020-11-07 21:35:54.755481", "stderr": "Traceback (most recent call last):\n  File \"/opt/admin/cmdsrv/scripts/iiab_update_menus.py\", line 26, in <module>\n    main()\n  File \"/opt/admin/cmdsrv/scripts/iiab_update_menus.py\", line 18, in main\n    adm.put_kiwix_enabled_into_menu_json()\n  File \"/usr/lib/python3/dist-packages/iiab/adm_lib.py\", line 522, in put_kiwix_enabled_into_menu_json\n    write_zim_versions_idx(zim_versions, iiab.CONST.kiwix_library_xml, CONST.zim_version_idx_dir, zim_menu_defs)\n  File \"/usr/lib/python3/dist-packages/iiab/adm_lib.py\", line 38, in write_zim_versions_idx\n    get_substitution_data(perma_ref, zim_versions, zims_installed, path_to_id_map)\n  File \"/usr/lib/python3/dist-packages/iiab/adm_lib.py\", line 97, in get_substitution_data\n    zim_id = path_to_id_map[path]\nKeyError: 'content/crashcourse_en_all_2020-05.zim'", "stderr_lines": ["Traceback (most recent call last):", "  File \"/opt/admin/cmdsrv/scripts/iiab_update_menus.py\", line 26, in <module>", "    main()", "  File \"/opt/admin/cmdsrv/scripts/iiab_update_menus.py\", line 18, in main", "    adm.put_kiwix_enabled_into_menu_json()", "  File \"/usr/lib/python3/dist-packages/iiab/adm_lib.py\", line 522, in put_kiwix_enabled_into_menu_json", "    write_zim_versions_idx(zim_versions, iiab.CONST.kiwix_library_xml, CONST.zim_version_idx_dir, zim_menu_defs)", "  File \"/usr/lib/python3/dist-packages/iiab/adm_lib.py\", line 38, in write_zim_versions_idx", "    get_substitution_data(perma_ref, zim_versions, zims_installed, path_to_id_map)", "  File \"/usr/lib/python3/dist-packages/iiab/adm_lib.py\", line 97, in get_substitution_data", "    zim_id = path_to_id_map[path]", "KeyError: 'content/crashcourse_en_all_2020-05.zim'"], "stdout": "Updating kiwix menus", "stdout_lines": ["Updating kiwix menus"]}
>
> PLAY RECAP *********************************************************************
> 127.0.0.1                  : ok=69   changed=60   unreachable=0    failed=1    skipped=4    rescued=0    ignored=0

holta, 2020-11-08:
The workaround seems to be to run `iiab-make-kiwix-lib` in advance of installing Admin Console.

Should this be fixed?  (Or is this not worth fixing / not easy to fix?)

tim-moody, 2020-11-08:
Once upon a time iiab-make-kiwix-lib was part of install or I considered making it so, but didn't want cost the usual install more time to have that run. I think the answer is to trap and report this error caused by zims and library.xml being out of sync.

holta, 2020-11-08:
This problem should now be mitigated by PR iiab/iiab-admin-console#363 "run iiab-make-kiwix-lib in case zims have been added manually"

PR #2627 might also be very useful as general cleanup.

holta, 2020-11-08:
See Also:

- PR iiab/iiab-admin-console#364 "Give error message and exit if iiab-make-kiwix-lib needs to run"

PR #2629 (2020-11-09): Kiwix role: fix counting of /library/zims/content/*.zim + in-line clarifs
Fixes #2626, building on @jvonau's PR #2627

Heavily tested with and without *.zim files in /library/zims/content
closed 2020-11-09 by holta