iiab/iiab #2628: Re-scrape 'Scale of the Universe' now that's it's JavaScript instead of .swf ?

holta, 2020-11-09:
@darkenvy can you take a quick look as to how easy/hard it is to scrape http://htwins.net/scale2/ ("sotu 2.1") for offline use?

Their rewrite (JavaScript instead of Flash?) should now work in all browsers, which is great as this single-page is urgently demanded/desired by most all countries/schools worldwide.

As background, its famous .swf (Flash-based) predecessor is here: http://oer2go.org/viewmod/en-scale-of-universe

PS the English version would certainly be sufficient for starters.  And if that turns out to be easy, the Spanish and French (if not a few others) would most certainly also be used heavily heavily too later !

shanti-bhardwa, 2020-11-09:
not clear how I can install JavaScript version?. the .swf fom old version plays fine if you have off-line flash-palyer on your desktop

darkenvy, 2020-11-09:
Yea, I will give a shot ripping it. I'm pretty optimistic that it shouldn't be too hard to offline the whole thing.

holta, 2020-11-10:
@darkenvy & @floydianslips should both get Nobel Prizes, for solving very practical front-end needs faced by people everywhere.

See "Scale of the Universe" halfway down http://iiab.me/jamaica in all 20 languages.

Published for all here: http://d.iiab.io/content/SofU.zip (14.3 MB)

holta, 2020-11-10:
Another JavaScript content glitch is here, if anyone has the answer?

"http://box/modules/en-algebra2go/ won't load over WiFi, i.e. when offline (LAN-side of IIAB)"
https://github.com/iiab/iiab/issues/2625

closed 2020-11-10 by holta

holta, 2020-11-10:
FYI an Internet-in-a-Box "menu def" [*] for @darkenvy's new scraping of "Scale of the Universe 2.1" is here:
https://github.com/iiab-share/js-menu-files/blob/master/menu-defs/en-scale-of-universe-2.json

[*] A "menu def" is a description and logo, for the main page of any IIAB, as summarized here:
https://github.com/iiab/iiab/wiki/IIAB-Menuing

holta, 2020-11-10:
IIAB Catalog http://med.iiab.me/catalog/iiab_catalog.json is an "auxiliary" catalog useful offline Content Packs that Internet-in-a-Box volunteers have helped make beautiful...to supplement other "offline content catalogs" like:

- http://oer2go.org
- http://download.kiwix.org/zims
- https://catalog.learningequality.org/
- Etc!

To help push this forward, @tim-moody has published a "reference copy" of @darkenvy's scrape of "sotu 2.1" to help with wider distribution:
http://med.iiab.me/modules/en-scale-of-universe-2/

And I've put a symlink here in case that helps more people find/understand/use Cary Huang & Matthew Martori great work here:
http://d.iiab.io/content/en-scale-of-universe-2.zip

holta, 2020-11-10:
This is now available for everyone to install (non-technical IIAB implementers included) using Admin Console > Install Content > Get OER2Go(RACHEL) Modules !

Thanks to @tim-moody's:

- https://github.com/iiab/iiab-admin-console/pull/368 "Add new scale of univ to catalog"
- https://github.com/iiab/iiab-admin-console/blob/master/assets/iiab_catalog.json

holta, 2020-11-11:
BAD NEWS FIRST: http://iiab.me/modules/en-scale-of-universe-2 warns you not to run it on smartphones :/

GOOD NEWS NEXT: if you force your mobile browser to request "Desktop site" [*] then Scale of the Universe 2 **works great,** in landscape mode especially!

[*] e.g. Android's Chrome browser lets you check a box ("Desktop site") on the top right, after clicking on the vertical ellipsis.  Firefox on Android similarly.

holta, 2020-11-11:
This mobile trick is now documented for all in [en-scale-of-universe-2.json](https://github.com/iiab-share/js-menu-files/blob/master/menu-defs/en-scale-of-universe-2.json#L10) here:

```
  "extra_description": "On a mobile device, use your browser settings (e.g. touch <b>&vellip;</b> in the top-right) to check the box for &quot;Desktop site&quot;.",
```

As visible halfway down the page if you do a hard-reload of http://iiab.me/jamaica

darkenvy, 2020-12-18:
Yea, it might be the best solution. The "Desktop Site" button expands the viewport in the browser. 
Once you get into the app, everything is drawn in canvas. So it's not easy to make modifications to make it friendly. I see why they made an iOS application.

Good solution @holta for now :)

matttt, 2022-04-17:
Hey all! I found this thread by chance when I happened to look up my name for other reasons 😅 I built the new version of Scale of the Universe and if there is anything I can do to help, like build a special version that disables the mobile warning or hardcodes a specific language I can definitely do so! This is so freaking cool.

tim-moody, 2022-04-25:
@matttt any chance it can be straight up html5, css, js and not need a running node app?