iiab/iiab #2632: Install Issue - Pi4-8g with SSD drive

JamesM92, 2020-11-11:
### Expected behavior and actual behavior:
Full install having an error

### Steps to reproduce the problem:
curl d.iiab.io/install.txt | suda bash
specifying type 3 install, 
installation went through completely, had an error at the end of the install stating that MongoDB didn't install properly.
I tried installing MongoDB manually and re-running but still getting the same error.

### IIAB version, OS version, hardware details:
IIAB:7.2
Raspberry Pi OS (32-bit) Lite : 2020-08-20
PI4-8g: Running off of USB ssd 2tb

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)
http://sprunge.us/Sh2YgA

holta, 2020-11-11:
> Command "/usr/bin/unzip" could not handle archive.
> Command "/bin/tar" could not handle archive."}

Are those 2 programs not installed out of curiosity?

What's the result if you run...?

```
ls -l /usr/bin/unzip /bin/tar
```

PS https://github.com/iiab/iiab/blob/master/roles/mongodb/tasks/install.yml#L28-L32 is the code in question!

holta, 2020-11-11:
Also, what's the result if you run...?

```
which unzip
which tar
```

holta, 2020-11-11:
Also, what's the result if you run...?

```
apt list unzip tar
```

JamesM92, 2020-11-11:
pi@raspberrypi:~ $ ls -l /usr/bin/unzip /bin/tar
-rwxr-xr-x 1 root root 352280 Apr 23  2019 /bin/tar
-rwxr-xr-x 2 root root 145240 Jul 30  2019 /usr/bin/unzip

pi@raspberrypi:~ $ which unzip tar
/usr/bin/unzip
/bin/tar

pi@raspberrypi:~ $ apt list unzip tar
Listing... Done
tar/stable,now 1.30+dfsg-6 armhf [installed]
unzip/stable,now 6.0-23+deb10u1 armhf [installed]

Also, I switched to using a microsd card, to see if the ssd was causing the issue. But i got the same error.

holta, 2020-11-11:
Thanks @JamesM92 this is extremely strange!

Can you also confirm that `/usr/bin/unzip` is working otherwise, to unzip other files in general?

> I tried installing MongoDB manually and re-running but still getting the same error.

Since you mention it, what approach did you try?

holta, 2020-11-12:
@JamesM92 I noticed you ran the exact same unzip code successfully, just 9 lines above: (as can be seen from the bottom of your log files at http://sprunge.us/Sh2YgA?en)

https://github.com/iiab/iiab/blob/master/roles/mongodb/tasks/install.yml#L19-L23

I tried to reproduce your issue on a 2GB RAM RPi 4 (also on 32-bit Raspberry Pi OS Lite) but could not &mdash; the following commands worked without any problem for me:

```
sudo su -
cd /opt/iiab/iiab
./runrole mongodb
./runrole --reinstall mongodb
```

Do you get the exact same failure(s) when you run the commands above?

In my case they work perfectly, as confirmed by:

```
pi@box:~ $ mongod --version
db version v3.0.14
git version: 08352afcca24bfc145240a0fac9d28b978ab77f3
```

holta, 2020-11-12:
Finally @JamesM92 if you you cannot figure this out (why your machine is different from others) please consider:

1. Allowing an IIAB engineer to make a remote connection to your machine to investigate and/or
2. Joining our 10AM EST / NYC Time Thursday (weekly) community call to discuss this with other engineers if you can?  Simply add your question(s) to the agenda here: http://minutes.iiab.io

JamesM92, 2020-11-12:
I've re-imaged the computer, I'll try and do a fresh install this weekend. and try your suggestions.

@holta  I tried installing mongodb using the : sudo apt-get install mongodb
and it installed fine but when i re-ran : sudo iiab 
i still got the same issue.

For reference I have gotten IIAB 1 line installer to install fine on a Pi4-2gb i have but the last time i tried it was months ago. So maybe the 8gb version is causing problems? Or I did change some settings on the Pi's EEPROM to try and boot from usb. But i got the same error even when on microsd.
(Like said im playing with this one to use a 2tb ssd instead of the micro sd card, plan to download EVERYTHING on it lol)

holta, 2020-11-12:
> I tried installing mongodb using the : sudo apt-get install mongodb

@JamesM92: Is it wise to run `apt purge mongodb` to help clean up, if the above ran?  Noting that Sugarizer (and hence IIAB) require a substantially more modern version of MongoDB than the OS (`apt`) can generally provide.

Please run these 3 commands on any afflicted IIAB &mdash; which should not take more than 1-2 minutes:

```
sudo su -
cd /opt/iiab/iiab
./runrole mongodb
```

Forwarding any error details that result!

Thanks again for pushing to get to the bottom of this (highly mysterious!) failure to unzip.

> I'm fine with opening it up for remote access if needed,

Thanks: please email holt @ unleashkids [dot] org to arrange that when you get a chance.

JamesM92, 2020-11-13:
I may have forgotten to mention that i was doing my install on wifi previously. Bot sure if that may have anything to do with it.

So i tried a new install, this time i used

Raspberry Pi OS (32bit) Lite : 2020-05-27

This time i got errors again during install, but different errors, each time i ran the "sudo iiab" it would run then give an error it wasn't able to perform _____ tasks install. I would run 

sudo apt-get update 
sudo apt-get upgrade

sudo iiab

and it would get a bit further till another program caused error. i had to repeat this process about 8 times for it to get through the entire install. The software works now, and have opted to try and install some media content on it.

Here is the diagnostics, (i forgot to do it before i started installing new stuff so not sure what all it'll capture.

http://sprunge.us/etw9oI

holta, 2020-11-13:
@JamesM92 are you able to install IIAB using an Ethernet cable to see if that avoids the 8 interruptions?

Or is that impossible in your current building?  ASIDE: some people (whose buildings do not offer Ethernet) have used their Windows/macOS laptops to act as an Ethernet-to-Internet router.

In short, they attached an Ethernet cable between their Raspberry Pi (IIAB) and Windows/macOS laptop, and then configured Windows/macOS to route that Ethernet traffic to the building's WiFi (and then onto the Internet!)  Easy to do _IF_ you can tolerate Windows/macOS network settings ;)

holta, 2020-11-13:
@JamesM92 two quick questions:

1) Out of curiosity, did you install a graphical desktop on top of Raspberry Pi OS Lite?  Which one if so?  Beware that adding a desktop environment can greatly affect WiFi and networking (best to test all WiFi and networking functionality _**before**_ adding any graphical desktop to be very sure !)

2) Is there any chance you connected your Raspberry Pi to your building/router's WiFi over 5GHz?  Just a warning that the current version of IIAB cannot support that, as so many/legacy "student" WiFi client device require 2.4GHz for now.  In any case, please do stick to 2.4GHz WiFi at this time, as we caution here:

   - https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes#known-issues
   - https://github.com/iiab/iiab/blob/master/vars/local_vars_big.yml#L10-L12

jvonau, 2020-11-14:
> I may have forgotten to mention that i was doing my install on wifi previously. Bot sure if that may have anything to do with it.
>

Using usbboot and usb3 has everything to do with flaky wifi on the first install where mongodb was an issue when you were trying:
>Raspberry Pi reference 2020-08-20
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 9a3a10bf1019ebb2d59053564dc6b90068bad27d, stage2

I've posted links to upstream trackers in other issues on the subject.
> So i tried a new install, this time i used
>
> Raspberry Pi OS (32bit) Lite : 2020-05-27

Ok the second install started out as stage2:
>Raspberry Pi reference 2020-05-27
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 825107f04027269db77426046f5085475b1ea22f, stage2
IIAB-EXPLANATION-OF-THE-ABOVE-------------------------------------------------
stage2 = Raspberry Pi OS Lite
stage4 = Raspberry Pi OS with desktop
stage5 = Raspberry Pi OS with desktop + recommended software

However there appears there is a desktop installed so I'm little confused as what was done behind the scenes:
>1643  1213 ?        00:00:00   lightdm
1644  1253 tty7     00:01:02     Xorg
1645  1629 ?        00:00:00     lightdm
1646  1733 ?        00:00:00       lxsession
1647  1818 ?        00:00:00         ssh-agent
1648  1908 ?        00:00:01         openbox
1649  1914 ?        00:00:00         lxpolkit
1650  1931 ?        00:00:02         lxpanel
1651  4953 ?        00:00:00           lxterminal
1652  4959 pts/0    00:00:00             bash
1653  1943 ?        00:00:00         pcmanfm
1654  1947 ?        00:00:00         xscreensaver
1655  1951 ?        00:03:04         chromium
1656  2145 ?        00:00:00           chromium
1657  2323 ?        00:05:56             chromium
1658  2148 ?        00:00:00           chromium
1659  2195 ?        00:00:00             chromium
1660  4415 ?        00:06:03               chromium
1661  6524 ?        00:00:00               chromium
1662  2326 ?        00:01:13           chromium

> This time i got errors again during install, but different errors, each time i ran the "sudo iiab" it would run then give an error it wasn't able to perform _____ tasks install. I would run
>
> sudo [...]

@JamesM92 Not sure what is being described, can this be expanded upon?
> Here is the diagnostics, (i forgot to do it before i started installing new stuff so not sure what all it'll capture.
> 
> http://sprunge.us/etw9oI

Lines of note: from iiab.ini
> iiab_network_mode = Gateway
iiab_network_mode_applied = LanController
hostapd_enabled = False

Lines of note: from iiab.env
>HOSTAPD_ENABLED=False

Lines of note from local_vars.yml
>hostapd_enabled: False

The above tells me that the admin-console was used to alter the network settings, could you expand on how you intend to use the network interfaces?

The contents of /etc/wpa_supplicant/wpa_supplicant.conf show multiple entries for the same SSID, what methods were used to configure the WiFi access to the internet? ie hand editing, raspi-config, admin-console or creating the connection in the desktop?

Can you check if your WiFi router offers 2.4Ghz and/or 5Ghz connections and if so are both bands use the same SSID name?

JamesM92, 2020-11-14:
@jvonau 
Desktop-
I tried to install a minimal gui after I had finished the whole install process. But I used the light image as the base. Installed IIAB, then tried to install a gui. (Like said i did a bunch of stuff before i ran the diagnostics.)

Network-
I tried to use the wifi control fields to entire SSID, somehow it seems the pi started broadcasting the same SSID, no clue what I did to cause that.

Also the 2020-05-27 image is one that I had made as a modified image with WPA-supllicant to connect to my wifi and a couple other small adjustments. So it wasn't a purely clean install.

Failures-
Not sure how to expand upon it. I would run the install and it would stop when it hit a failure. I didn't really keep track of which task it failed on.

But each time there was something new needing to be installed by 
Sudo apt-get update
Sudo apt-get upgrade

Then it would install a few more items before having another failed task.

-other
Currently working on a fresh install using a ethernet cord to handle the connection.

For anybody curious this is the case I'm using to house the pi with the laptop hard drive connection
-> https://www.argon40.com/argon-one-m-2-case-for-raspberry-pi-4.html

Is there a discussion board or forum? I've got a bit of a list for feature requests, but dont want to use bug reports to request them.

holta, 2020-11-14:
> Is there a discussion board or forum?

We have a Skype chat channel with about 20 people if you can stand that?

@JamesM92 email me if that's useful to you: holt @ unleashkids [dot] org

JamesM92, 2020-11-14:
Ok, so fresh install
Pi Os : 2020-08-05
Iiab:big
Local-vars: opted to install some that weren't included in the big by default. 

Node.js isnt able to install. Tried twice.

http://sprunge.us/9udi9Y

holta, 2020-11-14:
@JamesM92 what's the result of `node -v` ?

Note that the [pbx](https://github.com/iiab/iiab/tree/master/roles/pbx) role is not widely used at this time.  It might well need a community/maintainer to get it working properly on certain OS's like your 32-bit RaspiOS Lite.

holta, 2020-11-14:
@JamesM92 thanks for your feature requests and suggestions!

I hope it's Ok that I've reposted those 5 core ideas to:

http://minutes.iiab.io > Nov 19, 2020 > IIAB 8.0

holta, 2020-11-14:
@JamesM92 can you open new ticket(s) with specific/other priority issues, so we can close this umbrella ticket soon?

(Shedding light on the original MongoDB unzipping issue if you possibly can isolate the root cause, to help others if at all possible!)

JamesM92, 2020-11-14:
Ah sorry.

I reflashed the card.

The original issue of the mongoDB is unable to be replicated from a new install to be able to resolve.

So I guess this ticket should be closed.

Sorry about that.

holta, 2020-11-14:
> The original issue of the mongoDB is unable to be replicated from a new install to be able to resolve.

No worries, and thanks for following up!

closed 2020-11-14 by holta

jvonau, 2020-11-15:
> For anybody curious this is the case I'm using to house the pi with the laptop hard drive connection
> -> https://www.argon40.com/argon-one-m-2-case-for-raspberry-pi-4.html
> 

https://github.com/iiab/iiab/pull/2337#issuecomment-614134494
https://bugs.launchpad.net/ubuntu/+source/linux-raspi2/+bug/1861338
https://bugs.launchpad.net/ubuntu/+source/linux-raspi2/+bug/1857760
#2497

holta, 2020-11-15:
NEW WARNING (re: 2.4GHz WiFi interference with USB 3.0 boot devices) ADDED TO:

- http://FAQ.IIAB.IO > "44 Can I install IIAB onto a USB drive?"

(@jvonau & others please help me refine this warning if new info arises, Thanks!)