iiab/iiab #2636: make-kiwix-lib fails on non-standard language code

tim-moody, 2020-11-15:
if lang code lookup fails, e.g. if EN instead of en, a python error occurs

holta, 2020-11-16:
For those new to IIAB, this involves:

- https://github.com/iiab/iiab/blob/master/roles/kiwix/templates/iiab-make-kiwix-lib (bash, to invoke Python 3 below...)
- https://github.com/iiab/iiab/blob/master/roles/kiwix/templates/iiab-make-kiwix-lib3.py

These are run at the command line (or by IIAB Admin Console) as `/usr/bin/iiab-make-kiwix-lib` to index your ZIM files that live in `/library/zims/content`

...as briefly mentioned in http://FAQ.IIAB.IO > "33 How do I add ZIM files, like Wikipedia?"

tim-moody, 2020-11-16:
This should be fixed, though the remedy is not trivial. It really only occurs for a zim that was created with zimwriterfs and not with a proper language code.

- if lang not found, try lower case
- if still not found try one of the non-standard codes, like en should be eng and fr fra
- if still not found call it eng

tim-moody, 2023-03-20:
Is this still a problem? I don't see where it would fail in the code.

tim-moody, 2023-03-20:
definitely not a duplicate

closed 2025-03-11 by tim-moody