iiab/iiab #2638: What USB 3.0 stick/drive patterns degrade a Raspberry Pi's 2.4GHz WiFi?

holta, 2020-11-16:
Thanks to @jvonau and others who've uncovered some real problems here!  Let's try to identify a couple general guidelines for schools/clinics using USB 3.0 so they can avoid accidentally killing their Raspberry Pi's internal WiFi hotspot.

A big Thank You to _everybody_ who can report their experiences here:

- Can USB 2.0 flash drives be used safely in USB 3.0 slots?
- Can USB 3.0 flash drives be used safely in USB 2.0 slots?
- What fails when a USB3 stick is put in a USB3 slot ?
  - Presumably WiFi is affected much more when data is being retrieved from the USB3 stick / flash drive?
  - Is it possible WiFi isn't affected at all when data is not being read from [or written to] the USB 3.0 stick?
- What _other_ tests should we run to better understand this?

Refs:

- #2497
- #2585
- raspberrypi/firmware#1430

holta, 2020-11-17:
Does https://pibox.in/2020/08/29/usb-c-hubs-and-wifi-issues-fixed/ say it all?  Thanks to @jvonau for uncovering this!

Tin foil shielding of cable connectors[*] sounds like a JOKE —until you watch these videos.

_[*] and better cable connectors & EMI / RF interference ferrite chokes in general._

(What USB3 sticks &/or devices appear to be the highest risk?)

JamesM92, 2020-11-29:
Not sure if helpful, but I'm using a usb3 to sata m2 ssd to control my pi. I currently have it connected via hardwire ethernet.

It is able to broadcast its wifi. But anybody who tries to connect to it find that they cant authenticate their connection.

Also the pie is unable to connect to another wifi source. 

Would a diagnostics download be helpful? Or would somebody need to remote into it to be useful?

holta, 2020-11-30:
@JamesM92 can you send the URL resulting from running `sudo iiab-diagnostics` for starters?  Thanks !

jvonau, 2020-11-30:
A URL that has a picture and specs for of the device would be helpful also. Two things that possibility could cause the wifi drop out, RF interference or inadequate power to run the device.

JamesM92, 2020-12-01:
Diagnostics -> https://sprung.us/lxTZPd
Pi4-8gb
Case -> https://www.argon40.com/argon-one-m-2-case-for-raspberry-pi-4.html

jvonau, 2021-04-23:
#2739 This issue should really not have a milestone attached at all, this is more general info about suspect hardware out in the wild.

holta, 2022-12-09:
Just FYI for those experiencing similar, there are about 100 posts on the general topic of USB3 slowdowns with RPi 4's here:

https://forums.raspberrypi.com/viewtopic.php?t=245931