iiab/iiab #2639: Map AddOns Note

JamesM92, 2020-11-17:
Not a bug, just something that I think needs more explanation on the Map AddOns Page.

Its not stated anywhere on the page, but that the photos you download, are only to zoom 13, which is the same as the satellite photos of a standard map download. So if you've already downloaded a standard map for the region there is no reason to download the satellite photos.

I assumed it was more zoomed in than the standard maps. But that was on my part.

holta, 2020-11-17:
> the photos you download, are only to zoom 13, which is the same as the satellite photos of a standard map download. So if you've already downloaded a standard map for the region there is no reason to download the satellite photos.

I don't think this true!  @georgejhunt and other can confirm, but I believe:

Map Pack "continents" provide satellite photos with 10 levels of zoom (i.e. z0-z9, or 305-meter resolution).

Whereas Satellite Photos "squares" (100x100 km, 300x300 km, or 1000x1000 km) provide 4 additional levels of zoom (i.e. z10-z13, or 19-meter resolution).

JamesM92, 2020-11-17:
Hmm, if that's the case, then there may be a different issue?

I downloaded the world map, (actually all the maps lol), then i tried to install the 1000kx1000km of the State I live in. In  the Job status is saying over and over that a particular tile is already downloaded. 

(i don't believe Ive already installed any of the add-ons previously but I may be mistaken.)

[image]

{Hope my issue reports aren't a nuisance, I'm just goofing around and seeing whats breaking or confusing me in the process)

holta, 2020-11-17:
@JamesM92 your feedback here can be incredibly useful:

Personally I'd very much agree, and while not at all easy to make this happen (certainly in 2020), progress indicators and content context could be explained better in future years.

Even if these are hard to change today!  No Matter: your input helps keeps us all honest around such higher-level goals (regardless how hard it always is for newcomers to understand & customize their community's content — in general, as we should all ask what improving these UX's really means year-by-year — for curators/implementers/librarians and kids in all countries !)

tim-moody, 2020-11-17:
Neither answer is right. 'map packs' is a misnomer because two things that used to be bundled aren't really now. There are maps (tiles) and satellite photos. If you look at Get Map Regions there are Required Base Files which are automatically checked. The remaining map files are just what they say, map tiles with no photos. To get photos for more than 'Satellite to Zoom 9 931.18 MB' you go to Get Map Addons.

holta, 2020-11-17:
Aside: the reason that a lot of end users much prefer the concept of a Map Pack "continent" is that this provides them most everything they need without drowning in details: being pressed for time such folks demand "Content Bouquets" rather than granular lists of content details + metadata[*].

So the "Map Pack" notion is very useful for these higher-level folk wanting to quickly abstract away details of ~3 files (and other details they consider to be minutiae...that they truly don't have time for).

[*] as preferred by the more perfectionist librarians/curators/etc.  This tension is very natural with content curation (library science, where the public mostly wants nothing to do with the card catalog bureaucracy, whereas the librarian sees it as "biblical" !)  So our long-term challenge is to offer more high-level Content Bouquets (that hide laundry lists nitty-gritty details, as the vast majority prefer) _while at the very same time_ offering OCD curatorial perfectionists deeper insights...into the details of individual flowers (where possible!) as that fine-toothed gardening (while certainly not for everyone!) is ironically/also what keeps the larger garden hanging together for all ;)

tim-moody, 2020-11-17:
The "Map Pack" notion  may be preferred, but it is not currently available either in the installer or Admin Console.

tim-moody, 2020-11-17:
And are you saying that if I get a region I should get high resolution satellite photos for the entire region?

closed 2021-05-03 by JamesM92