iiab/iiab #2642: admin-console|cmdsrv fails to detect WiFi WAN

jvonau, 2020-11-21:
[image]
[image]

tim-moody, 2020-11-21:
likely. hasn't been changed in a long time, and not since up_down was implemented, so probably looking for wired interface.

jvonau, 2020-11-21:
Not likely, the working code uses ansible_default_ipv4 to detect a gateway. Taken from "Display Ansible Facts":

>  "ansible_default_ipv4": {},
  "ansible_default_ipv6": {},

>  "ansible_wlan0": {
    "active": false,
    "device": "wlan0",
    "macaddress": "dc:a6:32:20:b8:78",
    "module": "brcmfmac",
    "mtu": 1500,
    "pciid": "mmc1:0001:1",
    "promisc": false,
    "type": "ether"
  },

Think cmdsrv was started before wlan0 made a connection, restarting iiab-cmdsrv re-runs ansible setup with the correct device now being displayed. 
[image]

>  "ansible_default_ipv4": {
    "address": "192.168.0.18",
    "alias": "wlan0",
    "broadcast": "192.168.0.255",
    "gateway": "192.168.0.1",
    "interface": "wlan0",
    "macaddress": "dc:a6:32:20:b8:78",
    "mtu": 1500,
    "netmask": "255.255.255.0",
    "network": "192.168.0.0",
    "type": "ether"
  },
  "ansible_default_ipv6": {},

>   "ansible_wlan0": {
    "active": true,
    "device": "wlan0",
    "ipv4": {
      "address": "192.168.0.18",
      "broadcast": "192.168.0.255",
      "netmask": "255.255.255.0",
      "network": "192.168.0.0"
    },
    "ipv6": [
      {
        "address": "fd00:fc:8ddb:6692:7d01:fb3b:e0f8:dc24",
        "prefix": "64",
        "scope": "global"
      },
      {
        "address": "fe80::9616:78ec:4bb:6bb",
        "prefix": "64",
        "scope": "link"
      }
    ],
    "macaddress": "dc:a6:32:20:b8:78",
    "module": "brcmfmac",
    "mtu": 1500,
    "pciid": "mmc1:0001:1",
    "promisc": false,
    "type": "ether"
  },

holta, 2021-06-18:
@tim-moody @jvonau is there any progress here since November?

(Or is this possibly too hard to fix?)

tim-moody, 2021-06-18:
Only thing I can think of is a button to refresh status.

jvonau, 2021-06-19:
There could be better staging of when cmd-serve starts to allow the network to become fully operational before starting the service using a later systemd target.  Perhaps network.target might be late enough?

tim-moody, 2021-06-19:
what would the target be such that wlan0 is not only active but has authenticated and associated with an upstream hotspot? what would the timeout be?

tim-moody, 2021-06-19:
actually, as I think about it, I would not want to prevent a user from doing things that don't require an internet connection just because there isn't one.

holta, 2021-08-11:
I'm moving this tkt from the IIAB 7.2 milestone to the 8.0 milestone.

(But if there's any kind of short-term action / correctives useful in coming days, that's fine too.)

jvonau, 2021-08-11:
Delay the starting of the service to be later in the boot sequence using After= some appropriate target or service in the unit file to give the network time to become fully functional.

jvonau, 2025-04-06:
Still a known issue