iiab/iiab #2650: How many of these 10 sysctl kernel IPv4 IPv6 tweaks are still needed ~5 years later?

holta, 2020-11-27:
PRs like #2647 (@jvonau) and #2648 weren't quite enough, as the [sysctl](https://docs.ansible.com/ansible/latest/collections/ansible/posix/sysctl_module.html) Ansible module is also suddenly moved into of Ansible's new [ansible.posix](https://galaxy.ansible.com/ansible/posix) collection.

(Same story as with Ansible module [selinux](https://docs.ansible.com/ansible/latest/collections/ansible/posix/selinux_module.html), that we just faced in the above PR's.)

As a test I've commented out the 10 sysctl commands at https://github.com/iiab/iiab/blob/master/roles/2-common/tasks/main.yml#L24-L38 which is allowing the BIG-sized install to complete on Debian 11 VM 10.8.0.26 = deb11-srv-BIG-1127.  This VM installed successfully, which we can now use to investigate QUESTION in this ticket's subject line: _how many of these 10 `systctl` tweaks are truly still needed and necessary a half-decade later?_

```
- name: Use 'sysctl' to set 10 network/kernel settings, turning off IPv6 if possible
  sysctl:
    name: "{{ item.name }}"
    value: "{{ item.value }}"
  with_items:
    - { name: 'net.ipv4.ip_forward', value: '1' }
    - { name: 'net.ipv4.conf.default.rp_filter', value: '1' }
    - { name: 'net.ipv4.conf.default.accept_source_route', value: '0' }
    - { name: 'kernel.sysrq', value: '1' }
    - { name: 'kernel.core_uses_pid', value: '1' }
    - { name: 'net.ipv4.tcp_syncookies', value: '1' }
    - { name: 'kernel.shmmax', value: '268435456' }
    - { name: 'net.ipv6.conf.all.disable_ipv6', value: '1' }    # IPv6 disabled
    - { name: 'net.ipv6.conf.default.disable_ipv6', value: '1' }
    - { name: 'net.ipv6.conf.lo.disable_ipv6', value: '1' }
```

holta, 2020-11-27:
@jvonau @georgejhunt @tim-moody feel free to log into 10.8.0.26 if it helps us all answer this question?

jvonau, 2020-11-27:
Not in front of a vpn machine replying from my phone, I'll login later today, but
- { name: 'net.ipv4.ip_forward', value: '1' } does the masquerading from LAN -> Internet
- { name: 'net.ipv4.conf.default.rp_filter', value: '1' } Harder to explain in a few words search for your answer
- { name: 'net.ipv4.conf.default.accept_source_route', value: '0' } Harder to explain in a few words search for your answer 

These probability should be replaced:
- { name: 'net.ipv6.conf.all.disable_ipv6', value: '1' }    # IPv6 disabled
- { name: 'net.ipv6.conf.default.disable_ipv6', value: '1' }
- { name: 'net.ipv6.conf.lo.disable_ipv6', value: '1' }
with:
- { name: 'net.ipv6.conf.br0.disable_ipv6', value: '1' }

Not even sure if these 'net' values are still applicable with the latest kernels, haven't been keeping up on the subject.

holta, 2020-11-27:
Thanks @jvonau for the excellent preliminary summary above.

@georgejhunt all 10 are now commented out by PR #2651 on an interim basis, to allow IIAB installs to complete.  It would be great if you could take a look today or in coming days, as to which of these ~10 should really be around the long-haul now, in context with Jerry's suggestions above?

georgejhunt, 2020-11-27:
I think this is a release 7.3 issue. There is no error string attached to this issue which documents which ansible module is failing.

jvonau, 2020-11-29:
Think disabling ipv6 on the internet facing network device is going to bite some future new user who has ipv6 only access hard at some point soon but whatever....

holta, 2020-11-29:
> Think disabling ipv6 on the internet facing network device is going to bite some future new user who has ipv6 only access hard at some point soon but whatever....

Thanks for pointing this out.

To help everyone keep their head's up watching out for and/or preparing for this.

jvonau, 2022-04-05:
still relevant?

holta, 2022-04-05:
> still relevant?

Can't hurt, towards refining this going forward &mdash; or at least being ready for the future.

holta, 2022-06-23:
Somewhat related PR, makes some progress here:

- PR #3173