iiab/iiab #2658: iiab 7,2 Testing with Debian 10.7

shanti-bhardwa, 2020-12-06:
### Expected behavior and actual behavior:

### Steps to reproduce the problem:

### IIAB version, OS version, hardware details:

### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

shanti-bhardwa, 2020-12-06:
testing Debian 10.7 with DELL lattitude E6230, i5 4 GB of RAM, 500GB Drive

Image: debian-live-10.7.0-amd64-mate.iso

The .iso image installs, but does not have wifi firmware

I had to install it separately as root with following commands.

echo “deb http://ft[.se.debian.org/debian buster non-free contrin” >> etcapt/sources.list

apt-get update
apt-get install firmware-iwlwifi

apt reboot

Then apt-get install curl

The run the the 1 line install 

IIAB medium installed without any issues

but cannot use the SSID as these two services do not run

networking-service & iiab-dnsmasq

main reason seem to be br0 not up

get Failed to start network interfaces at boot time

I recall we had this problem before when testing Ubuntu or may be I don’t have the correct WIFI drives

see http://spruge.us/FB1m4I?en

shanti-bhardwa, 2020-12-06:
sorry closed in error

holta, 2020-12-06:
> echo “deb http://ft[.se.debian.org/debian buster non-free contrin” >> etcapt/sources.list

echo “deb http://ftp.se.debian.org/debian buster non-free contrib” >> /etc/apt/sources.list

> see http://spruge.us/FB1m4I?en

http://sprunge.us/FB1m4I?en

jvonau, 2020-12-07:
> > echo “deb [http://ft[.se.debian.org/debian](http://ft%5B.se.debian.org/debian) buster non-free contrin” >> etcapt/sources.list
> 
> echo “deb http://ftp.se.debian.org/debian buster non-free contrib” >> /etc/apt/sources.list
> 
That is not needed, the wifi device is detected. 

This is sort of like Ubuntu, NetworkManager is used but I'm sure systemd-networkd is present as networkctl responds, let just do the same as netplan does and use systemd-networkd. Give this branch a test spin:
`cd /opt/iiab/iiab/`
`sudo git checkout -b jv-test`
`sudo git pull https://github.com/jvonau/iiab debian_mate`
`sudo ./iiab-network`
`sudo reboot`

holta, 2020-12-07:
Thank you @shanti-bhardwa if you can help test @jvonau's recommendation just above.

shanti-bhardwa, 2020-12-07:
tested as above. iiab-dnsmaq service starts, br0 is up but networking.service does not start up, Status shows 

failed to bring up eth0

when in fact etho is UP and has an IP address assigned by my router.

root@box:~# systemctl status networking
● networking.service - Raise network interfaces
   Loaded: loaded (/lib/systemd/system/networking.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2020-12-07 21:21:09 GMT; 5min ago
     Docs: man:interfaces(5)
  Process: 566 ExecStart=/sbin/ifup -a --read-environment (code=exited, status=1/FAILURE)
 Main PID: 566 (code=exited, status=1/FAILURE)

Dec 07 21:21:09 box.lan dhclient[716]: than a configuration issue please read the section on submitting
Dec 07 21:21:09 box.lan dhclient[716]: bugs on either our web page at www.isc.org or in the README file
Dec 07 21:21:09 box.lan dhclient[716]: before submitting a bug.  These pages explain the proper
Dec 07 21:21:09 box.lan dhclient[716]: process and the information we find helpful for debugging.
Dec 07 21:21:09 box.lan dhclient[716]: 
Dec 07 21:21:09 box.lan dhclient[716]: exiting.
Dec 07 21:21:09 box.lan ifup[566]: ifup: failed to bring up eth0
Dec 07 21:21:09 box.lan systemd[1]: networking.service: Main process exited, code=exited, status=1/FAILURE
Dec 07 21:21:09 box.lan systemd[1]: networking.service: Failed with result 'exit-code'.
Dec 07 21:21:09 box.lan systemd[1]: Failed to start Raise network interfaces.

Also cannot connect more than one device to the SSID

http://sprunge.us/tMIMeC?en

shanti-bhardwa, 2020-12-07:
also can't connect to "uppstream" WIFI on my home router. keep getting these messages when I do dmesg

 1131.643933] iwlwifi 0000:02:00.0: beacon intervals don't match (0, 100)
[ 1131.644240] IPv6: ADDRCONF(NETDEV_UP): wlp2s0: link is not ready
[ 1131.685410] iwlwifi 0000:02:00.0: beacon intervals don't match (0, 100)
[ 1132.437711] iwlwifi 0000:02:00.0: beacon intervals don't match (0, 100)
[ 1132.437864] iwlwifi 0000:02:00.0: beacon intervals don't match (0, 100)
[ 1132.438060] iwlwifi 0000:02:00.0: beacon intervals don't match (0, 100

I guess issue with driver software
_

shanti-bhardwa, 2020-12-07:
is this supposed to use wpa_supplicant to connect to my router?
oot@box:~# systemctl status wpa_supplicant
● wpa_supplicant.service - WPA supplicant
   Loaded: loaded (/lib/systemd/system/wpa_supplicant.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2020-12-07 21:21:06 GMT; 37min ago
 Main PID: 574 (wpa_supplicant)
    Tasks: 1 (limit: 4915)
   Memory: 5.2M
   CGroup: /system.slice/wpa_supplicant.service
           └─574 /sbin/wpa_supplicant -u -s -O /run/wpa_supplicant

Dec 07 21:34:16 box.lan wpa_supplicant[574]: wlp2s0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="BT-3PAKNP" auth_failures=1 duration=10 reason=CONN_FAILED
Dec 07 21:34:26 box.lan wpa_supplicant[574]: wlp2s0: CTRL-EVENT-SSID-REENABLED id=0 ssid="BT-3PAKNP"
Dec 07 21:34:26 box.lan wpa_supplicant[574]: wlp2s0: SME: Trying to authenticate with e4:75:dc:39:81:86 (SSID='BT-3PAKNP' freq=2462 MHz)
Dec 07 21:34:26 box.lan wpa_supplicant[574]: wlp2s0: SME: Authentication request to the driver failed
Dec 07 21:34:26 box.lan wpa_supplicant[574]: wlp2s0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="BT-3PAKNP" auth_failures=2 duration=20 reason=CONN_FAILED
Dec 07 21:34:32 box.lan wpa_supplicant[574]: wlp2s0: Reject scan trigger since one is already pending
Dec 07 21:39:40 box.lan wpa_supplicant[574]: wlp2s0: Reject scan trigger since one is already pending
Dec 07 21:44:40 box.lan wpa_supplicant[574]: wlp2s0: Reject scan trigger since one is already pending
Dec 07 21:50:31 box.lan wpa_supplicant[574]: wlp2s0: Reject scan trigger since one is already pending
Dec 07 21:55:46 box.lan wpa_supplicant[574]: wlp2s0: Reject scan trigger since one is already pending
root@box:~#

shanti-bhardwa, 2020-12-07:
can't find wpa_supplicant.conf
oot@box:/etc/wpa_supplicant# ls -l
total 40
-rwxr-xr-x 1 root root   937 Mar 24  2020 action_wpa.sh
-rw-r--r-- 1 root root 25569 Mar 24  2020 functions.sh
-rwxr-xr-x 1 root root  4696 Mar 24  2020 ifupdown.sh

jvonau, 2020-12-07:
NetworkManager handles the wpa_supplicant part.

> networking.service does not start up

That is more of a red herring, without the full iiab-install.log I can't tell if we enabled the service purposely or was dragged as a dependency and was auto enabled by apt.

>2127 2020-12-07 17:35:33,229 p=2431 u=root n=ansible | TASK [Enable & Restart networkd-dispatcher.service] ****************************
2128 2020-12-07 17:35:33,527 p=2431 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Could not find the requested service networkd-dispatcher: host"}
2129 2020-12-07 17:35:33,530 p=2431 u=root n=ansible | PLAY RECAP *********************************************************************
2130 2020-12-07 17:35:33,530 p=2431 u=root n=ansible | 127.0.0.1                  : ok=91   changed=18   unreachable=0    failed=1    skipped=90   rescued=0    ignored=0   
2131 

Look like networkd-dispatcher is not installed can you varify if that is the case with `sudo apt list networkd-dispatcher`

shanti-bhardwa, 2020-12-07:
root@box:~#  apt list networkd-dispatcher
Listing... Done
networkd-dispatcher/stable,stable 2.0-2 all

and

root@box:~# systemctl status NetworkManager
● NetworkManager.service - Network Manager
   Loaded: loaded (/lib/systemd/system/NetworkManager.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2020-12-07 21:21:10 GMT; 1h 2min ago
     Docs: man:NetworkManager(8)
 Main PID: 638 (NetworkManager)
    Tasks: 4 (limit: 4915)
   Memory: 16.3M
   CGroup: /system.slice/NetworkManager.service
           ├─638 /usr/sbin/NetworkManager --no-daemon
           └─790 /sbin/dhclient -d -q -sf /usr/lib/NetworkManager/nm-dhcp-helper -pf /run/dhclient-eno1.pid -lf /var/lib/NetworkManager/dhclient-f933c927-7e44-4870-a6f4-b85282e1be4a-eno1.lease -cf /

Dec 07 22:10:21 box.lan NetworkManager[638]: <info>  [1607379021.2931] device (tun0): Activation: successful, device activated.
Dec 07 22:11:47 box.lan NetworkManager[638]: <info>  [1607379107.4083] device (wlp2s0): set-hw-addr: set MAC address to 86:A4:4F:15:7B:05 (scanning)
Dec 07 22:11:47 box.lan NetworkManager[638]: <info>  [1607379107.4576] device (wlp2s0): supplicant interface state: inactive -> disconnected
Dec 07 22:11:47 box.lan NetworkManager[638]: <info>  [1607379107.4632] device (wlp2s0): supplicant interface state: disconnected -> inactive
Dec 07 22:17:02 box.lan NetworkManager[638]: <info>  [1607379422.3994] device (wlp2s0): set-hw-addr: set MAC address to AE:86:2B:C9:25:AF (scanning)
Dec 07 22:17:02 box.lan NetworkManager[638]: <info>  [1607379422.4376] device (wlp2s0): supplicant interface state: inactive -> disconnected
Dec 07 22:17:02 box.lan NetworkManager[638]: <info>  [1607379422.4430] device (wlp2s0): supplicant interface state: disconnected -> inactive
Dec 07 22:22:17 box.lan NetworkManager[638]: <info>  [1607379737.4079] device (wlp2s0): set-hw-addr: set MAC address to 36:B9:26:C9:C5:D7 (scanning)
Dec 07 22:22:17 box.lan NetworkManager[638]: <info>  [1607379737.4581] device (wlp2s0): supplicant interface state: inactive -> disconnected
Dec 07 22:22:17 box.lan NetworkManager[638]: <info>  [1607379737.4637] device (wlp2s0): supplicant interface state: disconnected -> inactive

shanti-bhardwa, 2020-12-07:
does this make any sense

root@box:~# systemctl status networkd-dispatcher
Unit networkd-dispatcher.service could not be found.
root@box:~# systemctl start networkd-dispatcher
Failed to start networkd-dispatcher.service: Unit networkd-dispatcher.service not found.
root@box:~# systemctl |grep networkd
  systemd-networkd-wait-online.service                                                        loaded active exited    Wait for Network to be Configured                                           
  systemd-networkd.service                                                                    loaded active running   Network Service                                                             
  systemd-networkd.socket                                                                     loaded active running   Network Service Netlink Socket                                              
root@box:~# systemctl status systemd-networkd.service    
● systemd-networkd.service - Network Service
   Loaded: loaded (/lib/systemd/system/systemd-networkd.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2020-12-07 21:21:01 GMT; 1h 7min ago
     Docs: man:systemd-networkd.service(8)
 Main PID: 438 (systemd-network)
   Status: "Processing requests..."
    Tasks: 1 (limit: 4915)
   Memory: 3.6M
   CGroup: /system.slice/systemd-networkd.service
           └─438 /lib/systemd/systemd-networkd

Dec 07 21:21:02 box.lan systemd-networkd[438]: wlan0: Interface name change detected, wlan0 has been renamed to wlp2s0.
Dec 07 21:21:07 box.lan systemd-networkd[438]: br0: Lost carrier
Dec 07 21:21:07 box.lan systemd-networkd[438]: ap0: Gained carrier
Dec 07 21:21:07 box.lan systemd-networkd[438]: br0: Gained carrier
Dec 07 21:21:07 box.lan systemd-networkd[438]: br0: Configured
Dec 07 21:21:15 box.lan systemd-networkd[438]: eno1: Gained carrier
Dec 07 21:21:16 box.lan systemd-networkd[438]: eno1: Gained IPv6LL
Dec 07 21:21:23 box.lan systemd-networkd[438]: tun0: Gained carrier
Dec 07 22:10:19 box.lan systemd-networkd[438]: tun0: Lost carrier
Dec 07 22:10:21 box.lan systemd-networkd[438]: tun0: Gained carrier

jvonau, 2020-12-07:
> root@box:~# apt list networkd-dispatcher
> Listing... Done
> networkd-dispatcher/stable,stable 2.0-2 all

>Failed to start networkd-dispatcher.service: Unit networkd-dispatcher.service not found.
 
`sudo apt install networkd-dispatcher`   Temp fix for now 
`sudo systemctl disable networking.service`  Undo this one, not sure where ifupdown was enabled.
`cd /opt/iiab/iiab/`
`sudo ./iiab-network`
`reboot`

shanti-bhardwa, 2020-12-08:
Re-built my debian 10.7 Mate desktop again, but this time provided the correct wifi drivers. WIFI interface was recognised, and configured correctly and I could connect to upstream internet from my home router.
IIAB 7.2 installed fine but SSID of hotspot does not work. iiab-dnsmasq and dnsmasq services don't start.
dnsmasq: unknown interface br0

 because br0 is unknown and cannot be started. I applied above fix, but still the same problem.

See http://sprunge.us/MHXSvP?en

holta, 2020-12-08:
Thanks @shanti-bhardwa & @jvonau I hope we learn from the hard work here.

(Regardless if everybody abandons Debian 10 "Buster" for Debian 11 "Bullseye" in coming months, where presumably the same issues apply!)

shanti-bhardwa, 2020-12-08:
looks like we need to create wpa_supplicant.conf file here for Upstream WIFI

/etc/wpa_supplicant    # Upstream Wi-Fi

 and when I do  dmesg | grep iwlwifi I get lots of messages like "beacon intervals don't match (0,100) 
iwlwifi is the driver

shanti-bhardwa, 2020-12-09:
beacon intervals don't match (0,100) messages are associated with the way Access Point is set up.
They stop when I do iiab-hotspot-off and I can connect to upstream WIFI Router.
See here https://github.com/oblique/create_ap/issues/53

jvonau, 2020-12-09:
> They stop when I do iiab-hotspot-off and I can connect to upstream WIFI Router.

With the machine in that state can you grab `sudo /usr/sbin/iw dev` and `sudo systemctl status systemd-networkd` posting the results in this issue please. Think your upstream wifi might be a 5GHz connection, the channel in use should tell the tale.

> looks like we need to create wpa_supplicant.conf file here for Upstream WIFI
> 
> /etc/wpa_supplicant # Upstream Wi-Fi

No the wifi credentials are stored in your /etc/NetworkManager/system-connections/BT-3PAKNP.nmconnection file
and the wifi device is seen by NetworkManager
>1084 =IIAB==========================================================================
1085 COMMAND: /usr/bin/nmcli d    # NetworkManager status
1086 
1087 DEVICE  TYPE      STATE         CONNECTION         
1088 eno1    ethernet  connected     Wired connection 1 
1089 wlp2s0  wifi      disconnected  --                 
1090 br0     bridge    unmanaged     --                 
1091 lo      loopback  unmanaged     --                 
1092 ap0     wifi      unmanaged     --                 

The below are not good:
> systemd_networkd_active = False

>DIRECTORY /etc/systemd/network FILES WILL FOLLOW...IF THEY EXIST

>Dec 08 22:38:50 box.lan networkd-dispatcher[520]: WARNING: systemd-networkd is not running, output will be incomplete.

I can see that networkd-dispatcher was installed, are you sure that you applied my branch as per the [above](https://github.com/iiab/iiab/issues/2658#issuecomment-739661131) instructions as it appears that you are currently at the last commit on master and would explain why br0 doesn't have an ip address.
>  88 [runtime]
  89 iiab_stage = 9
  90 iiab_base_ver = 7.2
  91 iiab_revision = 0
  92 runtime_branch = master
  93 runtime_commit = 0fcd74ded083002ec6bd37bc05a31b328155085b
  94 runtime_date = 2020-12-08T22:34:53Z

jvonau, 2020-12-09:
@holta This is an error, the filename contains spaces, should be 'Wired connection 1'.
> 849 -IIAB--------------------------------------------------------------------------
 850 FILE DOES NOT EXIST: /etc/NetworkManager/system-connections/Wired
 851 -IIAB--------------------------------------------------------------------------
 852 FILE DOES NOT EXIST: /etc/NetworkManager/system-connections/connection
 853 -IIAB--------------------------------------------------------------------------
 854 FILE DOES NOT EXIST: /etc/NetworkManager/system-connections/1

jvonau, 2020-12-09:
>2302 Dec  8 22:47:17 box vmunix: [  529.625660] wlp2s0: authenticate with e4:75:dc:39:81:85
2303 Dec  8 22:47:27 box vmunix: [  539.860167] wlp2s0: authenticate with e4:75:dc:39:81:86
2304 Dec  8 22:47:39 box vmunix: [  552.023778] IPv6: ADDRCONF(NETDEV_UP): wlp2s0: link is not ready
2305 Dec  8 22:47:39 box vmunix: [  552.181645] IPv6: ADDRCONF(NETDEV_UP): wlp2s0: link is not ready

Interesting that there are 2 connection attempts with different ['bssid'](https://www.speedcheck.org/wiki/bssid/) that are based on the mac addresses of the radios involved. Given the strings are one character off from each other I'll venture to say the AP in question is using the same ssid for both radios for the 2.4 & 5 bands 

This is the connection that the machine is configured to auto-connect with when booted. The mac addresses :85 :86 above are different from this one.
>  829 [wifi]
 830 mac-address=8C:70:5A:BD:BA:48
 831 mac-address-blacklist=
 832 mode=infrastructure
 833 ssid=BT-3PAKNP

@shanti-bhardwa Were you trying to change the upstream wifi connection from above to a different router?

holta, 2020-12-10:
@shanti-bhardwa:

1) The `iiab-diagnostics` command should be fixed now if you can confirm, after running `git pull` etc?

2) If you have time, please also respond to @jvonau's questions above!

jvonau, 2021-04-23:
Why has this issue been moved to 8.0 when #2749 is listed as 7.2? #2739

holta, 2021-04-23:
Debian 10 is not our priority OS, but the question is fair.

So I've moved this tkt back to IIAB 7.2's milestone: https://github.com/iiab/iiab/milestone/7

The amount of testing and documentation needed is overwhelmingly large, but hopefully others will assist.  e.g. @deldesir in Haiti (who's been a Debian fan for years I believe?) has said he may be able to volunteer some time here this wkd, possibly assisting with QA of things like: PR #2742, #2744, PR #2747 etc.

holta, 2021-05-21:
FYI the `/var/mail/iiab-admin` growing bloat & associated errors are largely solved, thanks to 3 PR's off of #2777.

holta, 2021-05-29:
FYI the need for Debian 10 testing will likely end shortly, accelerated by the release of Raspbian 11[*] pre-announced for a June release in not so many weeks.

[*] based on Debian 11 Bullseye.

holta, 2021-06-18:
Good enough.  Debian 10.x has been tested repeatedly in the 6.5 months since.

@shanti-bhardwa please open a new ticket if this or similar recurs !

closed 2021-06-18 by holta