iiab/iiab #2659: Pink Ansible warning: "configured path /root/.ansible/collections does not exist"

holta, 2020-12-07:
Not a showstopper it appears, but maybe something we should clean up the loud/pink warning below?

```
root@box:~# ansible-galaxy collection list

# /usr/share/ansible/collections/ansible_collections
Collection           Version
-------------------- -------
ansible.netcommon    1.4.1
ansible.posix        1.1.1
community.general    1.3.0
community.kubernetes 1.1.1
community.mysql      1.1.1
google.cloud         1.0.1
[WARNING]: - the configured path /root/.ansible/collections does not exist.
```

FYI:

```
root@box:~# grep -n collections_path /etc/ansible/ansible.cfg
64:# collections_paths = ~/.ansible/collections:/usr/share/ansible/collections
```

@jvonau & all: should we added the line `collections_paths=/usr/share/ansible/collections` or `collections_paths = /usr/share/ansible/collections` to [/opt/iiab/iiab/ansible.cfg](https://github.com/iiab/iiab/blob/master/ansible.cfg) ?  (Which indeed makes the warning go away!)

Or perhaps just leave it as is, given the warning appears quite harmless ??

Ref: PR #2655 "(1) create /opt/iiab/iiab/collections.yml (2) move Ansible Collections from /root/.ansible to /usr/share/ansible"

jvonau, 2020-12-07:
Seems like a transient ansible thing, collections are still a little rough around the edges..

holta, 2020-12-07:
> Seems like a transient ansible thing, collections are still a little rough around the edges..

I agree.

Let's leave it as is for now, and monitor Ansible Collections's evolving recommendations in 2021.

holta, 2021-02-01:
2 months later this issue seems to have resolved itself -- presumably thanks to ansible-base 2.10.5 -- for example on Debian 10, see:

```
root@box:~# ansible-galaxy collection list

# /usr/share/ansible/collections/ansible_collections
Collection           Version
-------------------- -------
ansible.posix        1.1.1
community.general    2.0.0
community.mysql      1.2.0
community.postgresql 1.1.0
```

Refs: #2669 #2670 #2671 #2672

closed 2021-02-01 by holta

holta, 2021-02-21:
Just to confirm the latest Ansible Collection belows &mdash; that worked with a BIG-sized IIAB install on ansible-base 2.10.6 on kernel 5.10.0-3-amd64 on "desktop" pre-release of Debian 11 (Bullseye) VM &mdash; on 10.8.0.58 = 194-deb11-BIG-0221:

```
root@box:~# ansible-galaxy collection list

# /usr/share/ansible/collections/ansible_collections
Collection           Version
-------------------- -------
ansible.posix        1.1.1
community.general    2.1.1
community.mysql      1.2.0
community.postgresql 1.1.1
```

Ref: PR #2685 "Recommend ansible-base 2.10.6"

holta, 2021-04-06:
Just FYI the latest Ansible Collections (for IIAB) have all jumped forward...to:

```
root@box:~# ansible-galaxy collection list

# /usr/share/ansible/collections/ansible_collections
Collection           Version
-------------------- -------
ansible.posix        1.2.0
community.general    2.4.0
community.mysql      1.3.0
community.postgresql 1.2.0
```

holta, 2021-04-19:
Ansible Collection [community.mysql](https://github.com/ansible-collections/community.mysql) has made the big jump to 2.x:

```
# /usr/share/ansible/collections/ansible_collections
Collection           Version
-------------------- -------
ansible.posix        1.2.0
community.general    2.5.1
community.mysql      2.0.0
community.postgresql 1.2.0
```

Ref: PR #2488 "MariaDB [rework security, eliminate {{ mysql_root_password }}]"

holta, 2021-04-28:
FYI here's the latest today:

```
# ansible-galaxy collection list

# /usr/share/ansible/collections/ansible_collections
Collection           Version
-------------------- -------
ansible.posix        1.2.0
community.general    3.0.0
community.mysql      2.1.0
community.postgresql 1.2.0
```

(FWIW running the released version of ansible-core 2.11.0 on BIG-sized IIAB on 64-bit RaspiOS on RPi 4.)

holta, 2021-05-19:
Latest is... (after fresh-installing PR #2743 onto Ubuntu Server 20.04)

```
root@box:~# ansible-galaxy collection list

# /usr/share/ansible/collections/ansible_collections
Collection           Version
-------------------- -------
ansible.posix        1.2.0
community.general    3.1.0
community.mysql      2.1.0
community.postgresql 1.3.0
```

holta, 2021-07-18:
Latest is... (while fresh-installing PR #2850 onto Debian 11)

```
root@deb11-desk:~# ansible-galaxy collection list

# /usr/share/ansible/collections/ansible_collections
Collection           Version
-------------------- -------
ansible.posix        1.2.0
community.general    3.3.2
community.mysql      2.1.0
community.postgresql 1.4.0

holta, 2021-08-10:
Latest is...

```
root@u20-srv:~# ansible-galaxy collection list

# /usr/share/ansible/collections/ansible_collections
Collection           Version
-------------------- -------
ansible.posix        1.2.0
community.general    3.5.0
community.mysql      2.1.0
community.postgresql 1.4.0
```

holta, 2021-08-14:
```
root@u20-srv:/opt/iiab/iiab# ansible-galaxy collection list

# /usr/share/ansible/collections/ansible_collections
Collection           Version
-------------------- -------
ansible.posix        1.3.0
community.general    3.5.0
community.mysql      2.1.1
community.postgresql 1.4.0