iiab/iiab #2665: iptables moved in ubuntu-20 to /usr/sbin/iptables

georgejhunt, 2020-12-26:
But iiab-gen-iptables  in roles/network/templates/gateway/iiab-gen-iptable says:
{% if is_debuntu %}
IPTABLES=/sbin/iptables
IPTABLES_DATA=/etc/iptables.up.rules
{% else %}
IPTABLES=/usr/sbin/iptables
IPTABLES_DATA=/etc/sysconfig/iptables
{% endif %}

This surfaces because gateway_enabled is no longer working on my development machine. Maybe just a symbolic link from /sbin/iptables to /usr/sbin/iptables (that's what I did to fix in the short term)

tim-moody, 2020-12-26:
on my x86 ub20.10 /sbin is linked to /usr/sbin and /sbin/iptables -S returns results

holta, 2020-12-26:
@georgejhunt should this be marked as a bug?

tim-moody, 2020-12-26:
I get the same result on my ub 20.04 vm

holta, 2020-12-26:
VM 191-u2004-srv-BIG-1226 = 10.8.0.50 is now installing -- feel free to log in there if it helps?

holta, 2020-12-26:
> VM 191-u2004-srv-BIG-1226 = 10.8.0.50 is now installing -- feel free to log in there if it helps?

Just FYI on the above Ubuntu Server 20.04.1 (where BIG-sized IIAB has now finished installing) we have:

```
root@u20-srv:~# ls -l /sbin/iptables
lrwxrwxrwx 1 root root 26 Jul 31 12:29 /sbin/iptables -> /etc/alternatives/iptables
root@u20-srv:~# ls -l /usr/sbin/iptables
lrwxrwxrwx 1 root root 26 Jul 31 12:29 /usr/sbin/iptables -> /etc/alternatives/iptables
root@u20-srv:~# ls -l /bin/iptables
ls: cannot access '/bin/iptables': No such file or directory
root@u20-srv:~# ls -l /usr/bin/iptables
ls: cannot access '/usr/bin/iptables': No such file or directory

root@u20-srv:~# more /etc/issue.net
Ubuntu 20.04.1 LTS
root@u20-srv:~# uname -a
Linux box.lan 5.4.0-58-generic #64-Ubuntu SMP Wed Dec 9 08:16:25 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
```

holta, 2020-12-27:
@georgejhunt & @tim-moody: can you review proposed fix PR #2667, if you have time?

tim-moody, 2020-12-28:
>This surfaces because gateway_enabled is no longer working on my development machine,

From @jvonau's analysis this is the primary issue and the missing symlink is not the root cause, but rather a change in the behavior of netfilter-persistent save causes the masquerade rule not to be saved. This is what #2667 addresses.

holta, 2021-04-13:
Appears solved by PRs #2727 and #2728.

(Feel free to re-open as nec.)

closed 2021-04-13 by holta