iiab/iiab #2669: iiab 7.2 installation stuck on "Add 'runtime' variable values" Rpi 4 4GB Raspberry Pi OS 32-bit

chjohn22, 2021-01-28:
### Expected behavior and actual behavior:
After running the one-liner install, iiab should install all components. Instead, it does abort the installation after some minutes. Details below.
### Steps to reproduce the problem:
execute the one-liner, do not use custom .yaml, press enter to aknowledge that you can continue with `sudo iiab`, wait for the install to stop

### IIAB version, OS version, hardware details:
7.2, Raspberry Pi OS 32-bit with desktop and recommended software, Raspberry Pi 4 4GB, 256GB microSD
### Please run the `iiab-diagnostics` command, to generate a URL summarizing your IIAB configuration, for volunteers seeking to help you: (Include that short URL here, Thanks!)

Uploading to web pastebin does not work: `PUBLISHING TO URL... /usr/bin/iiab-diagnostics: Zeile 242: pastebinit: Kommando nicht gefunden`

The installation does run after executing: ` curl d.iiab.io/install.txt | sudo bash` but does not continue after the task named
`TASK [0-init : Add 'runtime' variable values to /etc/iiab/iiab.ini] ************`

Log:

`TASK [0-init : Add 'runtime' variable values to /etc/iiab/iiab.ini] ************
ok: [127.0.0.1] => (item={'option': 'iiab_stage', 'value': '0'})
ok: [127.0.0.1] => (item={'option': 'iiab_base_ver', 'value': 7.2})
ok: [127.0.0.1] => (item={'option': 'iiab_revision', 'value': 0})
ok: [127.0.0.1] => (item={'option': 'runtime_branch', 'value': 'master'})
ok: [127.0.0.1] => (item={'option': 'runtime_commit', 'value': 'f46d469eb7055176b94d8149d1798a28f40d78b2'})
changed: [127.0.0.1] => (item={'option': 'runtime_date', 'value': '2021-01-28T21:53:53Z'})
ok: [127.0.0.1] => (item={'option': 'ansible_version', 'value': '2.10.5'})
ok: [127.0.0.1] => (item={'option': 'kernel', 'value': '5.4.83-v7l+'})
ok: [127.0.0.1] => (item={'option': 'memory_mb', 'value': 3827})
ok: [127.0.0.1] => (item={'option': 'swap_mb', 'value': 99})
**failed: [127.0.0.1] (item={'option': 'product_id', 'value': ''}) => {"ansible_loop_var": "item", "changed": false, "item": {"option": "product_id", "value": ""}, "msg": "Parameter 'value' must not be empty if state=present and allow_no_value=False"}**
ok: [127.0.0.1] => (item={'option': 'gw_active', 'value': True})
ok: [127.0.0.1] => (item={'option': 'internet_available', 'value': True})
ok: [127.0.0.1] => (item={'option': 'rpi_model', 'value': 'Raspberry Pi 4 Model B Rev 1.2'})
ok: [127.0.0.1] => (item={'option': 'first_run', 'value': True})
ok: [127.0.0.1] => (item={'option': 'local_tz', 'value': 'Europe/Zurich'})
ok: [127.0.0.1] => (item={'option': 'no_NM_reload', 'value': False})
ok: [127.0.0.1] => (item={'option': 'is_F18', 'value': False})
ok: [127.0.0.1] => (item={'option': 'FQDN_changed', 'value': False})

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=30   changed=6    unreachable=0    failed=1    skipped=7    rescued=0    ignored=0`

I even tried it on my Pi 400, did wipe the microSD multiple times, did expand the file-system and set the locale and timezones as well as the wifi-country. The Pi is connected by ethernet and no DNS queries get blocked.

holta, 2021-01-28:
Let's investigate if Ansible 2.10.5 might suddenly no longer permit `{{ ansible_product_uuid }}` ?

https://github.com/iiab/iiab/blob/master/roles/0-init/tasks/main.yml#L140-L141

chjohn22, 2021-01-28:
> Let's investigate if Ansible 2.10.5 might suddenly no longer permit `{{ ansible_product_uuid }}` ?
> 
> https://github.com/iiab/iiab/blob/master/roles/0-init/tasks/main.yml#L140-L141

Can I help with something?

holta, 2021-01-28:
@chjohn22 if possible can you run:

```
cd /opt/iiab/iiab/scripts
sudo ./iiab-diagnostics
```

What's the resulting URL, that contains springer.us ?

chjohn22, 2021-01-28:
It does not upload, as I mentioned in the OP.
Error:
`PUBLISHING TO URL... ./iiab-diagnostics: Zeile 242: pastebinit: Kommando nicht gefunden
transl: PUBLISHING TO URL... ./iiab-diagnostics: Zeile 242: pastebinit: command not found`

holta, 2021-01-28:
> It does not upload, as I mentioned in the OP.
> Error:
> `PUBLISHING TO URL... ./iiab-diagnostics: Zeile 242: pastebinit: Kommando nicht gefunden`
> 
> `transl: PUBLISHING TO URL... ./iiab-diagnostics: Zeile 242: pastebinit: command not found`

Ok!  First run the following:

```
sudo su -
apt -y install pastebinit
```

chjohn22, 2021-01-28:
Sorry, yes I did install it a minute ago:
http://sprunge.us/wgPvzE

holta, 2021-01-28:
@chjohn22

Does everything just work if you comment out lines 140-141 within `/opt/iiab/iiab/roles/0-init/tasks/main.yml` and then re-try `sudo iiab` ?

chjohn22, 2021-01-28:
@holta The installation does run atm, seems as if it is working fine with the uncommented lines. I will post again if there is an issue with the installation. Thanks for developing and maintaining this great project, I appreciate it!

holta, 2021-01-28:
@chjohn22 commenting out those 2 lines helped?

(Or did not help?)

Thanks!!

chjohn22, 2021-01-28:
Yeah, I did comment 140 and 141 out, the installation does run smooth now.
I look forward to the stable Release of 7.2, the new SatMaps are amazing!

holta, 2021-01-28:
Thanks much @chjohn22 for the feedback.

We will continue to improve map + satellite photo options.

chjohn22, 2021-01-29:
It did stop on the moodle installation:
`TASK [moodle : Install Moodle if 'moodle_installed' not defined, e.g. in /etc/iiab/iiab_state.yml] ********************************
fatal: [127.0.0.1]: FAILED! => {"reason": "couldn't resolve module/action 'postgresql_user'. This often indicates a misspelling, missing collection, or incorrect module path.\n\nThe error appears to be in '/opt/iiab/iiab/roles/moodle/tasks/install.yml': line 109, column 3, but may\nbe elsewhere in the file depending on the exact syntax problem.\n\nThe offending line appears to be:\n\n\n- name: Create PostgreSQL db user Admin/changeme\n  ^ here\n"}`
In this state, the box is not accessible by http://box.lan, http://box or IP w/wo /admin. Always nginx 404, a reboot did not help.

holta, 2021-01-29:
@chjohn22 a quick workaround is to run:

```
sudo ansible-galaxy collection install community.postgresql
```

(And then continue with `sudo iiab`)

Another change resulting from Ansible changing underneath us, my apologies.

holta, 2021-01-29:
FYI both bugs should now be fixed, as a result of PRs #2670 and #2671.

closed 2021-01-29 by holta

chjohn22, 2021-01-29:
@holta Got another one:
`TASK [network : Add 'detected_network' variable values to /etc/iiab/iiab.ini] *****************************************
changed: [127.0.0.1] => (item={'option': 'has_ifcfg_gw', 'value': 'none'})
**failed: [127.0.0.1] (item={'option': 'prior_gateway_device', 'value': ''}) => {"ansible_loop_var": "item", "changed": false, "item": {"option": "prior_gateway_device", "value": ""}, "msg": "Parameter 'value' must not be empty if state=present and allow_no_value=False"}**
changed: [127.0.0.1] => (item={'option': 'dhcpcd_result', 'value': 'enabled'})
changed: [127.0.0.1] => (item={'option': 'network_manager_active', 'value': False})
changed: [127.0.0.1] => (item={'option': 'systemd_networkd_active', 'value': False})
changed: [127.0.0.1] => (item={'option': 'wan_in_interfaces', 'value': False})
changed: [127.0.0.1] => (item={'option': 'wireless_list_1(wifi1)', 'value': 'not found-1'})
changed: [127.0.0.1] => (item={'option': 'wireless_list_2(wifi2)', 'value': 'wlan0'})
changed: [127.0.0.1] => (item={'option': 'num_wifi_interfaces', 'value': '1'})
changed: [127.0.0.1] => (item={'option': 'discovered_wireless_iface', 'value': 'wlan0'})
changed: [127.0.0.1] => (item={'option': 'discovered_wired_iface', 'value': 'none'})
changed: [127.0.0.1] => (item={'option': 'exclude_devices', 'value': 'none'})
changed: [127.0.0.1] => (item={'option': 'num_lan_interfaces', 'value': '1'})
changed: [127.0.0.1] => (item={'option': 'gui_static_wan', 'value': False})
changed: [127.0.0.1] => (item={'option': 'iiab_lan_iface', 'value': 'br0'})
changed: [127.0.0.1] => (item={'option': 'iiab_wan_iface', 'value': 'eth0'})

PLAY RECAP ************************************************************************************************************
127.0.0.1                  : ok=423  changed=215  unreachable=0    failed=1    skipped=106  rescued=0    ignored=0   
`

holta, 2021-01-29:
Thanks @chjohn22  and @jvonau !

Discussion as to how to cleanly address ansible-base 2.10.5's sudden changes in behavior are ongoing here: #2670

jvonau, 2021-01-29:
From an older install:
>ubuntu@box:/opt/iiab/iiab$ ansible-galaxy collection list
\# /usr/share/ansible/collections/ansible_collections
Collection           Version
\-------------------- -------
ansible.netcommon    1.4.1  
ansible.posix        1.1.1  
community.general    1.3.0  
community.kubernetes 1.1.1  
community.mysql      1.1.1  
google.cloud         1.0.1  

`cd /opt/iiab/iiab`
`sudo scripts/ansible`

>ubuntu@box:/opt/iiab/iiab$ ansible-galaxy collection list
\# /usr/share/ansible/collections/ansible_collections
Collection           Version
\-------------------- -------
ansible.netcommon    1.4.1  
ansible.posix        1.1.1  
community.general    2.0.0  
community.kubernetes 1.1.1  
community.mysql      1.2.0  
google.cloud         1.0.1  

Interesting, according to https://github.com/ansible-community/ansible-build-data/blob/main/2.10/CHANGELOG-v2.10.rst#changed-collections the version of community.general should be at 1.3.5. I can confirm this new behavior with version 2.0.0, to check the installed version use `ansible-galaxy collection list` but the bigger question is why the difference in the version of community.general that is installed vs the release docs.

holta, 2021-01-29:
> why the difference in the version of community.general that is installed vs the release docs

I noticed that yesterday.

How do we say this politely...the quality (and timeliness) of Ansible documentation's has never been high..