iiab/iiab #2673: Feature proposition: selfhosted F-Droid Mirror

chjohn22, 2021-01-30:
It is nice if you can install a small set of apps, even when the internet does not work. Android smartphones get cheaper by the day and there are many lying around. F-Droid offers an interesting amount of well curated and secure apps, which can complement the goals, iiab tries to achieve. There is for example the Kiwix-App, Osmand as well as thousands of other useful tools, and many of them can do their job while being offline, too. F-Droid allows to build simple mirrors of their "repo" and "archive" repository. The main "repo" repository is manageable in size and I am convinced that it would be a great optional feature of IIAB. [The guide](https://f-droid.org/en/docs/Running_a_Mirror/) is simple and I think it should be technically possible to implement these things in IIAB. What do you think about it?

holta, 2021-01-30:
@m-anish has also supported this idea: it would just need someone to pull it together and make it happen!

Cultural Context: while schools and medical clinics generally don't need an F-Droid "app store" on their IIAB, and (wisely?) might even be concerned about infosec risks/liability...the general/grassroots community would indeed appreciate this, especially if the "repo" prominently highlights recent versions of the 100-or-so best Android apps.

chjohn22, 2021-01-30:
@holta I think it could even be great for schools. You could provide the Android clients for Nextcloud with F-Droid, to connect with the Nextcloud instance on IIAB. With advanced sharing features the IIAB could even be a hub for sharing the installed Kiwix ZIM-files with Android devices. As F-Droid does allow to share locally downloaded APKs with other clients over Wifi and Bluetooth, you could spread apps effectively offgrid.

holta, 2021-01-30:
Excellent point that many schools would benefit by distributing a curated (and "security/privacy-verified") small set of Android apps.

And F-Droid could very well be the best way to make this happen...for schools and others too!

chjohn22, 2021-01-30:
F-Droid does provide the settings to add custom repos too. You could define a subset of Apps to be included on IIAB instead of the standard "repo" F-Droid repository. Preconfigured repos ready to be mirrored by IIABs could be hosted on various platforms, I myself would be glad to host some custom repos on my servers.

tim-moody, 2021-01-31:
The medical version of IIAB does this, though without using F-Droid. The Kiwix android app is included with a link for sideloading, and installed zims can be loaded into it.