iiab/iiab #2679: Extending instance space by binding USB drives

Mortvert, 2021-02-13:
Hello,
Is it possible to use a USB stick/pendrive/drive/disk to extend the usable space of the instance?

I currently have a 16GB microsd in my raspberry pi but I do have a 128GB pendrive that I could use to extend the usable space.
For example, I wanted to download entire world map region which is ~76 GB
I do not own a microsd card big enough to download that content but I could bind the pendrive to contain all the maps. 
The pendrive itself would have to stay plugged in all the time but that does not seem like a problem.

How would one go about doing it? 
Thanks!

holta, 2021-02-13:
Personally I'd avoid this approach for reasons of USB device theft and fragility, but it's been done before and general guidance is here if necessary:

http://FAQ.IIAB.IO -> "43 Can I permanently attach an external USB drive, to add more content?"

jvonau, 2021-02-14:
#2614 @Mortvert, What filesystem (ntfs, ext4, etc..) is on the usbkey? It's doable just rsync /library to the root of the usbdrive, have the usbkey mount in a known location on boot, and bindmount /usbkey_mnt/library overtop of /library. Kind of like the FAQ 43 but without creating the symlink. I have rigged up usbmount to provide the auto bind mounting but works reliability with a single usbdrive if you are interested.

Mortvert, 2021-02-14:
> 
> 
> Personally I'd avoid this approach for reasons of USB device theft and fragility, but it's been done before and general guidance is here if you have Linux expertise and strongly prefer to split your disk (internal & external devices) in such ways:
> 
> http://FAQ.IIAB.IO -> "43 Can I permanently attach an external USB drive, to add more content?"

Ah, I did not notice that in FAQ. Sorry for being less than smart.
Shame it is not an option in the admin panel for the less tech-savy people.

@jvonau I formatted the USB key to EXT4 - fat32 will not fit that one big .zim file for english wikipedia.

jvonau, 2021-02-14:
> Shame it is not an option in the admin panel for the less tech-savy people.
> 
> @jvonau I formatted the USB key to EXT4 - fat32 will not fit that one big .zim file for english wikipedia.

The filesystem has everything to do with limits of what that can be done, with EXT4 the FAQ's method of placing /library on the external drive will work fine but temporarily bind mounting directories in place of permanently moving /library would work also.

tim-moody, 2021-02-14:
I think this is a case of just because you can doesn't mean you should. But if it is the only option or for experimental use it is not difficult to do.

rsync -av /library/ <your device>
edit /etc/fstab and add the line
/dev/<your device partition> /library ext4 defaults,nofail 0 0
test it by issuing mount -a
if it works should be able to reboot

jvonau, 2021-02-14:
On should really list the usb device via LABLE or UUID in fstab, booting with 2 usb drives present could upset the ordering, \/dev/sda may become \/dev/sdb. 

The advantage of bind-mounting is one can mix and match locations, some individual files can live within \/library on the SDcard while files on the usbdrive are introduced within \/library without actually residing within \/library. Not permanent, easily reversible, without moving some of database files and the rest of \/library are some of the advantages so one does not fully lose functionality if the usbkey holding \/library is not present, just the apps that are missing the files to serve.

Mortvert, 2021-02-18:
> Personally I'd avoid this approach for reasons of USB device theft and fragility, but it's been done before and general guidance is here if you have Linux expertise and strongly prefer to split your [...]

If it's easy to steal the USB drive - you might as well grab the entire device (if it is a raspberry pi or similar) so it's not really that much of a problem IMO.
Am asking because it is cheaper for me to grab a 128GB USB stick than 128GB microsd card to play around with.

@tim-moody will test that as soon as I'm able and report back.