iiab/iiab #2680: KA Lite no longer actively developed.

Mortvert, 2021-02-18:
Hello again.

No idea if this is known (I did search issues and found nothing) but apparently KA Lite is being replaced by Kolibri (as per https://kalite.learningequality.org/)

Are there any plans to replace KA Lite entirely in IIAB?

Please close this issue if it is irrelevant to the project as a whole, thank you and sorry if that's the case.

holta, 2021-02-18:
KA Lite is much more compact and lightweight as compared to Kolibri.

IIAB continues to support both, as they each serve a very distinct and valuable purpose.

Those who have lots of disk and CPU and are fortunate enough to have a large budget for Curricular Alignment generally prefer Kolibri.

Conversely many others strongly prefer KA Lite, as they need reliable access to Khan Academy, with a very small budgetary footprint.

Again, each of these 2 products delivers an incredibly valuable service — but in a very different way — depending on the specific field need.

holta, 2023-04-09:
FYI KA Lite is unlikely to make sense for anybody in 2030 — as Kolibri & similar bring far more modern UX, and increasingly far more modern content!

But just FYI @Mortvert, KA Lite has been given a small lease on life here, in case that helps you and others in the meantime:

- PR #3527

holta, 2023-04-14:
> FYI @Mortvert, KA Lite has been given a small lease-on-life here, in case that helps you and others in the meantime:
> 
> * PR [Python2 fix [for KA Lite on Debian 12 or Ubuntu 23.04 / 23.10] #3527](https://github.com/iiab/iiab/pull/3527)

FYI further refined by:

- PR #3535

holta, 2023-05-25:
Further refined (for Ubuntu 23.10+) by:

- PR #3582
- PR #3587