iiab/iiab #2695: What's the best way to slim down ZIM videos?

georgejhunt, 2021-02-28:
Target is top learning videos in ~10G ZIM files: 
how do we get there?  
In rough order of priority the top 3 are:
1. farm.openzim.org/recipes/teded_en_all (26G)
2. farm.openzim.org/recipes/crashcourse_en_all (35G)
3. farm.openzim.org/recipes/cest-pas-sorcier_fr_all (74G)

### Proposal for how we get there ###

1. Zimdump to file system tree.
2. De-NameSpace, and verify that links still work
3. Based upon the direct relationship with youtube, fetch the "view_count", and merge with "Zim_size" -- perhaps in a sqlite file.
4. In sql:"select sum(Zim_size) from data where view_count > first_cut".
5. Keep on changing "first_cut" until the summed size ~ 10GB.
6. Cut out, and prune playlist (if necessary) minus the videos that did not get selected vi sql above
7. apply zimwritterfs to the new reduced file tree (hopefully autogenerating index properly).
8. Might be possible to re-compress, lowering data rate or frame rate, before applying zimwritterfs.

holta, 2021-02-28:
1) This will be an absolute Godsend.  These very compact collections of popular + approachable science videos are urgently needed by kids, and not just in poor countries.  Also in Spanish and other languages too — e.g. @thotmx suggests we try to get permission to do the exact same thing with https://en.wikipedia.org/wiki/Beakman%27s_World (as these videos already exist in Spanish, sorely needed !)

2) It does not matter if the ordered list of individual videos is criticized by different communities during initial months.

Regardless whether videos are ordered/ranked/prioritized by quality, by
popularity, by pedagogical/curricular relevance, by child's age, by science/naturw topic, in chronological order — or even if randomly ordered in initial ZIM files.

In fact: this will be the most constructive feedback of all — crucial to evolving these ~10 GB extracts — allowing these subset selections to become more-and-more valuable Learning Packs to kids and educators as every year goes by.

Likewise see this monthly scoring (ordered list) of English Wikipedia's 6.3 Million articles that is published every month — which is incredibly valuable — no matter its many flaws:

http://download.openzim.org/wp1/enwiki_2021-02/tops/
http://download.openzim.org/wp1/enwiki_2021-02/scores.tsv.zip
https://github.com/openzim/wp1_selection_tools

3) It does not matter if grassroots educators later prefer 2GB or 20GB extracts.  Or if these entire extracts are later rebundled into their own peer-redistributable Android APK apps etc.

The key is that we're able to very rapidly iterate — delivering 10GB (or such, very compact, very compelling) learning vid extracts for Learning Cultures/Communities around the entire planet.

georgejhunt, 2021-02-28:
I can pare down TedED from 1731 videos to about 900  that will occupy about 10GB if I choose videos that have been viewed on youtube more than 500000 times.

tim-moody, 2021-02-28:
What is the advantage of putting TED videos in a ZIM?

holta, 2021-02-28:
> I can pare down TedED from 1731 videos to about 900 that will occupy about 10GB if I choose videos that have been viewed on youtube more than 500000 times.

That would be absolutely incredible.

If this works, we should do likewise for C'est Pas Sorcier (a French one if needed) and for CrashCourse.

> What is the advantage of putting TED videos in a ZIM?

Kiwix's current ZIM files are dysfunctionally large (is there a more polite word for dramatically overweight?) eliminating them from practical contention in most every real-world microSD low-income scenario.

Tragic as popular science-like video content is exactly what's needed in almost every such country, in schools and outside of schools too.  In the end it's the "hoarding mentality" (in fact hoarding disorder is in fact classified as an OCD medical condition by many psychologists) and that's may be fine for the affluenza wealthy &mdash; but unfortunately it's destroying viable community action in non-affluent (e.g. microSD not disk) communities.

> Does search actually work or is it just not working on the old zims on iiab.me?

Hopefully search works, but time will tell after these are assembled.  ASIDE: Search is far less important in a YouTube-like use experience, so long as similarly interesting videos are easily accessible.  However for teachers and parents, search will be more important if it hopefully works later, possibly including paragraphs + metadata about each video etc.

holta, 2021-02-28:
@darkenvy can you review the above strategy, to see if there's any general approaches or implementation tricks possibly overlooked?  Thanks Much!!

holta, 2021-02-28:
> (More Generally: the advantage of packaging up these collections of videos into ZIM files is that this provides a very decent auto-generated navigation interface for videos, that is far superior to flat directories — and avoids the high ongoing maintenance costs of manually-created-navigation with RACHEL/OER2Go modules.)

Of course @georgejhunt's selection of the 900 most popular video[*] is the key.  If for example Kolibri (of Moodle, or WordPress, or whatever) deliver a friendlier UX than Kiwix in future, that's 100% fine too!

[*] Regardless if in future this changes to the ~900 most age-appropriate vids, or the ~900 most recent, or the ~900 most related to environmental science, or whatever~

georgejhunt, 2021-02-28:
* Adam, are you saying that you think the index that Kiwix generates for youtube videos (TedED is only one I have examined) is preferred to the playlist navigation provided in the zim? 
* If so, a lot of the work to prune the data.json monster file goes away. If we were to modify the landing page, such that it did not have the playlist, what would it look like?

holta, 2021-02-28:
> * Adam, are you saying that you think the index that Kiwix generates for youtube videos (TedED is only one I have examined) is preferred to the playlist navigation provided in the zim?
> * If so, a [...]

The more Visual, Simple, Friendly the UX, the better.  I assumed that meant video ZIM files' current UX, as designed by Kiwix apparently, but perhaps I'm wrong? 

Certainly visual thumbnails (hinting at the contents of all 30-or-so videos on each page of the ZIM file) are VERY important.  (Certainly FTP-style listings of filenames are no longer acceptable to schools in 2020s.)

> * Maybe we'd just  select a generally interesting search term into the index search. (I wonder what that search term would be).

Can you explain a bit more?  While not the most urgent need, specialized subsets of C'est Pas Sorcier...e.g. videos related to environmental science...e.g. removal of the nationalistic French military videos...might be very useful in the future.

georgejhunt, 2021-02-28:
* I have so much to learn. Tim seems to suggest that Kiwix indexing of videos does not work on older ZIM's. We may need to understand how the xapian indexer gets its source, and tweak it a bit such that it works correctly for videos.
* I'm trying to conceive of ZIM reformatting in the context of a jupyter notebook, because I think we need to get out of the job of creating one off ZIM's for people with special requests. To facilitate this, I need to supply tools which summarize all of the available characteristics of each video, and accept a list of youtube ID's to be included in the package.

tim-moody, 2021-02-28:
should probably discount what I said about the zims as they are from 2018. We should update them on iiab.me

holta, 2021-02-28:
> I think we need to get out of the job of creating one off ZIM's for people with special requests

It's the opposite of isolated special requests in this case.

It's the universal request from essentially every school around the planet for modern science-like videos, in their own language — bundled into 1 compact Learning Pack — to avoid bureaucracy.

Such tools (or recipe) will be so valuable to fulfill this essentially universal global need, that these exact tools will almost inevitably later be applicable to other YouTube subsets — repackaging the best learning vids from other less-science-centric domains, likewise into ZIM files and Android apps (or Kolibri Channels, etc) of 1-to-10GB for myriad different kinds of home schooling, orphanages, prisons, etc.

holta, 2021-03-05:
1) Huge Thanks @georgejhunt for all the progress in https://github.com/georgejhunt/iiab-factory/blob/zimtools/content/kiwix/ted-tools/zimshrink.md &mdash; coincidentally the top 1000 TED-Ed videos make up almost exactly 10G which is perfect!  (FWIW these top 1000 videos average 10.3M each, while the bottom 762 videos average 19.3M each.)

2) What's the easiest way to find out the total number of videos in a ZIM files like the 5 below and/or directly from their YouTube channel (playlists?) and/or ZIMFARM command/recipe?  In the past I could view the ZIM file in a browser, then multiply the number of pages by ~40.  But that no longer works, now that categories are the predominant UX of such ZIM files scraped from YouTube...as shown in each of these ZIM files below having an (all too long in some cases!) popup menu of categories in the top-right:

   - http://iiab.me/kiwix/teded_en_all_2021-01/ (27G)
   - http://iiab.me/kiwix/crashcourse_en_all_2021-01/ (36G)
   - http://iiab.me/kiwix/cest-pas-sorcier_fr_all_2021-01/ (74G)
   - http://iiab.me/kiwix/la_chaine_de_maths_et_tiques_fr_all_2021-01/ (42G)
     - from [farm.openzim.org/recipes/la_chaine_de_maths_et_tiques](https://farm.openzim.org/recipes/la_chaine_de_maths_et_tiques/config)
   - http://iiab.me/kiwix/universcience-tv_fr_all_2021-01/ (70G)
     - from [farm.openzim.org/recipes/universcience-tv_fr_all](https://farm.openzim.org/recipes/universcience-tv_fr_all/config)

   I don't yet understand how such ZIM files' categories arise, and how they work in general, but hopefully @georgejhunt can explain more in coming weeks.

3) A key number will be the size (in GB) of the top 1000 videos for any high-quality collection of videos, like the above 5, and hopefully Spanish etc one day soon too!

   _Spanish language collections of science-like videos are very urgently needed (and other languages too of course...please if anybody runs into any solid options here contact us!!)_

4) Kiwix's scores.tsv files published monthly at http://download.openzim.org/wp1/ are an absolute Godsend, allowing automated trimming down of English Wikipedia (and eventually other Wikipedias) to ~10G and similar subsets.  Can we learn any vital lessons from Kiwix's http://download.openzim.org/wp1/enwiki_2021-02/README pasted in here?

   > pageviews.tsv: page_title view_count
pages.tsv: page_id page_title page_size is_redirect
pagelinks.tsv: source_page_id target_page_title
langlinks.tsv: source_page_title language_code target_page_title
redirects.tsv: source_page_id target_page_title
ratings.tsv: page_title project quality importance
vital.tsv: level page_title
all.tsv: page_title page_id page_size pagelinks_count langlinks_count pageviews_count [rating1] [rating2] ...
scores.tsv: page_title score
tops/*tsv: page_title (one file per TOP selection)
projects/*tsv: page_title (one file per project)
customs/*tsv: page_title (one file per custom selection)

5) Fully automated ~10GB ZIM file creation (or 1000 top videos may be preferable in many cases, as people very much appreciate "Top 10" lists) is the priority so schools can move beyond bureaucracy in 2021, using these right away.

   While NOT the priority today, in later years it's likely that polished/smaller subsets for Kolibri Channels (and/or for other LMS's like Moodle, so students/teachers can comment on very tight videos Learning Packs, Android APK's etc) will be demanded &mdash; by folks who lack disk/finance/electricity &mdash; and most importantly, time.

holta, 2021-03-05:
> 2. What's the easiest way to find out the total number of videos in a ZIM file

@georgejhunt is the _recent popularity_ of any video available?
e.g. total views over the past year?
(Rather than tabulating views across all of time &mdash; so as not to be biased in favor of older videos &mdash; that have been in existence for a decade or more?)

holta, 2021-03-05:
@georgejhunt how might we identify very important/popular/high-quality _new videos_ (e.g. from https://www.ted.com/talks?q=ted-ed&sort=newest or https://www.youtube.com/playlist?list=PLJicmE8fK0EiEzttYMD1zYkT-SmNf323z or similar?)

e.g. that have not yet appeared within Kiwix's sometimes stagnant ZIM file production pipeline, that can often be many months out of date?  (Sometimes even a year or more, which is only natural and no reason to blame their many "industrial" workflows!)

holta, 2021-03-05:
> how might we identify very important/popular/high-quality new videos published in recent months

Quick Example:

- [ted.com/talks/alex_gendler_building_the_world_s_largest_and_most_controversial_power_plant](https://www.ted.com/talks/alex_gendler_building_the_world_s_largest_and_most_controversial_power_plant) shows "377,700 views"
- It's also available at [youtu.be/dcZ0BXJYlUA](https://youtu.be/dcZ0BXJYlUA) which shows "357,710 views"

Either way, this 3-month-old video would appear to be among the 1000 most popular TED-Ed videos of all time, despite not appearing at all within the ranking of 1762 videos listed at the bottom of https://github.com/georgejhunt/iiab-factory/blob/zimtools/content/kiwix/ted-tools/zimshrink.md

Note the above video appears within the latest Kiwix TED-Ed ZIM file from 2021-01-16:
http://iiab.me/kiwix/teded_en_all_2021-01/A/building_the_world_s_largest_and_most_controversial_power_plant_alex_gendler.html

Despite the fact that search is unfortunately broken for the above video (autocomplete appears to work, but is not clickable, perhaps due to the apostrophe) &mdash; even with the latest kiwix-serve 3.1.2-4 here: http://iiab.me/kiwix/teded_en_all_2021-01/

holta, 2021-03-05:
@georgejhunt I certainly don't understand, but do you have any idea why the 1762 videos listed at the bottom of https://github.com/georgejhunt/iiab-factory/blob/zimtools/content/kiwix/ted-tools/zimshrink.md make up less than 50% of TED-Ed's total videos?

(If one takes at face value the "3600+ talks to stir your curiosity" advertised at the top of https://www.ted.com/talks?q=ted-ed ?)

There seems to be a serious discrepancy here, when ZIM files like http://download.kiwix.org/zim/ted/teded_en_all_2021-01.zim are named "all" as if they're supposed to...contain _all_ TED-Ed videos ??

holta, 2021-03-06:
@georgejhunt if the average video length (in min:sec) can be calculated, that metadata will be extremely useful for teachers.

e.g. if hypothetically this (average min:sec) turns out to something we can auto-generate — and somehow mention with the final product ZIM files / Learning Packs — e.g. in their "menu def" footnotes etc?

tim-moody, 2021-03-06:
I think the following should be considered in enhancing video collections for IIAB:

https://meta.wikimedia.org/wiki/VideoWiki

holta, 2021-03-06:
A couple examples just to show "inflation" (the polite word for hoarding) is increasingly more out of control year-by-year &mdash; making automated/thoughtful "weight loss programs" especially urgent:

```
root@iiab-me:/library/zims/content# du -h cest* univ*

41G     cest-pas-sorcier_fr_all_2018-06.zim
53G     cest-pas-sorcier_fr_all_2020-04.zim
74G     cest-pas-sorcier_fr_all_2021-01.zim

14G     universcience-tv_fr_all_2015-03.zim
70G     universcience-tv_fr_all_2021-01.zim
```

(Specifically if such quality materials are to be relevant in French-speaking schools across West and North Africa and around the world &mdash; these need to be made available in a far more compact way &mdash; as we work with such communities whenever/wherever possible to face down the eternal struggle against yet more bloat!)

georgejhunt, 2021-03-08:
#### Progress:
* Reordered the video list by views per year, rather than total views (which favors the older ones). See https://github.com/georgejhunt/iiab-factory/blob/zimtools/content/kiwix/ted-tools/zimshrink.md
* I'm wanting to put this project on hold to give Emanuel a chance to fix the indexing problem I discovered: https://github.com/openzim/zim-tools/issues/230.   His response suggests to me that it will get fixed.
* As Tim suggested above, asking essentially "why Zims?", I feel that the index adds value. But absent the indexing, a tar or zip might be almost as good -- maybe more flexible, and easier to learn.

holta, 2021-03-08:
> * Reordered the video list by views per year, rather than total views (which favors the older ones). See https://github.com/georgejhunt/iiab-factory/blob/zimtools/content/kiwix/ted-tools/zimshrink.md

1) According to this new ranking, the top 1000 videos make up 10.8G, which is really great!

   And the top 314 videos make up 3.37G (to celebrate Pi Day 3/14, i.e. March 14th this coming Sunday) which might sounds silly &mdash; but might be right on target &mdash; with the order of magnitude urgently needed by smaller communities, schools and families in dire poverty.

2) @georgejhunt I need to figure out how many videos are contained in http://iiab.me/kiwix/cest-pas-sorcier_fr_all_2021-01/ or https://www.youtube.com/channel/UCENv8pH4LkzvuSV_qHIcslg &mdash; what's the best way to get a solid approximation?  Or is it possibly easier to extract this estimate (or exact number) using any of the following, which are (supposed to be) equivalent?

   - https://farm.openzim.org/recipes/cest-pas-sorcier_fr_all/config
   - http://download.kiwix.org/zim/other/cest-pas-sorcier_fr_all_2021-01.zim

   As I've found some of these numbers to be completely off over the years &mdash; not aligning with https://en.wikipedia.org/wiki/C%27est_pas_sorcier which claims 599 episodes of 26min each.  Possibly these numbers changed in post-production / in later years &mdash; e.g. if the videos were chopped up or repackaged in different ways?

georgejhunt, 2021-03-08:
>    March 14th this coming Sunday) which might sounds silly — but might be
>    right on target — with the order of magnitude urgently needed by smaller
>    years — not aligning with
>    in later years — e.g. if the videos were chopped up or repackaged in

The only way we can really be sure is to download it and zimdump it. Do
your have zimdump?

holta, 2021-03-08:
> The only way we can really be sure is to download it and zimdump it. Do your have zimdump?

Is there a version of zimdump that you'd recommend to put on iiab.me, to get visibility into what's inside ZIM files?

(Tools like zimdump, and ideally your own tools to poll YouTube channels directly, are definitely the missing piece to be able to move forward.)

georgejhunt, 2021-03-08:
The zimwriterfs and zimdump are all in flux right now -- they are part of
the same repo. I recommend that we wait until the dust settles

holta, 2021-03-08:
> The zimwriterfs and zimdump are all in flux right now -- they are part of the same repo. I recommend that we wait until the dust settles

Ok.

At the same time we should try to scope out if a YouTube Channel can be made to cough up (reveal) the very basics &mdash; without the need for any ZIM file &mdash; e.g. how many videos any public YouTube channel (or playlist?) contains at the bare minimum?

(Even if the details can come later -- i.e. how many minutes long + how disk-heavy etc, that YouTube Channel/playlist's vids are on average!)

georgejhunt, 2021-03-08:
> made to cough up (reveal) the very basics — without the need for any ZIM
> file — e.g. how many videos any public YouTube channel (or playlist?)

I zimdumped and then did 'ls 1|wc". That tells me that there are 1853
videos in cest_sorcier

holta, 2021-03-08:
> I zimdumped and then did 'ls 1|wc". That tells me that there are 1853 videos in cest_sorcier

Thanks!!

74GB / 1853 means these videos are likely most/all right around 40MB each.  So the "Pi Day" packaging strategy (314 videos bundled up, as a motivational goal for communities) looks potentially very promising — leading to a polished deliverable vid pack of about 12.5 GB.

_ASIDE:_ As in recent years, the Wikipedia article https://en.wikipedia.org/wiki/C%27est_pas_sorcier continues to be highly misleading.  We should fix it once we're sure, when we make progress here.

georgejhunt, 2021-03-21:
> At the same time we should try to scope out if a YouTube Channel can be made to cough up (reveal) the very basics — without the need for any ZIM file — e.g. how many videos any public YouTube channel (or playlist?) contains at the bare minimum?

I made a list of the channels in TEDed, and then used a python program:
https://github.com/georgejhunt/iiab-factory/blob/zimtools/content/kiwix/ted-tools/channel_list_info.py
 to create:
https://github.com/georgejhunt/iiab-factory/blob/zimtools/content/kiwix/ted-tools/teded_channel_info -- the "#" column is the number of videos in that particular playlist (playlists id's start with "PL", and channel id's with "Ch").

* I suspect that the recent dates on the playbooks correspond more to changing the playbook than to really new videos.

holta, 2021-03-21:
> https://github.com/georgejhunt/iiab-factory/blob/zimtools/content/kiwix/ted-tools/teded_channel_info -- the "#" column is the number of videos in that particular playlist (playlists id's start with "PL", and channel id's with "UC").

The above is an incredible survey of 14 different science/demo-oriented YouTube Channels (with 5 Playlists from each).  With Channels #11 and #12 being TED-Ed.  Quick questions:

- Are the Playlist ID's cut off on Lines 44, 45 and 117 for a reason?  (Or is that some kind of bug, if it matters?)
- Is there an easy way to auto-extract the copyright license e.g. the specific Creative Commons license or similar from a Channel and/or Playlist?  (Or does the license change too much across each individual video, such that there's no unique license for each Channel and Playlist?)

georgejhunt, 2021-03-22:
These channels were the ones in the <zimroot>/I/channels/ directory of the
TEDeD zim file. I suspect the license is moot since Emanuel  has already
taken the risk.

holta, 2021-03-22:
> These channels were the ones in the <zimroot>/I/channels/ directory of the TEDeD zim file.

Are you saying that http://iiab.me/kiwix/teded_en_all_2021-01/ contains videos from outside of TED-Ed ?

(Personally I see only TED-Ed videos, when quickly browsing the link above.)