iiab/iiab #2713: Let's provide a less confusing way for Kolibri communities to save to USB drives

holta, 2021-03-19:
@fisherbryan eventually got it to work at the Command-Line as root, using:

```
umount /media/usb0
mount -o rw,users,umask=000 /dev/sda1 /media/usb0
```

(I don't know if all those -o options are truly needed?  Possibly something as simple as `-o rw` might be enough?)

But FWIW Kolibri is now confirmed working for him — as a result of the above be is now able to save materials from Kolibri to his VFAT-formatted 8 GB USB stick.

IIAB should offer a hassle-free option that does something very similar to the above.  Let's investigate how to do that herein:

https://github.com/iiab/iiab/tree/master/roles/usb_lib

holta, 2021-03-19:
Possibly related concern, unearthed by @jvonau: (is it possible Kolibri 0.14.x doesn't play well with `fuse` e.g. filesystems like VFAT / FAT32 and NTFS ?)

https://github.com/learningequality/kolibri/blame/release-v0.14.x/kolibri/core/discovery/utils/filesystem/posix.py

holta, 2021-03-19:
FYI @jvonau suspects this entire ticket is possibly a non-issue &mdash; that would not arise at all (i.e. the inability to save from Kolibri) if the USB stick/drive had been cleanly unmounted[*] the last time it was used.

@fisherbryan please do confirm/deny that when you can (with multiple confirmed-cleanly-ejected USB sticks if possible!)

holta, 2021-03-19:
https://raspberrypi.stackexchange.com/a/29954 (April 2015) says that `/etc/usbmount/usbmount.conf` affects read-only versus read/write permissions etc &mdash; apparently one uses 'umask=0000' to set rwx-rwx-rwx explicitily in this configuration file &mdash; similar to what Bryan did above at the command-line.

So something like the following in the above `usbmount.conf` might work &mdash; e.g. if there's a desire/need to make the option of read/write versus more-protected media more explicit (and customizable) for IIAB operators:

```
FS_MOUNTOPTIONS="-fstype=vfat,umask=0000"
```

holta, 2021-03-19:
It's possible the `noexec` in Line 38 of `/etc/usbmout/usbmount.conf` is causing the problems that several people claim (is preventing them from writing to USB media).  By default that line is: (on RaspiOS on RPi 4)

```
MOUNTOPTIONS="sync,noexec,nodev,noatime,nodiratime"
```

Thanks to @jvonau for noticing this!

holta, 2021-03-20:
FWIW @fisherbryan says Kolibri can export to USB drive but not import.  This sound serious, but details are needed from him or others:

If anybody can reproduce this, please include the output of `lsblk`, `blkid` and `mount` (as well as [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md)) so that Kolibri developers can assist in understanding and explaining how this might be possible.

holta, 2021-03-20:
> @fisherbryan please do confirm/deny that when you can (with multiple confirmed-cleanly-ejected USB sticks if possible!)

FYI Bryan has reconfirmed that this original failure (#2713) to export is indeed repeatable.

He did that by carefully unmounting a USB stick (to ensure it's clean and without filesystem problems) and then used a freshly-booted IIAB to confirm that he was (once again) unable to export from Kolibri-on-IIAB to that inserted USB stick.

jvonau, 2021-03-20:
>mount -o rw,users,umask=000 /dev/sda1 /media/usb0

Think 'users' is what allows Kolibri's user full access to the device without sudo. I'm going to assume that exporting from Kolibri is where is might of occurred. The next step I would suggest edit /etc/usbmount/usbmount.conf and set ENABLED=0, reboot,and test kolibri standalone noting where and with what permissions the device get mounted, report back. Next try adding FS_MOUNTOPTIONS="-fstype=vfat,users,umask=000" and setting ENABLED=1 in /etc/usbmount/usbmount.conf, reboot, retry Kolibri, report back.

jvonau, 2021-03-20:
> FWIW @fisherbryan says Kolibri can export to USB drive but not import. This sound serious, but details are needed from him or others:

Think that is a mis-spoke, this is a writing issue and is reproducible when not using sudo to write files as a normal user.
>ubuntu@box:~$ touch /media/usb0/test.touch2
touch: cannot touch '/media/usb0/test.touch2': Permission denied

after the above edit and reboot
>touch /media/usb0/test.touch2
ls /media/usb0/test.touch2
/media/usb0/test.touch2

jvonau, 2021-03-20:
This change would be a step backward from a security standpoint, kolibri's user needs to become a member of a group that has the correct privileges to access the filesystem on the usb device (that is why sudo works). https://kolibri.readthedocs.io/en/latest/install/ubuntu-debian.html#changing-the-owner-of-kolibri-system-service "Prior to v0.10, kolibri user account was the owner of the system service, while from v0.10 and later, desktop user’s account is preferred, in order for Kolibri to access the local USB storage."

jvonau, 2021-03-20:
https://kolibri.readthedocs.io/en/latest/install/provision.html
Instead of trying to fix what is not broken, the user just needs to use sudo and just cp /home/kolibri/.kolibri to /media/usbN/ after stopping the kolibri service, needs sudo for that anyway, "kolibri manage deprovision" and reverse the steps to import... Just taking a local backup, then omit the "kolibri manage deprovision" step.

closed 2021-03-25 by holta