iiab/iiab #2720: Admin Consoles GET-EXTDEV-INFO failed when partitioned external device is present

holta, 2021-04-04:
Details: https://github.com/iiab/iiab-admin-console/issues/388

tim-moody, 2021-04-05:
I thought there was consensus that kiwix will always be installed, and Admin Console expects this to be true.

holta, 2021-04-05:
We can set the policy either way.

(Either way things can occasionally get messy!?)

PS I've run into a few communities and NGO's that want IIAB without Kiwix, but these are very rare.

holta, 2021-04-05:
I closed this by accident.

I don't know if Admin Console should require Kiwix or not.

Aside: Certainly kiwix-serve changes quite dramatically approximately annually, but that's a reality that we live with.

holta, 2021-04-08:
Good enough as is.  Yes there are a few IIAB implementers who do not want Kiwix (e.g. Kolibri-only shops) but more generally Kiwix is a soft requirement for those who want to install IIAB.

Certainly IIAB implementers who want to use Admin Console should install Kiwix, as changing Admin Console to be able to work without Kiwix is too much work at this time.

closed 2021-04-08 by holta