iiab/iiab #2735: Question: is iiab.ini still used?

tim-moody, 2021-04-16:
is the data in this file ever used? is it guaranteed to be correct?

holta, 2021-04-16:
Some folk seem to use `/etc/iiab/iiab.ini` visually, to get a quick understanding of what's installed on an IIAB.

As such I've tried to make it more internally consistent in recent years, as each Ansible role was writing to iiab.ini in its own peculiar/personal way.  So it's not never been very consistent, but apparently it's still very useful for these kinds of [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics) troubleshooting purposes:

https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md

Finally, for various IIAB networking purposes, it's possible that iiab.ini is used programmatically in a couple places possible?  (@jvonau would know much more.)

holta, 2021-04-16:
Are these 11 files worth focusing on?

```
roles/0-init/tasks/first_run.yml:1
roles/0-init/tasks/iiab_ini.yml:7
roles/0-init/tasks/main.yml:5

roles/network/tasks/computed_network.yml:2
roles/network/tasks/computed_services.yml:2
roles/network/tasks/detected_network.yml:2
roles/network/tasks/enable_services.yml:6
roles/network/tasks/squid.yml:4
roles/network/tasks/wondershaper.yml:2
roles/network/templates/gateway/iiab-gen-iptables:1

roles/pylibs/templates/iiab_const.py.j2:1
```

(Above excerpted from the 55 files below in `/opt/iiab/iiab/roles`, that work with `/etc/iiab/iiab.ini`)

```
root@u21-srv:/opt/iiab/iiab# grep -rl iiab_ini_file roles | grep -v 0-DEPRECATED-ROLES | xargs grep -c iiab_ini_file | sort

roles/0-init/tasks/first_run.yml:1
roles/0-init/tasks/iiab_ini.yml:7
roles/0-init/tasks/main.yml:5
roles/awstats/tasks/main.yml:2
roles/azuracast/tasks/main.yml:2
roles/bluetooth/tasks/main.yml:2
roles/calibre/tasks/main.yml:2
roles/calibre-web/tasks/main.yml:2
roles/captiveportal/tasks/main.yml:2
roles/cups/tasks/main.yml:2
roles/elgg/tasks/main.yml:2
roles/gitea/tasks/main.yml:2
roles/httpd/tasks/main.yml:2
roles/iiab-admin/tasks/main.yml:2
roles/internetarchive/tasks/main.yml:2
roles/jupyterhub/tasks/main.yml:2
roles/kalite/tasks/main.yml:2
roles/kiwix/tasks/main.yml:2
roles/kolibri/tasks/main.yml:2
roles/lokole/tasks/main.yml:2
roles/mediawiki/tasks/main.yml:2
roles/minetest/tasks/main.yml:2
roles/mongodb/tasks/main.yml:2
roles/monit/tasks/main.yml:2
roles/moodle/tasks/main.yml:2
roles/mosquitto/tasks/main.yml:2
roles/munin/tasks/main.yml:2
roles/mysql/tasks/main.yml:2
roles/network/tasks/computed_network.yml:2
roles/network/tasks/computed_services.yml:2
roles/network/tasks/detected_network.yml:2
roles/network/tasks/enable_services.yml:6
roles/network/tasks/squid.yml:4
roles/network/tasks/wondershaper.yml:2
roles/network/templates/gateway/iiab-gen-iptables:1
roles/nextcloud/tasks/main.yml:2
roles/nginx/tasks/main.yml:2
roles/nodejs/tasks/main.yml:2
roles/nodered/tasks/main.yml:2
roles/openvpn/tasks/main.yml:2
roles/openvpn/templates/announcer.j2:2
roles/osm-vector-maps/tasks/main.yml:2
roles/pbx/tasks/main.yml:2
roles/phpmyadmin/tasks/main.yml:2
roles/postgresql/tasks/main.yml:2
roles/pylibs/templates/iiab_const.py.j2:1
roles/samba/tasks/main.yml:2
roles/sshd/tasks/main.yml:2
roles/sugarizer/tasks/main2.yml:2
roles/transmission/tasks/main.yml:2
roles/usb_lib/tasks/main.yml:2
roles/vnstat/tasks/main.yml:2
roles/wordpress/tasks/main.yml:2
roles/www_base/tasks/html.yml:2
roles/yarn/tasks/main.yml:2
```

holta, 2021-04-16:
Only Tangentially Related to iiab.ini but to surface more of IIAB's internal structures, here's the equivalent summary and horizontal bar chart (histogram) for the 29 files in `/opt/iiab/iiab/roles` that reference `/etc/iiab/iiab.env` :

```
root@u21-srv:/opt/iiab/iiab# grep -rl iiab_env_file roles | grep -v 0-DEPRECATED-ROLES | xargs grep -c iiab_env_file | grep -v 1$ | sort

roles/network/tasks/computed_network.yml:6
roles/network/tasks/enable_services.yml:3
roles/network/tasks/hostapd.yml:2
```

```
root@u21-srv:/opt/iiab/iiab# grep -rl iiab_env_file roles | grep -v 0-DEPRECATED-ROLES | xargs grep -c iiab_env_file | sort

roles/0-init/defaults/main.yml:1
roles/0-init/tasks/main.yml:1
roles/1-prep/tasks/main.yml:1
roles/2-common/tasks/main.yml:1
roles/2-common/templates/iiab-startup.sh:1
roles/3-base-server/tasks/main.yml:1
roles/4-server-options/tasks/main.yml:1
roles/5-xo-services/tasks/main.yml:1
roles/6-generic-apps/tasks/main.yml:1
roles/7-edu-apps/tasks/main.yml:1
roles/8-mgmt-tools/tasks/main.yml:1
roles/9-local-addons/tasks/main.yml:1
roles/network/tasks/computed_network.yml:6
roles/network/tasks/detected_network.yml:1
roles/network/tasks/enable_services.yml:3
roles/network/tasks/hostapd.yml:2
roles/network/templates/gateway/iiab-gen-iptables:1
roles/network/templates/gateway/iiab-internet-off:1
roles/network/templates/gateway/iiab-internet-on:1
roles/network/templates/network/iiab-hotspot-off:1
roles/network/templates/network/iiab-hotspot-on:1
roles/openvpn/templates/announcer.j2:1
roles/pylibs/templates/iiab_const.py.j2:1
roles/pylibs/templates/iiab_env.py.j2:1
roles/usb_lib/tasks/main.yml:1
roles/usb_lib/templates/iiab-usb_lib-show-all-off:1
roles/usb_lib/templates/iiab-usb_lib-show-all-on:1
roles/usb_lib/templates/mount.d/70-usb-library:1
roles/www_base/templates/iiab-refresh-wiki-docs.sh:1
```

jvonau, 2021-04-17:
> is the data in this file ever used? 

Well you should know, your code is a consumer of the data generated. Note cmdserve reads iiab.ini upon access to /admin with GET-IIAB-INI, with a couple of data structures being programmatically acted upon in 20-configure.html via
https://github.com/iiab/iiab-admin-console/blob/master/roles/console/files/js/admin_console.js#L1034-L1043

I find the structure implemented within iiab.ini very helpful to unearth the history of the machine in question with install, runtime, network, and the installed optional roles presented in a chronological order of occurrence. I'm the reason why the layout is way it is, I was the one fielding issues and these data points can be useful when analyzing a complaint.

>is it guaranteed to be correct?

The writing of the data should be correct as of the last running of ansible and ansible does update iiab.ini with the current variables when you pass though the code a second time. This has be maintained since its inception in 2014ish time frame, with a ridiculous amount of time spent with the primary job of first, recording the role was installed, second recording the enabled state of the service. The third part you more or less had the design layout in mind when you thought reading the ( - option: description, value: 'bla' ) for the installed role/service was useful and used in the web facing portal, but that was tossed to the wayside at some point in the past. The data structure (iiab_ini.minetest.minetest_enabled) could of been used just as well as reading local_vars for minetest_enabled although local_vars would need toggling anyway to make the change. If anything it's a routine that was implemented then abandoned but is still useful to those to can make sense of the data presented by it. Guess iiab.ini is safe to remove now. I won't be looking.

jvonau, 2021-04-17:
> Some folk seem to use `/etc/iiab/iiab.ini` visually, to get a quick understanding of what's installed on an IIAB.
> 
> As such I've tried to make it more internally consistent in recent years, as each Ansible role was writing to iiab.ini in its own peculiar/personal way. So it's never been very consistent, but apparently it's still very useful for these kinds of [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics) troubleshooting purposes:
> 
> https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md
> 
> Finally, for various [IIAB networking](https://github.com/iiab/iiab/wiki/IIAB-Networking) purposes, it's possible that iiab.ini is used programmatically in a couple places possibly? (@jvonau would know much more.)

What is your whine about, the roles recording information inconsistently or network recording lots of information? No network just uses local_vars and perhaps iiab_state.

holta, 2021-04-17:
@jvonau mentioned yesterday ( https://github.com/iiab/iiab/issues/2525#issuecomment-821724634 ) that Squid is no longer in use.

Wondershaper, DansGuardian and others probably fall in the same category (i.e. definitely no longer needed 9-to-14 years later).  Possibly things like avahi and named are also no longer needed.

In any case, [/opt/iiab/iiab/roles/network](https://github.com/iiab/iiab/tree/master/roles/network) contains 120 files and 13 directories many of which are vital, but some of which will need a spring cleaning in due course:

```
root@u21-srv:/opt/iiab/iiab# tree roles/network

roles/network
├── defaults
│   └── main.yml
├── README.rst
├── tasks
│   ├── avahi.yml
│   ├── computed_network.yml
│   ├── computed_services.yml
│   ├── dansguardian.yml
│   ├── debian.yml
│   ├── detected_network.yml
│   ├── dhcpd.yml
│   ├── dnsmasq.yml
│   ├── down-debian.yml
│   ├── enable_services.yml
│   ├── fedora
│   │   ├── create_ifcfg.yml.deprecated
│   │   ├── detected_redhat.yml
│   │   ├── edit_ifcfg.yml
│   │   ├── enable_wan.yml
│   │   ├── hosts.yml.deprecated
│   │   ├── ifcfg_mods.yml
│   │   ├── NM.yml
│   │   ├── redetect.yml
│   │   └── static.yml
│   ├── hostapd.yml
│   ├── main.yml
│   ├── named.yml
│   ├── netplan.yml
│   ├── NM-debian.yml
│   ├── restart.yml
│   ├── rpi_debian.yml
│   ├── squid.yml
│   ├── sysd-netd-debian.yml
│   └── wondershaper.yml
└── templates
    ├── avahi
    │   ├── portal.service
    │   └── schoolserver.service
    ├── captive-portal
    │   └── captive-portal.py.j2
    ├── dhcp
    │   ├── dhcpd-env.j2
    │   ├── dhcpd-iiab.conf.j2
    │   └── dhcpd.service
    ├── gateway
    │   ├── iiab-gen-iptables
    │   ├── iiab-internet-off
    │   └── iiab-internet-on
    ├── hostapd
    │   ├── 00-iiab-debug
    │   ├── 50-hostapd
    │   ├── 70-persistent-net.rules
    │   ├── clone-wifi.service.j2
    │   ├── hostapd.conf.j2
    │   ├── hostapd.legacy.j2
    │   ├── hostapd.service.j2
    │   ├── netd-disp
    │   ├── netd-disp2
    │   ├── README.ap0
    │   ├── test-wifi
    │   └── wifi-test.service.j2
    ├── named
    │   ├── bind9.service
    │   ├── dns-jail.conf
    │   ├── dummy
    │   ├── localdomain.zone
    │   ├── localhost.zone
    │   ├── named
    │   ├── named.blackhole
    │   ├── named.broadcast
    │   ├── named-iiab.conf.j2
    │   ├── named.ip6.local
    │   ├── named.j2
    │   ├── named.local
    │   ├── named.rfc1912.zones
    │   ├── named.root
    │   ├── named.root.hints
    │   ├── named.service
    │   ├── named.zero
    │   ├── school.external.zone.db
    │   ├── school.internal.zone.16.in-addr.db.j2
    │   ├── school.internal.zone.32.in-addr.db.j2
    │   ├── school.internal.zone.48.in-addr.db.j2
    │   ├── school.internal.zone.db
    │   ├── school.internal.zone.in-addr.db.j2
    │   └── school.local.zone.db
    ├── network
    │   ├── ap0-manage.conf
    │   ├── br0.j2
    │   ├── bridge-br0
    │   ├── debian-auto.j2
    │   ├── dhcpcd.conf.j2
    │   ├── dnsmasq.conf.j2
    │   ├── dnsmasq-iiab
    │   ├── dnsmasq.service.rh
    │   ├── dnsmasq.service.u18
    │   ├── dnsmasq.sh.j2
    │   ├── hostname.j2
    │   ├── hosts-dnsmasq.j2
    │   ├── ifcfg.j2
    │   ├── ifcfg-slave.j2
    │   ├── ifcfg-WAN.j2
    │   ├── iiab-hotspot-off
    │   ├── iiab-hotspot-on
    │   ├── manage.conf.j2
    │   ├── netplan.j2
    │   ├── NetworkManager.conf
    │   ├── NM-static.j2
    │   ├── resolv.conf.j2
    │   ├── resolvconf.j2
    │   ├── rpi.j2
    │   ├── sysconfig.network.j2
    │   ├── systemd-br0.j2
    │   ├── systemd-br0-network.j2
    │   ├── systemd-br0-slave.j2
    │   ├── systemd.j2
    │   ├── systemd-static-net.j2
    │   └── wifi-slave.j2
    ├── squid
    │   ├── allowregex.rules
    │   ├── dansguardian.conf.centos.j2
    │   ├── dansguardian.conf.debian.j2
    │   ├── dansguardian.conf.j2
    │   ├── denyregex.rules
    │   ├── dstaddress.rules
    │   ├── iiab-httpcache.j2
    │   ├── sites.whitelist.txt
    │   ├── squid-iiab.conf.j2
    │   └── squid.sysconfig
    └── wondershaper
        ├── wondershaper.conf
        ├── wondershaper.j2
        └── wondershaper.service
```

tim-moody, 2021-04-17:
iiab.ini was designed by Santiago many years ago, including the English text. Adm Cons reads it to display it, but doesn't use the data. Seems reasonable to continue that.

closed 2021-04-17 by tim-moody