iiab/iiab #2738: RFC: how to determine enabled status

tim-moody, 2021-04-17:
I am assuming that installed status can be had from state.yml.

On the last call it was said that enabled should be taken from systemctl is-enabled. This is a reasonable idea if performance does not suffer too much.

I have created a data structure to determine how to gather this status. I think there are 4 sources of enabled status:

* service enabled
* nginx conf exists
* apache conf exists (if there still are any)
* installed implies enabled

I have not included dependent services like nginx or mysql. I also assume that any dependencies that are not part of core will be installed and enabled when one of these roles is run.
```
iiab_services={
     "awstats": {"menu_def": None, "stat_src": "nginx", "stat_arg": "awstats-nginx.conf"},\
     "azuracast": {"menu_def": "en", "stat_src": "install", "stat_arg": None},\
     # check
     "calibre": {"menu_def": "en", "stat_src": "service", "stat_arg": "calibre-serve.service"},\
     "calibre-web": {"menu_def": "en-calibre", "stat_src": "service", "stat_arg": "calibre-serve.service"},\
     "captiveportal": {"menu_def": None, "stat_src": "nginx", "stat_arg": "capture.conf"},\
     "cups": {"menu_def": "en-cups", "stat_src": "service", "stat_arg": "cups.service"},\
     "elgg": {"menu_def": "en-elgg", "stat_src": "nginx", "stat_arg": "elgg-nginx.conf"},\
     "gitea": {"menu_def": "en-gitea", "stat_src": "service", "stat_arg": "gitea.service"},\
     "internetarchive": {"menu_def": "en-internetarchive", "stat_src": "service", "stat_arg": "internetarchive.service"},\
     "jupyterhub": {"menu_def": "en-jupyter", "stat_src": "service", "stat_arg": "jupyterhub.service"},\
     "kalite": {"menu_def": "en-kalite", "stat_src": "service", "stat_arg": "kalite-serve.service"},\
     "kiwix": {"menu_def": "en-test_zim", "stat_src": "service", "stat_arg": "kiwix-serve.service"},\
     "kolibri": {"menu_def": "en-kolibri", "stat_src": "service", "stat_arg": "kolibri.service"},\
     "lokole": {"menu_def": "en-lokole", "stat_src": "service", "stat_arg": "lokole.service"},\
     "mediawiki": {"menu_def": "en-calibre", "stat_src": "nginx", "stat_arg": "mediawiki-nginx.conf"},\
     "minetest": {"menu_def": "en-minetest", "stat_src": "service", "stat_arg": "minetest-server.service"},\
     "monit": {"menu_def": None, "stat_src": "service", "stat_arg": "monit.service"},\
     "moodle": {"menu_def": "en-moodle", "stat_src": "nginx", "stat_arg": "moodle-nginx.conf"},\
     "mosquitto": {"menu_def": Nonw, "stat_src": "service", "stat_arg": "mosquitto.service"},\
     "munin": {"menu_def": None, "stat_src": "nginx", "stat_arg": "munin24-nginx.conf"},\
     "nextcloud": {"menu_def": "en-nextcloud", "stat_src": "nginx", "stat_arg": "nextcloud-nginx.conf"},\
     "nodered": {"menu_def": "en-nodered", "stat_src": "service", "stat_arg": "nodered.service"},\
     "openvpn": {"menu_def": None, "stat_src": "service", "stat_arg": "openvpn@xscenet.service"},\
     "osm-vector-maps": {"menu_def": "en-calibre", "stat_src": "nginx", "stat_arg": "osm-vector-maps-nginx.conf"},\
     "pbx": {"menu_def": None, "stat_src": "service", "stat_arg": "freepbx.service"},\
     "phpmyadmin": {"menu_def": None, "stat_src": "apache", "stat_arg": "phpmyadmin.conf"},\
     "samba": {"menu_def": None, "stat_src": "service", "stat_arg": "{{ smb_service }}"},\
     "sugarizer": {"menu_def": "en-sugarizer", "stat_src": "service", "stat_arg": "sugarizer.service"},\
     "transmission": {"menu_def": None, "stat_src": "service", "stat_arg": "transmission-daemon.service"},\
     "usb_lib": {"menu_def": "en-usb", "stat_src": "service", "stat_arg": usbmount@.service},\
     "vnstat": {"menu_def": None, "stat_src": "service", "stat_arg": "vnstat.service"},\
     "wordpress": {"menu_def": "en-wordpress", "stat_src": "nginx", "stat_arg": "wordpress-nginx.conf"}\
}

```

holta, 2021-04-17:
> - apache conf exists (if there still are any)

The Apache -> NGINX transition status doc is here, to (hopefully!) help answer questions like this:

https://github.com/iiab/iiab/tree/master/roles/nginx

holta, 2021-04-17:
This looks like a very promising strategy.

If I'm remembering correctly, @jvonau mentioned this kind of approach about a year ago.

tim-moody, 2021-04-17:
>If I'm remembering correctly, @jvonau mentioned this kind of approach about a year ago.

so you said on the call. I'm not taking credit for the idea; just trying to implement it.

by apache I mean apache only.

holta, 2021-04-17:
> I am assuming that installed status can be had from state.yml

Yes.  To be specific, for people new to IIAB, the test is to look inside `/etc/iiab/iiab_state.yml` for existence (or non-existence!) of lines like...

```
calibreweb_installed: True
kolibri_installed: True
moodle_installed: True
wordpress_installed: True
...
```

tim-moody, 2021-04-17:
get_roles_status() added to adm_lib
performance is fine

jvonau, 2021-04-17:
> I am assuming that installed status can be had from state.yml.
> 
> On the last call it was said that enabled should be taken from systemctl is-enabled. This is a reasonable idea if performance does not suffer too much.

I'm assuming this is more geared towards admin-console, as "menu_def" is used and during the install the services are not yet installed.
> 
> I have created a data structure to determine how to gather this status. I think there are 4 sources of enabled status:
> 
>     * service enabled
>
>     * nginx conf exists
>
>     * apache conf exists (if there still are any)
> 
>     * installed implies enabled
> 
>

Lets clarify '* installed' first, is that meaning local_vars' *_install  or *_installed from iiab_state?

local_vars' `_enabled: True` should imply `_install: True` to keep current validation happy while going the other way, `_install: True` with `_enabled: False` is also valid, just means install the app/service but don't make it available. 

I find '*_install: True' with '_enabled: False' in local_var is very useful with image creation, for example one could install based on what is in 'medium' local_vars but enable based on 'min" (or any combination of _install _enabled for the optional roles) because the end-user's hardware's capabilities is an unknown variable and I would prefer one 'official iiab image' per distro. The images I produced for the past 2 years with this method would correctly display the disabled services with admin-console's current code base, what is missing is being able in install roles that were not previously installed. Installing these 'new' roles is a nice to have in the future for the end-user but for the most part this is more of a what should be on an image by default issue, installing more roles but have them disabled can be done *TODAY* with no changes what so ever. Initial images become bigger but is offset by not using the internet once the image is written and booted. 

> I have not included dependent services like nginx or mysql. I also assume that any dependencies that are not part of core will be installed and enabled when one of these roles is run.
>
> ``` [...]

The above looks like you intend to shift away from using iiab-from-console.yml to implement individual changes, as the stat_src and stat_arg info is already part of each role.

tim-moody, 2021-04-17:
>I'm assuming this is more geared towards admin-console

yes. for now it is iinstalled with adm cons, though it could be moved to iiab_lib if there's a reason to do so.

>the stat_src and stat_arg info is already part of each role.

where is that?

>the above looks like you intend to shift away from using iiab-from-console.yml to implement individual changes

looking at it, but this is just about getting actual status as opposed to desired status.

jvonau, 2021-04-17:
> > I'm assuming this is more geared towards admin-console
> 
> yes. for now it is iinstalled with adm cons, though it could be moved to iiab_lib if there's a reason to do so.
> 
> > the stat_src and stat_arg info is already part of each role.
> 
> where is that?
> 

By what the role does, each role should install/delete the role's specific nginx.conf file and record the current state in iiab.ini. Roles that have a service backend should have the additional commands that starts|stops enables|disables the service present to do so.

> > the above looks like you intend to shift away from using iiab-from-console.yml to implement individual changes
> 
> looking at it, but this is just about getting actual status as opposed to desired status.

If you gathering data 'is running', ie pid file present would be useful.

tim-moody, 2021-04-17:
I added systemctl is-active

jvonau, 2021-04-17:
> By what the role does, each role should install/delete the role's specific nginx.conf file and record the current state in iiab.ini. Roles that have a service backend should have the additional [...]

One of the down sides of the initial implementation of nginx support is the template is written directly to /etc/nginx/conf.d/ and the only copy is deleted when disabled. What we should do is update a copy in /etc/nginx/sites-available/ and use symlinks in /etc/nginx/sites-enabled/ as per normal nginx usage.

jvonau, 2021-04-18:
> What we should do is update a copy in /etc/nginx/sites-available/ and use symlinks in /etc/nginx/sites-enabled/ as per normal nginx usage. 

The /etc/nginx/sites-* directories might be an ubuntu add-on, don't have a debian|RaspOS booted to confirm.

holta, 2021-04-18:
> > What we should do is update a copy in /etc/nginx/sites-available/ and use symlinks in /etc/nginx/sites-enabled/ as per normal nginx usage.
> 
> The /etc/nginx/sites-* directories might be an ubuntu add-on, don't have a debian|RaspOS booted to confirm.

They definitely exist on RaspiOS, Debian 11 as well as Ubuntu.  (I'm not worried about Debian 10, which we won't be supporting for much longer, but someone can check that if nec.)

holta, 2021-04-18:
FYI /etc/nginx/sites-available/ and /etc/nginx/sites-enabled/ also exist on Debian 10 (if apt package nginx-common is installed).

jvonau, 2021-04-18:
The design choice was made at the time, held my nose and looked the other way. The alternate method was discussed but since ansible was always to be used there is no effective difference from where the file is referenced, the role-nginx.conf would be replaced if needed, and there was still a bunch more work to do with little time.

tim-moody, 2021-04-18:
@jvonau are you planning to make it possible to invoke role/tasks/enable_disable.yml independently of the rest of the role?

tim-moody, 2021-04-18:
what is the test for calibre (vs calibre-web)?

jvonau, 2021-04-18:
> @jvonau are you planning to make it possible to invoke role/tasks/enable_disable.yml independently of the rest of the role?

No, the use of enable-or-disable is not consistent across the roles and the iiab.ini writing is usually in main.yml 
>ubuntu@box:/opt/iiab/iiab$ ls roles/*/tasks/*enable*
roles/azuracast/tasks/enable-or-disable.yml
roles/bluetooth/tasks/enable.yml
roles/captiveportal/tasks/enable-or-disable.yml
roles/cups/tasks/enable-or-disable.yml
roles/kiwix/tasks/enable-or-disable.yml
roles/minetest/tasks/enable.yml
roles/mongodb/tasks/enable-or-disable.yml
roles/monit/tasks/enable-or-disable.yml
roles/mosquitto/tasks/enable-or-disable.yml
roles/network/tasks/enable_services.yml
roles/nginx/tasks/enable-or-disable.yml
roles/openvpn/tasks/enable-or-disable.yml
roles/pbx/tasks/freepbx_enable.yml
roles/phpmyadmin/tasks/enable-or-disable.yml
roles/samba/tasks/enable-or-disable.yml
roles/sshd/tasks/enable-or-disable.yml
roles/sugarizer/tasks/enable-or-disable.yml
roles/transmission/tasks/enable-or-disable.yml
roles/vnstat/tasks/enable-or-disable.yml

>drwxr-xr-x  5 root root 4096 Jan  6 11:26 awstats
drwxr-xr-x  5 root root 4096 Jan  6 11:26 azuracast
drwxr-xr-x  5 root root 4096 Jan  6 11:26 bluetooth
drwxr-xr-x  5 root root 4096 Jan  6 11:26 calibre
drwxr-xr-x  6 root root 4096 Jan  6 11:26 calibre-web
drwxr-xr-x  6 root root 4096 Jan  6 11:26 captiveportal
drwxr-xr-x  5 root root 4096 Jan  6 11:26 cups
drwxr-xr-x  5 root root 4096 Jan  6 11:26 elgg
drwxr-xr-x  4 root root 4096 Jan  6 11:26 firmware
drwxr-xr-x  5 root root 4096 Jan  6 11:26 gitea
drwxr-xr-x  5 root root 4096 Jan  6 11:26 httpd
drwxr-xr-x  5 root root 4096 Jan  6 11:26 iiab-admin
drwxr-xr-x  5 root root 4096 Jan  6 11:26 internetarchive
drwxr-xr-x  5 root root 4096 Mar 28 08:08 kalite
drwxr-xr-x  6 root root 4096 Jan  6 11:26 kiwix
drwxr-xr-x  5 root root 4096 Jan  6 11:26 kolibri
drwxr-xr-x  5 root root 4096 Jan  6 11:26 lokole
drwxr-xr-x  5 root root 4096 Jan  6 11:26 mediawiki
drwxr-xr-x  5 root root 4096 Jan  6 11:26 minetest
drwxr-xr-x  5 root root 4096 Jan  6 11:26 mongodb
drwxr-xr-x  4 root root 4096 Jan  6 11:26 monit
drwxr-xr-x  5 root root 4096 Jan  6 11:26 moodle
drwxr-xr-x  5 root root 4096 Jan  6 11:26 mosquitto
drwxr-xr-x  5 root root 4096 Jan  6 11:26 munin
drwxr-xr-x  4 root root 4096 Jan  6 11:26 mysql
drwxr-xr-x  5 root root 4096 Jan  6 11:26 network
drwxr-xr-x  5 root root 4096 Apr 11 23:56 nextcloud
drwxr-xr-x  5 root root 4096 Jan  6 11:26 nginx
drwxr-xr-x  4 root root 4096 Jan  6 11:26 nodejs
drwxr-xr-x  5 root root 4096 Jan  6 11:26 nodered
drwxr-xr-x  5 root root 4096 Jan  6 11:26 openvpn
drwxr-xr-x  6 root root 4096 Jan  6 11:26 osm-vector-maps
drwxr-xr-x  5 root root 4096 Jan  6 11:26 pbx
drwxr-xr-x  5 root root 4096 Jan  6 11:26 phpmyadmin
drwxr-xr-x  5 root root 4096 Jan  6 11:26 postgresql
drwxr-xr-x  4 root root 4096 Jan  6 11:26 pylibs
drwxr-xr-x  6 root root 4096 Jan  6 11:26 samba
drwxr-xr-x  4 root root 4096 Jan  6 11:26 sshd
drwxr-xr-x  5 root root 4096 Jan  6 11:26 sugarizer
drwxr-xr-x  5 root root 4096 Mar 28 08:08 transmission
drwxr-xr-x  5 root root 4096 Mar 28 08:08 usb_lib
drwxr-xr-x  4 root root 4096 Jan  6 11:26 vnstat
drwxr-xr-x  6 root root 4096 Jan  6 11:26 wordpress
drwxr-xr-x  5 root root 4096 Jan  6 11:26 www_base
drwxr-xr-x  4 root root 4096 Jan  6 11:26 www_options
drwxr-xr-x  3 root root 4096 Jan  6 11:26 yarn

jvonau, 2021-04-18:
> what is the test for calibre (vs calibre-web)?

Sorry I don't understand the question, but they are separate roles if that helps.

tim-moody, 2021-04-18:
thought they had the same service, but it's calibre-serve and calibre-web

jvonau, 2021-04-18:
calibre-web's sevice name is calibre-web.service while calibre uses calibre-serve.service, different roles, different programs, different service names, perhaps @holta could add something.

tim-moody, 2021-04-18:
on rpi 4 returned in approx 2 seconds:

`adm.get_roles_status()`

`{'awstats': {'installed': True, 'enabled': True, 'active': True}, 'azuracast': {'installed': False, 'enabled': False, 'active': False}, 'calibre': {'installed': False, 'enabled': False, 'active': False}, 'calibre-web': {'installed': False, 'enabled': False, 'active': False}, 'captiveportal': {'installed': True, 'enabled': False, 'active': False}, 'cups': {'installed': False, 'enabled': False, 'active': False}, 'elgg': {'installed': False, 'enabled': False, 'active': False}, 'gitea': {'installed': False, 'enabled': False, 'active': False}, 'internetarchive': {'installed': False, 'enabled': False, 'active': False}, 'jupyterhub': {'installed': False, 'enabled': False, 'active': False}, 'kalite': {'installed': False, 'enabled': False, 'active': False}, 'kiwix': {'installed': True, 'enabled': True, 'active': True}, 'kolibri': {'installed': False, 'enabled': False, 'active': False}, 'lokole': {'installed': False, 'enabled': False, 'active': False}, 'mediawiki': {'installed': True, 'enabled': True, 'active': True}, 'minetest': {'installed': False, 'enabled': False, 'active': False}, 'monit': {'installed': False, 'enabled': False, 'active': False}, 'moodle': {'installed': False, 'enabled': False, 'active': False}, 'mosquitto': {'installed': False, 'enabled': False, 'active': False}, 'munin': {'installed': True, 'enabled': True, 'active': True}, 'nextcloud': {'installed': False, 'enabled': False, 'active': False}, 'nodered': {'installed': False, 'enabled': False, 'active': False}, 'openvpn': {'installed': True, 'enabled': True, 'active': False}, 'osm-vector-maps': {'installed': False, 'enabled': False, 'active': False}, 'pbx': {'installed': False, 'enabled': False, 'active': False}, 'phpmyadmin': {'installed': False, 'enabled': False, 'active': False}, 'samba': {'installed': False, 'enabled': False, 'active': False}, 'sugarizer': {'installed': False, 'enabled': False, 'active': False}, 'transmission': {'installed': False, 'enabled': False, 'active': False}, 'usb_lib': {'installed': True, 'enabled': True, 'active': False}, 'vnstat': {'installed': True, 'enabled': True, 'active': True}, 'wordpress': {'installed': False, 'enabled': False, 'active': False}}
`

jvonau, 2021-04-18:
Looks good, time is not that bad, think the biggest hit time wise would be the is_active if you find the query taking too long for your liking. For the future could get fancy and display a big warning when is_enabled and is_active don't match up, just a thought and not required.

holta, 2021-04-18:
> could get fancy and display a big warning when is_enabled and is_active don't match up

If a system sanity-check is later needed, to verify services are working as expected etc.

In the distant future one could imagine visual warnings on the main IIAB page (http://box.lan) to warn folks/operators/everyone that critical LMS is not working...or disk is nearly full...etc (along the same lines as RaspiOS visually warning users on-screen when the CPU/motherboard are overheating).

closed 2023-03-20 by tim-moody