iiab/iiab #2740: Public Relations

jvonau, 2021-04-18:
Not sure if you are aware or not, one of the better sources of exposure https://opensource.com/article/17/5/internet-in-a-box-raspberry-pi has links under "Finding and using content" for a live demo that point to schoolserver.xsce.org resolving to 173.246.159.89 for me. Now whois tells me "Updated Date: 2021-03-23T15:45:23Z" and nmap is telling 'filtered' so something is amiss on the host or is schoolserver.xsce.org going away?

holta, 2021-04-18:
Important observation, thanks.

Even if worst case it turns out little can be done about this particular situation?!

tim-moody, 2021-04-18:
This is a little bizarre, and thanks for catching it. I didn't realize xsce.org was still used anywhere and was going to let the registration expire. BUT, BlueHost renewed it automatically, so I guess I should just point it to iiab.me?

jvonau, 2021-04-18:
If the same content and ports are available with the same path as advertised on the webpage, should work.

tim-moody, 2021-04-18:
Looking at the links, I see they are to very old content, which is not on iiab.me. I put in a redirect, but it doesn't seem to work. I also don't know where the 173 is coming from as it is not in the domain dns.

jvonau, 2021-04-18:
With the content from the 2016-2017 era might be useful to spin up a machine|instance just for that purpose.

jvonau, 2021-04-18:
> Looking at the links, I see they are to very old content, which is not on iiab.me. I put in a redirect, but it doesn't seem to work. I also don't know where the 173 is coming from as it is not in the domain dns.

messages crossed, what doe 173 refer to?

tim-moody, 2021-04-18:
173.246.159.89 tried adding cname of schoolserver.xsce.org to iiab.me, but it said there was an 'issue' and failed. tried adding an A record but means hardcoding the ip. I guess I need a dynamic dns record. have to find my account.

jvonau, 2021-04-18:
I see what 173 refers to:
>ubuntu@box:/opt/iiab/iiab$ dig schoolserver.xsce.org pointer +noall +answer
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 48507
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1
;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 65494
;; QUESTION SECTION:
;schoolserver.xsce.org.		IN	A
;; ANSWER SECTION:
schoolserver.xsce.org.	14400	IN	A	173.246.159.89
;; Query time: 55 msec
;; SERVER: 127.0.0.53#53(127.0.0.53)
;; WHEN: Sun Apr 18 08:20:55 CDT 2021
;; MSG SIZE  rcvd: 66
 
> ubuntu@box:/opt/iiab/iiab$ whois 173.246.159.89 
NetRange:       173.246.128.0 - 173.246.159.255
CIDR:           173.246.128.0/19
NetName:        TEKSAVVY
NetHandle:      NET-173-246-128-0-1
Parent:         NET173 (NET-173-0-0-0-0)
NetType:        Direct Allocation
OriginAS:       AS5645, AS20375
Organization:   TekSavvy Solutions, Inc. (TEKSAV)

Hope it helps..

tim-moody, 2021-04-18:
Yes, I know what it is. It's my old teksavvy ip address, no longer mine since I have had Bell for 2 years. Problem is how to fix it.

jvonau, 2021-04-18:
Ok xsce.org at 162.241.24.98 is a dead site but responds, I would just re-purpose 162.241.24.98 as needed for the webpage and point schoolserver.xsce.org to xsce.org

jvonau, 2021-04-18:
First get schoolserver.xsce.org to respond the same way as xsce.org does with the dead site, that might be*(it's been awhile) just a cname for schoolserver pointing to xsce.org should you want to reuse 162.241.24.98.

tim-moody, 2021-04-18:
162.241.24.98 is just a parking site. I have no control over it. I'm moving registrars and trying redirect again

tim-moody, 2021-04-18:
iiab.me is probably going to choke on all those port 3000 urls that don't have a valid zim

jvonau, 2021-04-18:
You still have the old zims? Can you load the zims but not add the menu-defs?

tim-moody, 2021-04-18:
Actually there are plenty of other problems. The code points to xsce. The wiki points to kiwix. Probably more.

tim-moody, 2021-04-18:
So I changed registrar and name servers and schoolserver.xsce.org still points to 173... How is that even possible?

jvonau, 2021-04-18:
> ubuntu@box:/opt/iiab/iiab$ dig xsce.org any +noall +answer
xsce.org.		76121	IN	SOA	ns1.bluehost.com. root.box5862.bluehost.com. 2019062705 86400 7200 3600000 300
xsce.org.		3290	IN	A	162.241.24.98

2019062705 is when my upstream dns thinks the domain was lasted updated, it's going to take awhile for the change to propagate though the internet's dns system. If this was a new domain the change would be instantaneous as there would be no cached records to update in the wild, the client lookup would be forced to contact the SOA for the info.

jvonau, 2021-04-18:
https://www.bluehost.com/help/article/dns-management-add-edit-or-delete-dns-entries
>Pro Tip! Your new DNS settings will begin to propagate as soon as you save your changes. Each ISP updates at a different rate, usually taking 24 - 72 hours to complete worldwide. During this time, your emails will be down, and your web traffic will be directed to either the old or new host, depending on the visitor's ISP.

I'll post when my ISP updates their cache.

tim-moody, 2021-04-18:
still doesn't explain why xsce.org goes to iiab.me and schoolserver.xsce.org doesn't. both are redirected in the same way, but sure let's see if it resolves

jvonau, 2021-04-19:
Think xsce.org is alive at its new home.
>Non-authoritative answer:
Name:	xsce.org
Address: 192.64.119.238

jvonau, 2021-04-19:
Don't create the schoolserver record until after the machine is configured, so the record is not cached encase you need to revise the ip address before going live.

tim-moody, 2021-04-19:
I don't want to create a separate machine for xsce.org. It is pointed at iiab.me and goes there because I created a CNAME record that basically redirects. Try http://xsce.org.

I would like schoolserver.iiab.me to do the same and created a CNAME for that as well, but it doesn't get found.

Ah maybe the problem is iiab.me has no record for schoolserver.iiab.me

jvonau, 2021-04-19:
> I don't want to create a separate machine for xsce.org. It is pointed at iiab.me and goes there because I created a CNAME record that basically redirects. Try http://xsce.org.

That works to display the menu but shows iiab.me in the address bar so that won't work with the links on the page. The address bar must retain (schoolserver).xsce.org for the links to work seamlessly.

> 
> I would like schoolserver.iiab.me to do the same and created a CNAME for that as well, but it doesn't get found.
> 

That doesn't look right, you mean schoolserver.xsce.org?

> Ah maybe the problem is iiab.me has no record for schoolserver.iiab.me

or nginx's server_name directive on iiab.me needs to know it is now serving http for xsce.org?

jvonau, 2021-04-20:
The rewrite in the address bar is gone, but schoolserver does not resolve yet.

holta, 2021-06-18:
We tried.  Good enough for now.

closed 2021-06-18 by holta