iiab/iiab #2744: Testing Ubuntu 21.04 on (RPi+PC) x (Server+Desktop). Does WiFi firmware 7.45.18.0 work on RPi?

holta, 2021-04-19:
Looks great so far, but starting on 2021-04-22 testing of the official/released OS (Ubuntu "Hirsute Hippo" 21.04) will be important.

Refs: #823, #2585, #2610

jvonau, 2021-04-20:
Testing against the [daily server images](https://cdimage.ubuntu.com/ubuntu-server/daily-preinstalled/current/) seems to work ok with my wifi-only testing, no wired connection internet via wlan0.

holta, 2021-04-23:
FYI Ubuntu 21.04 is now released:
https://releases.ubuntu.com/21.04/

Also for Raspberry Pi:
https://ubuntu.com/download/raspberry-pi

_Huge thanks to the many volunteers who are able to get to know this better[*] and thereby help with IIAB testing:_
http://download.iiab.io

[*] While RaspiOS 11 will officially be a derivative of Debian 11 when both are released in a couple months or so...certain Ubuntu advances are also likely to make their way into RaspiOS ?!

holta, 2021-04-28:
Just FYI testing is ongoing (on Ubuntu Server 21.04 and Ubuntu Desktop 21.04 VM's, and increasingly on RPi too) thanks to many people, but especially @jvonau and @deldesir:

- http://10.8.0.42 = 196-u21-srv-VM-BIG
- http://10.8.0.46 = 197-u21-desk-VM-BIG

Refs:

- #2756 "Lokole install blows up"
- PR #2757 "lokole/tasks/main.yml: fix illegal char EM DASH \u2014 in iiab.ini description"

holta, 2021-08-11:
Ubuntu 21.04 testing seems good enough.

It has been tested across many different scenarios, even if it hasn't been tested to death.

Which is OK as others OS's like the imminent Raspberry Pi 11 "Bullseye" (based on Debian 11 "Bullseye") are likely much more important in days as Saturday Aug 14 approaches!)  Also Ubuntu 21.10 "Impish Indri" will be released exactly 2 months after that, on Oct 14:

- PR #2749
- PR #2820
- PR #2818