iiab/iiab #2749: Testing IIAB on the new Debian 11 Bullseye Release Candidate(s)

holta, 2021-04-23:
Debian 11 Release Candidate 1 was released today!

https://www.debian.org/devel/debian-installer/

Two years later, this Debian 11 ("Bullseye") should be the basis for the new Raspberry Pi OS 11 (formerly known as Raspbian) in coming weeks/months...

We're likely already in extremely good shape here, thanks to hard work from @deldesir and @jvonau starting back in October 2020.

But of course final testing in coming weeks will be critical as this converges~

Refs: #2744 #2748

jvonau, 2021-04-23:
Known issue on desktop with 2 or more internet devices #2658 
Proposed fix #2693

holta, 2021-04-23:
Just FYI "Errata for Bullseye RC 1"
https://www.debian.org/devel/debian-installer/errata

holta, 2021-05-21:
FWIW a colorful graph of Debian "release-critical" bugs over the past 4 years:

https://bugs.debian.org/release-critical/

And over the past 30 days:

https://bugs.debian.org/release-critical/graph-month.png

(Neither makes clear when Debian 11 "Bullseye" will be released — but they will give some clues in coming weeks!)

holta, 2021-06-14:
Debian 11 "Bullseye" is converging more slowly than prior versions.  It will enter "Full Freeze" on Saturday 2021-07-17.  Possibly this means an August release?

https://lists.debian.org/debian-devel-announce/2021/06/msg00000.html

I assume Raspberry Pi OS 11 (announced for Q2 2021) will likely also be somewhat delayed as a result.

holta, 2021-06-14:
Debian 11 Release Candidate 2 is now official:

https://lists.debian.org/debian-devel-announce/2021/06/msg00001.html

kmw, 2021-06-19:
I tried installing iiab on https://raspi.debian.net/verified/20210413_raspi_4_bullseye.img.xz and it didn't work. Has anyone else done anything with this image? I see the Debian 11 Release Candidate 2 is now official but where can I download it.
I'm headed to Haiti late tomorrow and I'm scrambling to get a French based iiab on 64g and 128g cards. Maybe I should stick to debian 10 for now.

holta, 2021-06-19:
@kmw stick with 32-bit Raspberry Pi OS (based on Debian 10 Buster) if you can?

That is by far the most heavily tested!

Also: see http://iiab.me/haiti for example content you might want to download?

Keep in Touch!

kmw, 2021-06-19:
how do I install the http://iiab.me/haiti ? Do I need to locate and select
each package from a list or is there a script or way to grab all of them at
once?

holta, 2021-06-19:
> how do I install the http://iiab.me/haiti ? Do I need to locate and select each package from a list or is there a script or way to grab all of them at once?

They will not all fit within 64G or 128G.

But one subset is available (approx 100G) if you use Admin Console -> Install Content -> Quick Pick -> School with Instruction in French.

PS Debian 11 Bullseye us tentatively scheduled for release on 2021-07-31: https://release.debian.org

kmw, 2021-06-19:
I also have a 256g chip but I might not have enough time to download that
much between now and when I leave.

holta, 2021-06-19:
> But one subset is available (approx 100G) if you use Admin Console -> Install Content -> Quick Pick -> School with Instruction in French.

Also beware the above is very new!  Picking your own Content Packs (within Admin Console -> Install Content) is the safest way especially if you're targeting an exact total size in GB (e.g. 50 of a 64GB microSD card?  Or 100 of a 128GB microSD card?)

kmw, 2021-06-19:
I do not see Quick Pick.

holta, 2021-06-19:
> I do not see Quick Pick.

Possibly you should install a fresh copy of Internet-in-a-Box on a fresh OS:

```
curl d.iiab.io/install.txt | sudo bash
```

(This and other changes were mainlined about 4 days ago.)

kmw, 2021-06-19:
ok I'll do debian 10 https://raspi.debian.net/daily/raspi_4_buster.img.xz

holta, 2021-06-19:
> ok I'll do debian 10 https://raspi.debian.net/daily/raspi_4_buster.img.xz

I don't know anything about the above image.

FYI these are the 3 OS images (32-bit Raspberry Pi OS, based on Debian 10) that are regularly tested:

https://www.raspberrypi.org/software/operating-systems/

holta, 2021-07-09:
@kmw @deldesir @jvonau and others: while IIAB does not support Debian 11 on Raspberry Pi hardware as of today (it has glitches apparently?) here are links for those that might want to get involved in making this work:

https://wiki.debian.org/RaspberryPi4

Whether or not Raspberry Pi OS 11 is released in the coming days, possibly solving these issues in their own way?!

holta, 2021-07-09:
@kmw you might also want express your opinion here:

- #2853

Related:

- #2820

holta, 2021-07-20:
Debian 11 Bullseye "fully frozen" on 2021-07-17:
https://lists.debian.org/debian-devel-announce/2021/07/msg00002.html

deltacloud9, 2021-07-20:
Unable to access the box web page when testing in Debian 11 VM.

http://sprunge.us/XiDxUz?en

Receiving this error message:
[image]

holta, 2021-07-20:
> Unable to access the box web page when testing in Debian 11 VM.
> 
> http://sprunge.us/XiDxUz?en

Did you test http://192.168.122.173 from another PC on your network?

Did you test IIAB's standard URL's from within the VM?  e.g. did you run `lynx box.lan` at the command-line (using a text browser) if the VM has not graphical desktop?

- http://box
- http://box.lan
- http://127.0.0.1
- http://172.18.96.1
- http://192.168.122.173

If you installed a graphical desktop with the Debian 11 VM, which one did you install?  (e.g. Gnome or otherwise?)

deltacloud9, 2021-07-20:
> > Unable to access the box web page when testing in Debian 11 VM.
> > http://sprunge.us/XiDxUz?en
> 
> Did you test http://192.168.122.173 (This does not work from another PC) from another PC on your network?
> 
> Did you test IIAB's standard URL's from within the VM? e.g. did you run `lynx box.lan` at the command-line (using a text browser) if the VM has no graphical desktop?
> 
> * http://box - This works from within the VM
> * http://box.lan - This does not work from within the VM
> * http://127.0.0.1 - This one works from within the VM
> * http://172.18.96.1 - This one spins endlessly from within the VM
> * http://192.168.122.173 - This one works from within the VM
> 
> If you installed a graphical desktop with the Debian 11 VM, which one did you install? (e.g. Gnome or otherwise?)
I installed the KDE desktop for this VM.

@holta Please see my responses above

holta, 2021-07-20:
> * http://box - This works from within the VM

> * http://127.0.0.1 - This one works from within the VM

> * http://192.168.122.173 - This one works from within the VM

The above 3 work, great!

> * http://172.18.96.1 - This one spins endlessly from within the VM

That's likely normal, considering your VM has not created a hotspot.

> * http://box.lan - This does not work from within the VM

That one's more concerning!

@jvonau do you have any recommendations that @deltacloud9 should try out, to adjust networking?

jvonau, 2021-07-20:
> > Did you test http://192.168.122.173 (This does not work from another PC) from another PC on your network?

That is unexpected, if this is VM I suspect only the host of the VM can access 192.168.122.173 directly.

> > 
> > * http://box - This works from within the VM

Good hostname set 
> > * http://box.lan - This does not work from within the VM

Ok, '.lan' is only really applicable to wifi clients. Need dnsmasq to be running on br0 for that to resolve.
> > * http://127.0.0.1 - This one works from within the VM

This should always work
> > * http://172.18.96.1 - This one spins endlessly from within the VM

Should not really be present with a single interface appliance (no br0) and is not.
> > * http://192.168.122.173 - This one works from within the VM
> > 

Should work from a client within the same subnet ie 192.168.122.xxx
> > If you installed a graphical desktop with the Debian 11 VM, which one did you install? (e.g. Gnome or otherwise?)
> > I installed the KDE desktop for this VM.

holta, 2021-07-20:
> > http://box.lan - This does not work from within the VM

> Ok, '.lan' is only really applicable to wifi clients. Need dnsmasq to be running on br0 for that to resolve.

Ah well, ok.

Possibly this is something we can improve in future, as much of our documentation/conversations assumes that `box.lan` should almost always work.

>> Did you test http://192.168.122.173 from another PC on your network?

> (This does not work from another PC)

@deltacloud9 can you ssh into 192.168.122.173 from another PC on your network?

Do you need to adjust your VM configuration to make 192.168.122.173's ports (especially port 80) available to other PC's on your network?

(If using VirtualBox, one sometimes needs to change from NAT to `Bridged Adapter` within [actual VM] > Network > Adapter 2 > Enable Network Adapter > Attached to > Bridges Adapter.)

deltacloud9, 2021-07-20:
> > > http://box.lan - This does not work from within the VM
> 
> > Ok, '.lan' is only really applicable to wifi clients. Need dnsmasq to be running on br0 for that to resolve.
> 
> Ah well, ok.
> 
> Possibly this is something we can improve in future, as much of our documentation/conversations assumes that `box.lan` should almost always work.
> 
> > > Did you test http://192.168.122.173 from another PC on your network?
> 
> > (This does not work from another PC)
> 
> @deltacloud9 can you ssh into 192.168.122.173 from another PC on your network?
> I tried using PuTTy to ssh into this IP from my main workstation and was unable to estaplish a connection.

> Do you need to adjust your VM configuration to make 192.168.122.173's ports (especially port 80) available to other PC's on your network?
> 
> (If using VirtualBox, one sometimes needs to change from NAT to `Bridged Adapter` within [actual VM] > Network > Adapter 2 > Enable Network Adapter > Attached to > Bridges Adapter.)
>I tried changing my VM network setup from NAT to Bridged Adapter and it seems that when doing that my KDE doesn't load after the initial login screen? Not sure if the two are related but when switching back to NAT the KDE desktop then loads after logging in.

deltacloud9, 2021-07-21:
Reinstalling my Debian 11 build and noticed that Debian 11 doesn't include /usr/sbin by default.  The iiab install script is calling /usr/sbin/useradd. I modified my PATH var to include it.

Small catch but may cause issues down the line.

holta, 2021-07-21:
> Reinstalling my Debian 11 build and noticed that Debian 11 doesn't include /usr/sbin by default. The iiab install script is calling /usr/sbin/useradd. I modified my PATH var to include it.

Never seen that before but good catch if this is failing...

Are you talking about https://github.com/iiab/iiab-factory/blob/master/iiab#L94 or something else?

Please include a photo or screenshot for context, Thanks Much @deltacloud9.

jvonau, 2021-07-21:
Couldn't hurt to define PATH in 'iiab' to cover all locations like zzz_iiab.sh does as an installed hack as /usr/sbin is intentally not on the PATH as a 'feature'. Using a full path to useradd is an alternative also.

deltacloud9, 2021-07-21:
@holta you are correct. It lies in the install.txt. Here is a screenshot.
[image]

@jvonau it may be better to define PATH just in case but yeah using full path to useradd is probably fine too

holta, 2021-07-21:
`/usr/sbin/useradd` is now fixed by https://github.com/iiab/iiab-factory/commit/f1cea5834b0307a278541ecbe8b39e4d45fcece9

This raises another question too — as to whether command-line use of `sudo iiab` will also fail on Debian.

This strongly suggests that the `iiab` install script should probably be saved to `/usr/bin` — instead of the current `/usr/sbin`

jvonau, 2021-07-21:
Once zzz_iiab.sh takes effect in /etc/profile.d the PATH should include /usr/sbin and should find iiab

holta, 2021-07-21:
> Once zzz_iiab.sh takes effect in /etc/profile.d the PATH should include /usr/sbin and should find iiab

Good point, but unfortunately that will only happen after a reboot, making our `sudo iiab` instructions too frustrating and confounding.

`/usr/sbin/iiab` really needs to move to `/usr/bin/iiab` — so Debian newcomers aren't driven crazy.

deltacloud9, 2021-07-21:
Current install of Debian 11 seems to be running smoothly.

Current diagnostics: http://sprunge.us/uA1dn5?en

When rebuilding this VM I set the network to bridged network. (I think it was having issues when trying to change it later after the VM was created).

Small bug I am noticing is that iiab-admin's password isn't working in the http://box/admin console.

Works fine in the terminal but not in the console.

I reset the password in the terminal a few times and retested every time but it just says login failed.

jvonau, 2021-07-21:
> This strongly suggests that the `iiab` install script should probably be saved to `/usr/bin` — instead of the current `/usr/sbin`

The basic old school linux convention is for 'root level access' to go into /sbin as a reminder to use sudo with the command... but systemd and friends are in /usr/bin.. ick, progress?

jvonau, 2021-07-21:
> > Once zzz_iiab.sh takes effect in /etc/profile.d the PATH should include /usr/sbin and should find iiab
> 
> Good point, but unfortunately that will only happen after a reboot, making our "keep trying `sudo iiab`" command-line instructions too frustrating and confounding.
>

The only reason for the install to stop is to reboot for an apt update..  That is the only time one has to type `sudo iiab` to continue.
 
> `/usr/sbin/iiab` really needs to move to `/usr/bin/iiab` — so Debian newcomers aren't driven crazy.

Don't switch now, two potential copies floating around the path is troublesome, stick with the correct way.

jvonau, 2021-07-21:
Or teach new users the correct layout https://www.geeksforgeeks.org/linux-file-hierarchy-structure/ and why /usr/sbin is not normally on the path as a security feature and alter the docs to use 'sudo /usr/sbin/iiab'

holta, 2021-07-21:
> The only reason for the install to stop is to reboot for an apt update.. That is the only time one has to type `sudo iiab` to continue.

IIAB installs fail due to networking and mirroring glitches all the time.

That's the core reason that documentation is lucky emphatic that `sudo iiab` needs to tried repeatedly.

holta, 2021-07-21:
> Or teach new users the correct layout https://www.geeksforgeeks.org/linux-file-hierarchy-structure/ and why /usr/sbin is not normally on the path as a security feature and alter the docs to use 'sudo /usr/sbin/iiab'

I disagree.

With respect, it really is important that things be very hassle-free for newcomers.

jvonau, 2021-07-21:
/usr/local/sbin or /usr/local/bin should be also be on the default path provided by the OS. The reset of the iiab-* utilities that need sudo live in /usr/bin anyways so when in rome...  If iiab moves to /usr/bin then the ugly zzz_iiab.sh hack can be, in the interests of security, removed also?

holta, 2021-07-21:
1) I'm now thinking that PR iiab/iiab-factory#174 is possibly not necessary, as `sudo iiab` currently works even when `/usr/sbin` is not in a regular user's $PATH — the reason is that root itself should always have `/use/sbin` in its own PATH.

   FYI I've tested this on Debian 10.10 (headless).

   Though `/usr/local/sbin` and `/usr/local/bin` are also viable options if necessary — as mentioned below.

> /usr/local/sbin or /usr/local/bin should be also be on the default path provided by the OS. The reset of the iiab-* utilities that need sudo live in /usr/bin anyways so when in rome... If iiab moves to /usr/bin then the ugly zzz_iiab.sh hack can be, in the interests of security, removed also?

2) Excellent Points All Around.

   I'm in favor of deprecating [zzz_iiab.sh](https://github.com/iiab/iiab/blob/master/roles/2-common/templates/zzz_iiab.sh) if we can modernize, to pave the way for a 1st pass security audit in the coming year.

holta, 2021-07-21:
Now I'm confused:

Running `sudo iiab` from a non-root account (lacking `/use/sbin` in its path, on Debian 10) also works to invoke `/usr/sbin/useradd` !

@deltacloud9 is there any chance you accidentally ran...

```
curl d.iiab.io/install.txt | bash
```

...instead of...?

```
curl d.iiab.io/install.txt | sudo bash
```

_Or is Debian 11 somehow profoundly different?!_

jvonau, 2021-07-21:
Could of sworn I saw an issue in my email about iiab-admin's password...  @deltacloud9 all good?

holta, 2021-07-21:
> Running `sudo iiab` from a non-root account (lacking `/usr/sbin` in its path, on Debian 10) also _works_ to invoke a `/usr/sbin` script containing `useradd` !

I've reconfirmed this with the very latest pre-release version of Debian 11 (without any graphical desktop) on a x86_64 / amd64 VM.

holta, 2021-07-21:
FYI my installation of [BIG-sized](http://faq.iiab.io/#What_services_.28IIAB_apps.29_are_suggested_during_installation.3F) IIAB onto Debian 11 Server VM went flawlessly overnight &mdash; its [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) output is here:

http://sprunge.us/Phf3oi?en

I'll need to catch up with @deltacloud9, as my own non-root account `iiab-admin2` (that installed `iiab-admin` and then IIAB) faced no problems at all when running `curl d.iiab.io/install.txt | sudo bash` == `sudo iiab` &mdash; even despite the fact that this non-root user (itself) does not have `/usr/sbin` in its $PATH !

Conclusion?  `/usr/sbin/useradd` (including full path) as implemented by https://github.com/iiab/iiab-factory/commit/f1cea5834b0307a278541ecbe8b39e4d45fcece9 and PR https://github.com/iiab/iiab-factory/pull/174 both appear (to me) as 100% useless for those who install Debian 10/11 in the conventional manner.  So my suspicion (I might be entirely wrong!) is that @deltacloud9 is installing Debian in a different way from most every one else &mdash; e.g. possibly in a way that causes root have `/usr/sbin` removed from its $PATH ??

jvonau, 2021-08-30:
Any reason to keep this issue open given Bullseye was released 2 weeks ago?

holta, 2021-08-30:
> Any reason to keep this issue open given Bullseye was released 2 weeks ago?

We should close it soon, good idea.

@deltacloud9 are you able to do an [IIAB install](https://download.iiab.io) onto Debian 11 "Bullseye" on bare metal, e.g. if you have an old laptop or similar, with a working internal AP ideally?

deltacloud9, 2021-08-30:
Yes, I will run an install of this this week and let you know my findings.

holta, 2021-08-30:
> Yes, I will run an install of this this week and let you know my findings.

Many thanks @deltacloud9 !

Please post output from [sudo iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) if poss, whether or not you get the internal AP [Access Point, i.e. hotspot] working!

holta, 2021-11-10:
Good enough!

Thanks to @icebox827 and many others who've helped testing IIAB quite rigorously on VM's and others Debian "Bullseye" machines!

closed 2021-11-10 by holta