iiab/iiab #2753: Sentinel-2 Satellite Product

webmap2015, 2021-04-27:
Hi,

In this project, Sentinel-2 cloudless 2018 is being used.
According to https://s2maps.eu/, there are much more Sentinel-2 data products between 2016 - 2020.
Is there any info about usage policy for downloading and using at non-internet network for free?

Thanks

holta, 2021-04-27:
Can you please be somewhat more specific...explaining what you're trying to accomplish?

webmap2015, 2021-04-27:
For example, "OpenStreetMap Database" or "SRTM Digital Elevation Data" are open source data, so they can be freely downloaded, maps can be produced in them, and used freely. Similarly, sentinel2 data is free and some of the data is **cloudy**.

According to https://cloudless.eox.at/, it is a stunning view of the Earth, without a single cloud in sight.

And this product also seems to be paid.

[image]