iiab/iiab #2758: and usb1 -> /media/usb1 can still appear in the mounted device

jvonau, 2021-04-28:
and usb1 -> /media/usb1 can still appear in the mounted device

_Originally posted by @tim-moody in https://github.com/iiab/iiab/issues/2282#issuecomment-603999619_

jvonau, 2021-04-28:
Created by /etc/usbmount/mount.d/00_create_model_symlink, supplied by the usbmount package.

jvonau, 2021-04-28:
https://github.com/iiab/iiab/pull/2282#issuecomment-604426628

holta, 2021-04-28:
Anyone feel free to revise the ticket subject line, if that helps clarify the core question(s) ?!

jvonau, 2021-04-28:
[unmount could be cleaner?]  Think the unmounting is fine, the issue is none of the /etc/usbmount/unmount.d/ scripts are triggered anymore needing workarounds like #2326, so the link is never cleaned up. Just remove the mount.d/00_create_model_symlink action and the problem link is not created to begin with, there would be nothing to clean up upon unmounting.

holta, 2021-04-29:
Should be resolved by PR #2760.

(Or re-open as nec!)

closed 2021-04-29 by holta